lvgl/lvgl #3614: tabview pages not being CLICK_FOCUSABLE causes bug for dropdown lists.

RubBra, 2022-08-24:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

in lv_tabview.c (line 75) the page in the content (i.e. the background of the content area) is set to not be LV_OBJ_FLAG_CLICK_FOCUSABLE. This causes widgets that change on LV_EVENT_DEFOCUSED to not change as expected. One example of this is lv_dropdown.c (line 659) where an expanded list doesn't get closed when dragging from one tab to another.

### To Reproduce

```c
void minimal_example(void) {
lv_obj_t * tv = lv_tabview_create(lv_scr_act(),LV_DIR_TOP,LV_VER_RES*0.1);
lv_obj_t * t1 = lv_tabview_add_tab(tv, "Tab1");
lv_obj_t * t2 = lv_tabview_add_tab(tv, "Tab2");
lv_obj_t * dd = lv_dropdown_create(t1);
lv_obj_align(dd,LV_ALIGN_TOP_MID,0,0);
lv_dropdown_set_options(dd, "item1\n"
                            "item2\n"
                            "item3");
}
```

### Expected behavior
The dropdown list should close when draging the background

### Screenshots or video
[image]

kisvegabor, 2022-08-28:
I see the problem and your solution sounds good. Can you send a pull request?

RubBra, 2022-08-28:
I am not sure what the solution is. 
Removing lv_tabview.c (line 75) fixes the bug, but there must be a reason for that line, thus I am not confident that unilaterally removing it doesn't cause unintended side effects.  I'll make a PR with the change and my concerns.

kisvegabor, 2022-08-28:
I thought people usually don't want the tabview page to be focused, but now I'm not sure it's really true. E.g. it can be focused to scroll it with arrows of a keyboard. 

But v9 is still in development so we can get feedback how it work in the practice.

commit a81dbd33 (2022-08-28): fix: remove tabview content not click focusable causes bugs in widgets reacting to focus changes(#3633)

fixes #3614
closed 2022-08-28 by kisvegabor