lvgl/lvgl #3618: Adding a new GPU accelerator (GD32 IPA), however run into a bug and not sure how to proceed

trollcop, 2022-08-25:
I am working on custom hardware with GD32F450I on it, and while the performance is fine, I wanted to try IPA (GD32's version of STM32 DMA2D) and see how it works. I am using `lv_widgets_demo` to exercise lvgl, with nothing else running on hardware.

I've made a copy of stm32_dma2d and replaced everything with gd32_ipa and wrote all the hardware-specific stuff following the same things STM32 version did.

To rule out everything else, I've #if 0'd all GPU code other than the simplest FILL implementation. I don't mind that blending/etc isn't done, as I'm OK with the black/random noise rectangles.

```
static void lv_draw_gd32_ipa_blend_fill(lv_color_t * dest_buf, lv_coord_t dest_stride, const lv_area_t * fill_area,
                                           lv_color_t color)
{
    /*Simply fill an area*/
    int32_t area_w = lv_area_get_width(fill_area);
    int32_t area_h = lv_area_get_height(fill_area);
    invalidate_cache();

    IPA_CTL = IPA_FILL_UP_DE;
    IPA_DPV = color.full;
    IPA_DMADDR = (uint32_t)dest_buf;
    IPA_DLOFF = dest_stride - area_w;
    IPA_IMS |= ((area_w << 16U) | (area_h));

    /*start transfer*/
    IPA_CTL |= IPA_CTL_TEN;
}
```

As you can see, it's pretty much 1:1 copy with same parameters as the STM32 version:

```
    /*Simply fill an area*/
    int32_t area_w = lv_area_get_width(fill_area);
    int32_t area_h = lv_area_get_height(fill_area);
    invalidate_cache();

    DMA2D->CR = 0x30000;
    DMA2D->OMAR = (uint32_t)dest_buf;
    /*as input color mode is same as output we don't need to convert here do we?*/
    DMA2D->OCOLR = color.full;
    DMA2D->OOR = dest_stride - area_w;
    DMA2D->NLR = (area_w << DMA2D_NLR_PL_Pos) | (area_h << DMA2D_NLR_NL_Pos);

    /*start transfer*/
    DMA2D->CR |= DMA2D_CR_START_Msk;
```

Which basically is exactly same thing, with same inputs and should be same outputs.
However, after ~4 calls to this function, program ends up in hardfault with a backtrace of ~10 functions (screenshot attached), which probably means the DMA2D fill is overflowing one of the 2 image buffers in SRAM and crapping over stack and/or other vars.
[image]

Now I've read the GPU docs and I see that as far as I'm doing things, everything is done exactly as it should be.
However, I have a suspicion since I'm running two SRAM buffers, of the size 800x20 (lines), and the copy takes ~5 calls to crash, it has something to do with the copy not limiting Y-coord for the fill/whatever routine and it ends up going past the SRAM buffer into somewhere else.

Also, looking at the LCD, the copies are correctly reflected at the top of the screen until it reaches around ~40? pixels down and then crashes, and there's no distortion in the copied image etc, so the copy is working correctly (with the parameters given).

However, if this is the case, and STM32 DMA2D is using exactly same parameters, that means somewhere down the line (between 8.x?? when it worked and 8.3 (what I am using now), something regressed.

I do have a STM32F429-Disco that I should be able to hookup and test on, but it would take some time to setup. ( I will do it anyway, just not immediately ).

Soo... that said, does one of the maintainers have a STM32 + LTDC + DMA2D test environment where they can check building with `LV_USE_GPU_STM32_DMA2D` enabled and see if it actually works?

Eventually once I bugfix this I plan to contribute the GD32 IPA driver if there's interest, so it can be useful to others.

kisvegabor, 2022-08-25:
Hi,

I've verified `LV_USE_GPU_STM32_DMA2D` with the `release/v8.3` branch on an STM32F769 Discovery and it worked well.

I agree with your conclusion, that probably it's buffer out of bounds write. Are you sure the color format of your GPU is the same as `LV_COLOR_DEPTH`. This is typical issue e.g. if the GPU is in ARGB8888 mod and LVGL is in 16 bit mode. 

To debug it, I suggest manually setting e,g, `area_w = 10` and `area_h = 2` and check if really the expected number of pixel are filled with the expected format.

trollcop, 2022-08-25:
Hi,

Yes, the pixel format is same (RGB565). Just like in STM32 version, it's set during ipa_init:

```
    IPA_DPCTL &= ~(IPA_DPCTL_DPF);
    IPA_DPCTL = LV_IPA_COLOR_FORMAT;
```

But I've also added it into the simple_fill part as well, just in case. Anyhow, I stepped through the code and DPCTL remains correct pixel format, so I don't think that's the problem.

Since the first ~40 or so rows are copied distortion free I don't think its related to pixel format, either.
See attached screenshot.
[image]

> I've verified LV_USE_GPU_STM32_DMA2D with the release/v8.3 branch on an STM32F769 Discovery and it worked well.

Can you share the buffering/framebuffer arrangement in your case?
I believe 769-Disco is using MIPI for LCD, right? Is framebuffer still in SDRAM?

My current arrangement is:
1) Framebuffer in SDRAM (full screen)
2) LTDC Layer 0 mapped to the framebuffer from (1)
3) 2 buffers of 800x20 pixels (800x20x2 bytes) declared for double-buffered operation in `lv_disp_draw_buf_init`.
4) flush_cb starts a line-by-line M2M DMA which copies from temporary buffers to SDRAM, and then fires off another DMA transfer in transfer_complete callback, until entire updated region is copied, then calls `lv_disp_flush_ready`.

I believe this is a normal/standard way of handling things.

Looks like it's time for me to try and repeat the same issue on F429-disco.

kisvegabor, 2022-08-25:
It's great that you have found to root of the problem. 

Could you measure the performance of the GPU compared to SW fill? STM's DMA2A is usually slightly slower than SW rendering. Hopefully GD_IPA can do a better job. 

I have contact to GigaDevice and will inform that about this issue.

> My current arrangement is [...]

I think it's the fastest configuration, assuming the small buffers are in the internal RAM

trollcop, 2022-08-25:
OK, I ran lv_demo_benchmark() and it crashes (even with extra line expanded framebuffers). I've glanced on the FPS numbers compared to running without GPU, up to the moment where it crashes, but they look similar, probably within +- 1 FPS.
The code is compiled with -Ofast (armclang), tho running with O0 doesn't have any effect on crashing either.

So not only buggy, but likely not worth it performance-wise. meh.

kisvegabor, 2022-08-28:
If it's buggy, and not faster I think it's not a good idea to support it right now. :slightly_frowning_face:

trollcop, 2022-08-28:
Yes i agree, however having support out there might get the hardware fixed in next revision. I've opened a ticket with GD but not holding my breath.

As it is right now, unless I'm missing something fundamental about it's usage, it's not usable for 2d acceleration.

I'm going to try using it for copying from temp buffers to SDRAM (where it can't screw up other vars) and where it should be more visual to see what/ how it's screwing up.

trollcop, 2022-08-29:
I have some updates. It seems that if the DMA is entire width of LCD, it works.
If the region is smaller (i.e. DLOFF/FLOFF are involved), the copy will be random:

Flushing 0,305-799,324 (800x20) - 20000b34
Flushing 0,325-799,344 (800x20) - 200094b4
Flushing 0,345-799,364 (800x20) - 20000b34
Flushing 0,365-799,384 (800x20) - 200094b4
Flushing 0,385-799,404 (800x20) - 20000b34
Flushing 0,405-799,424 (800x20) - 200094b4
Flushing 0,425-799,444 (800x20) - 20000b34
Flushing 0,445-799,464 (800x20) - 200094b4
Flushing 0,465-799,479 (800x15) - 20000b34
these succeed (full-screen refresh)
Flushing 382,0-799,37 (418x38) - 200094b4
Flushing 382,38-799,74 (418x37) - 20000b34
these look like crap (random noise)

I've also did a test replacing line-by-line M2M DMA for framebuffer->SDRAM copy with IPA, and the results were similar, full-width updates worked, anything off-center caused random noise:

```c
    int32_t dest_w = act_x2 - act_x1 + 1;
    int32_t dest_h = act_y2 - act_y1 + 1;
    uint32_t offset = ((LCD_WIDTH) * act_y1 + act_x1);

    IPA_CTL = IPA_FGTODE | IPA_INT_FTF;
    IPA_FMADDR = (uint32_t)color_p;
    IPA_DMADDR = (uint32_t)&framebuffer[offset];
    IPA_FLOFF = LCD_WIDTH - dest_w;
    IPA_DLOFF = LCD_WIDTH - dest_w;
    IPA_FPCTL = IPA_DPF_RGB565;
    IPA_IMS = ((dest_w << 16U) | (dest_h));
    fprintf(stderr, "Flushing %d,%d-%d,%d (%dx%d) - %p\n", act_x1, act_y1, act_x2, act_y2, dest_w, dest_h, color_p);

    /*start transfer*/
    IPA_CTL |= IPA_CTL_TEN;
```

(not shown is IPA_IRQHandler which basically checks FTF and calls `lv_disp_flush_ready`.)

I've also tried switching LVGL to 32-bit mode (ARGB8888) and using IPA_FGTODE_PF_CONVERT to convert src 8888 to 565, with identical effects: full-width conversion worked, stuff in the middle of the screen didn't.

trollcop, 2022-08-31:
OK, did some more testing. If destination for fill is in SRAM, there's a possibility of corruption. I moved both draw buffers to SDRAM temporarily, and the crashes stopped. There's a crash at benchmark 86 or so (colored/opa arcs) in `remove_free_block` but that might be just a problem with 8.3.0, I did not test it on master.

Anyway, since corruption only occurs when destination is SRAM this is definitely something GD32 related as same stuff in DMA2D works.

I've got a contact window open with GD32 FAE but I'm not hoping for much here, because first thing they did was asked me where I got Device_limitations_of_GD32F45x_F40x_Rev1.0.pdf from (the GD32F4 errata pdf, linked together with the rest of the docs off gd32mcu.com).

kisvegabor, 2022-08-31:
So there are good and bad news too: the GPU is working but it's really broken is some cases. 

Meanwhile I got this from GD. I'm not sure if it helps once you found that the bug is in the hardware for sure. 
[lv_gpu_stm32_dma2d.c.zip](https://github.com/lvgl/lvgl/files/9464870/lv_gpu_stm32_dma2d.c.zip)

trollcop, 2022-09-01:
OK, so I *was* doing the same thing, but only backing up one pixel *and* my backups were not volatile.
After using thier method of doing 2 backups, waiting for GPU to finish, and then restoring 2 memory locations, the entirety of benchmark suite passes (including no longer crash at test 86 or so.

Was there any explanation with the code they sent, like why they're backing up 2 pixels and not just one which was according to errata?

kisvegabor, 2022-09-02:
No, that's all what I got. :slightly_frowning_face: 
If you send me an email address to kisvegabor@gmail.com I can cc you to the conversation.

embeddedt, 2022-09-17:
Discussion continued in #3622.

closed 2022-09-17 by embeddedt