lvgl/lvgl #3620: LodePNG: Incompatible pointer type in "decoder_open"

jacquesdmernok, 2022-08-25:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
In the simulator I can use the lodePNG lib fine, but when I port to STM32, I get incompatible pointer type errors in the decoder_open function.

### To Reproduce

kisvegabor, 2022-08-25:
Thanks for the report, I've pushed a fix.

closed 2022-08-25 by kisvegabor

jacquesdmernok, 2022-10-11:
> Closed #3620<https://github.com/lvgl/lvgl/issues/3620> as completed via 74cbed2<https://github.com/lvgl/lvgl/commit/74cbed21e6085fa5e61d27f2beeef6cee1a9aa01>.
>
>
> Email Disclaimer | This email is subject to a disclaimer which may be found on our website by clicking on this link: http://mernok.co.za/images/ME-DOC-LEG-DIS-002-01_EmailDisclaimer.pdf | Everything in this email and its attachments relating to the official business of MERNOK ELEKTRONIK is proprietary to MERNOK ELEKTRONIK. If the email is used other than for official business of MERNOK ELEKTRONIK or the views and opinions expressed in the email are not authorised by MERNOK ELEKTRONIK, the views and opinions expressed are those of the individual sending the email. The content of this email is confidential, legally privileged and protected by law. The person addressed in the email is the sole authorised recipient. Please notify the sender immediately if this email and its attachments have unintentionally reached you; do not read, copy, disseminate or use the content in any way and delete the email and any copies of it. Whilst all reasonable precautions are taken to ensure the accuracy and integrity of the information, and that this email and its attachments are free from any virus, MERNOK ELEKTRONIK accepts no liability however arising or responsibility whatsoever in this regard, and in keeping with good computing practice, the scanning of files and attachments is advised.
>
> Email Disclaimer | This email is subject to a disclaimer which may be found on our website by clicking on this link: http://mernok.co.za/images/ME-DOC-LEG-DIS-002-01_EmailDisclaimer.pdf | Everything in this email and its attachments relating to t [... cut ...] curacy and integrity of the information, and that this email and its attachments are free from any virus, MERNOK ELEKTRONIK accepts no liability however arising or responsibility whatsoever in this regard, and in keeping with good computing practice, the scanning of files and attachments is advised.

Thanks so much for the fix, but it seems as though there were two instances of this warning. Just a few lines up from where you implemented the fix, there is another instance.

Kind Regards,
Jacques Duminy
Junior Electronic Design Engineer
Mernok Elektronik (Pty) Ltd
tel +27 12 661 4828
fax +27 86 542 9007
***@***.******@***.***>
www.mernok.co.za<https://eur02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fprotect-za.mimecast.com%2Fs%2FV7P-CRgnPlcy2DkFNi1SH%3Fdomain%3Dmernok.co.za%2F&data=05%7C01%7Chabib.c%40powermite.co.za%7Cb430d8ac6cf441d1008908da749ff627%7Cf66ea97f8cc74a2896e5539105b2cf59%7C0%7C0%7C637950527627834019%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=pQ0mtqwKoACG7MABw4YWbfRdFBjYyNGtvPnbW93CT24%3D&reserved=0>
***@***.***<https://eur02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fprotect-za.mimecast.com%2Fs%2F-AMSCVmr7vuzGnLuyBU-w%3Fdomain%3Dtickets.tixsa.co.za&data=05%7C01%7Chabib.c%40powermite.co.za%7Cb430d8ac6cf441d1008908da749ff627%7Cf66ea97f8cc74a2896e5539105b2cf59%7C0%7C0%7C637950527627834019%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=ZP7ACLzt%2BtPnntz73HWe8bOs30dYgzqX7jmSedJHK9I%3D&reserved=0>
***@***.***

kisvegabor, 2022-10-11:
These seem to be fixed too:
https://github.com/lvgl/lvgl/blob/master/src/libs/png/lv_png.c#L170-L171

commit 74cbed21 (2022-08-25): fix(png): fix warning

fixes #3620