lvgl/lvgl #3628: lv_obj_is_valid can return true after lv_obj_del

boraozgen, 2022-08-26:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

`lv_obj_is_valid` can return true after `lv_obj_del` if the address pointed by the object has been allocated to another object.

### To Reproduce

```c
lv_obj_t* obj1 = lv_obj_create(NULL);
LV_ASSERT(lv_obj_is_valid(obj1));
lv_obj_del(obj1);
LV_ASSERT(!lv_obj_is_valid(obj1)); // passes
lv_obj_t* obj2 = lv_obj_create(NULL);
LV_ASSERT(!lv_obj_is_valid(obj1)); // fails
```

### Expected behavior

IMHO this is not a solvable problem, therefore maybe a warning should be added to the documentation or the function should be removed.

### Screenshots or video

embeddedt, 2022-08-26:
I *think* this is intended behavior. The idea of the function is to check that the provided pointer points to any valid LVGL object, not necessarily the originally allocated one. Let's see what @kisvegabor says.

kisvegabor, 2022-08-26:
> IMHO this is not a solvable problem, therefore maybe a warning should be added to the documentation or the function should be removed.

I agree that it's not solvable problem. If `obj1` points to valid object (intentionally or by accident) than it's a valid object. I think it's the user responsibility to set `obj1 = NULL` when it's deleted.

boraozgen, 2022-08-29:
I thought so. Thank you for the clarification.

closed 2022-08-29 by boraozgen