lvgl/lvgl #3638: New line with \n in the span widget

kisvegabor, 2022-08-29:
### Introduce the problem
Adding `\n` in spangroup's span does nothing 

### Examples and cases
```c
	lv_obj_t * sg = lv_spangroup_create(lv_scr_act());
	lv_obj_center(sg);
	lv_obj_set_style_bg_opa(sg, 100, 0);
	lv_obj_set_style_bg_color(sg, lv_color_hex3(0xf00), 0);
	lv_span_t * s1 = lv_spangroup_new_span(sg);
	lv_span_set_text(s1, "abc\n");
	lv_span_t * s2 = lv_spangroup_new_span(sg);
	lv_span_set_text(s2, "1\n2\n3\n");
```
[image]

To mention a real life example, I need to realize this "label group" and wanted to use a single span group:
[image]

### Suggested solution

By looking into the code I haven't found any obvious solutions. @guoweilkd could you take a look at it?

closed 2022-09-21 by lvgl-bot

dns-xyz-glk, 2023-04-25:
Shame that this idea has be abandoned. It would be really nice to be able to split text into lines or paragraphs. Doesn't make really much sens for a span to hold an uninterruptible sheet of text of different styles, like in the docs example.

kisvegabor, 2023-04-26:
Unfortunately, I didn't  have time to dive into it yet. 

If you are interested in addig support for `\n`, a Pull request would be very welcome!