lvgl/lvgl #3643: Multi-threaded drawing

anheiqq, 2022-08-31:
Hello, I want to implement multi-threaded drawing. For example, I want to use 'lv_print_line' to draw lines directly to a memory canvas in a sub-thread, and these drawings are not output to the screen. I have rewritten the 'init_fake_disp' function and excluded the influence of the '_lv_refr_get_disp_refreshing' function, but I found that the drawing lines are still affected by 'lv_mem_buf', '_lv_draw_mask_list', etc. What should I do, and are there any plans to support multithread drawing in the future?

kisvegabor, 2022-08-31:
In v9 (under development) we are planning to implement multi thread drawing. It requires significant rework in the algorithms and a lot of things should be considered. 

Before jumping into it there should be a longer discussion to the find out what would be the best way to deal with it. But once you opened this issue, let's start this discussion :slightly_smiling_face: 

The questions I see:
- How to divide the the tasks among threads?
   - With partial buffering (e.g. 1/10 screen sized buffer), 1 part to 1 thread? We need as many draw buffer as many threads are used.
   - With normal double buffering we also need to divide the rendering into part so can we say it's working only with partial buffering?    
- How to start a new thread?
  - E.g. is there a `draw_thread_create_cb` in the display driver and the max threads are set too? The thread exists when it finished its area and a new thread is created if needed.    
- All global render related data (e.g. masks) should be moved to `draw_ctx`. I see no problem with that.
- How to handle GPUs? Any special consideration?  cc @GorgonMeducer 

As an example use case we can consider 
- ESP32 with 2 cores
- Raspberry with 4 cores?
- Other?

anheiqq, 2022-08-31:
About the opinion All global render related data should be moved to draw_ctx,I think is very correct!  and it should not be affected by the global variable 'disp_refr' when drawing. 
Also when using ’lv_canvas_draw_line‘ to draw, every time the 'lv_canvas_draw_line' function is executed, 'lv_obj_invalidate(canvas)' will be called to refresh, which will make the cup occupancy very high. It is recommended to freely control when to execute 'lv_obj_invalidate' or 'lv_obj_invalidate_area' to do all or partial refresh,  For example, the drawing of real-time ECG, as shown in the following figure。

[image]

GorgonMeducer, 2022-08-31:
@kisvegabor For multi-threaded drawing, GPU should be treated as a normal thread at least in the eyes of the upper layers. 
So:

- SW drawing works in an OS task
- GPU drawing works in parallel 

Both of them work asynchronously. 

A typical way of handling this is 

- Introduce a pipeline of drawing-tasks
- provide a descriptor for each drawing-task
- dispatch drawing tasks to SW thread or GPU in the backend
- Implement a dedicated barrier to sync those drawing tasks when required

kisvegabor, 2022-09-01:
@anheiqq 

> It is recommended to freely control when to execute 'lv_obj_invalidate' or 'lv_obj_invalidate_area' to do all or partial refresh, For example, the drawing of real-time ECG, as shown in the following figure。

`lv_disp_enable_invalidation(disp, false/true);` can be sued to enable/disable the internal invalidations. E.g. 
```c
lv_disp_enable_invalidation(disp, false);
do_something();
lv_disp_enable_invalidation(disp, true);
invalidate_manually();
``` 

@GorgonMeducer 
> SW drawing works in an OS task
GPU drawing works in parallel

If the CPU and GPU works on the same area how can we avoid to keep he CPU waiting for GPU when the GPU can't draw something? For OpenGL or similar high-end GPUs (where we assume that GPU can draw everything and there is no mixed SW*GPU rendering) I see that it works like pushing commands to a buffer and the CPU waits for completion. However, it doesn't apply to lower-end MCUs.

My idea with this feature is to utilize multiple CPU cores with or without GPU. It's a question if it's really a reasonable goal though.

cc @nicusorcitu

GorgonMeducer, 2022-09-01:
@kisvegabor please find my comments below:

- "CPU waiting" isn't a bad thing, especially when LVGL works in a dedicated thread of an RTOS. Hence, when CPU waiting, LVGL can block the thread and yield.
- For any method that can take a drawing task, we call it Processing Element. It can be a dedicated processor core, a GPU, a DMA or even a local thread. Dispatching drawing tasks needs some algorithm to balance the performance and latency, i.e. if a GPU is busy, dispatcher might decide to change a different PE, for example a local Thread. 
- multiple CPU cores have no different than GPU; they are PEs. Hence we can use the same programmers' model / low level interface for all PEs.

In summary:
- the abstraction of PE is important
- the algorithm for dispatching is important
- CPU waiting isn't bad as long as it can yield. 
    * Multi-dispatching can mitigate the so-called CPU-waiting issue.

kisvegabor, 2022-09-04:
> "CPU waiting" isn't a bad thing, especially when LVGL works in a dedicated thread of an RTOS. Hence, when CPU waiting, LVGL can block the thread and yield.

But we won't get more FPS this way even if the MCU has 8 cores.

> For any method that can take a drawing task, we call it Processing Element. It can be a dedicated processor core, a GPU, a DMA or even a local thread. Dispatching drawing tasks needs some algorithm [...]

Sounds good to think in an abstract Processing Elements. To estimate the load of a PE the PE needs to know its future tasks. In any given moment we can easily decide only if it's busy or not. 

> multiple CPU cores have no different than GPU; they are PEs. Hence we can use the same programmers' model / low level interface for all PEs.

I don't fully agree. It'd be true if PEs were exchangeable. However, embedded GPUs are limited in features. SW rendering is the fallback now, but not sure it will be abel to render everything in the future. 

What do you mean exactly by "Dispatching drawing tasks"? What is a "drawing task"? 
1. A button with radius, gradient, shadow, etc
2. Only the shadow of the button
3. The button with all its children
4. an area to to redraw which includes the button
5. Something else?

2) makes sense to me. If LVGL wants to draw something it asks which PE can draw it. But we should consider that we cannot assign a CPU core to draw a button's gradient bg and an other to draw its border because they would draw to the same area. So there is a dependency between the drawing tasks. Which  can be easily resolved by working on independent areas in parallel. Which leads to 4)

nicusorcitu, 2022-09-05:
If we use a task/thread only for 2) will not lead into creating too many of those? A larger parallelism will lead into bigger CPU loading. The boards with a single CPU will suffer.

I believe 4) is a good solution too. The purpose while using the GPU is to make it work in parallel with CPU as much as possible.
One of the idea is to have a drawing descriptor for each area (it can be 1/10 of the screen) and compare with the GPU capabilities. If there is a match then it can be dispatched to GPU that task.
The GPU is effective if we submit in its queue multiple commands that it can handle. Having an area handled completely by the GPU will clearly increase the performance.
The problem is what happens if we don't have any area that can be fully supported by the GPU? Will we go further to split even more that area into peaces that can be processed by GPU?!

kisvegabor, 2022-09-05:
> The problem is what happens if we don't have any area that can be fully supported by the GPU? Will we go further to split even more that area into peaces that can be processed by GPU?!

Good questions. E.g. a large panel might not work with GPU because it has 2px radius. So the rectangle could be drawn by the GPU except the tiny corners. 

To have a practical example: let's say there is a 800x80 area (1/6 of a 800x480 screen) with buttons, labels, charts, images, etc. We subdivide this area into 80x80  blocks in the hope that GPU can fully render a smaller block.

So we have 10 areas. Each can contain fragments of buttons, labels, charts, sliders etc. We can say that the CPU and GPU are not equal as we prefer the GPU to render. So we can say: "Hey GPU, can you fully render area1?" If it say yes, fine, render it. While it's working check if "area2" can be rendered with GPU or not. And so on... Let say  4 out the 10 areas were working with the GPU and remaining 6 is CPU only or mixed where the GPU can help with some parts. We can leave this areas to the CPU as the GPU is too busy and it isn't worth to wait for it just to fill a 80x80 area.

So in summary:
1. Subdivide the area
2. CPU checks which areas are fully GPU compatible
3. Start rendering the areas with GPU. I.e. collect and load the commands to the GPU to make it work in the background.
4. While the GPU is working render the CPU only or mixed area with CPU or multiple CPU cores

I'm not sure it's the right direction, just thinking aloud. However it makes sens to me. this way simpler GPUs (like NXP PXP or STM Chroma Art) could be used more effectively because there wouldn't be "wait for the GPU ready" lock but the CPU(s) and GPU could work fully in parallel.

It requires some deep rework in LVGL ot effectively collect the draw tasks and check if it GPU compatible or not, but it seems doable.

GorgonMeducer, 2022-09-05:
> I don't fully agree. It'd be true if PEs were exchangeable. However, embedded GPUs are limited in features. SW rendering is the fallback now, but not sure it will be able to render everything in the future.

Do you remember I mentioned first that each PE would have a capability descriptor? so it is not something "exchangeable" but a purely capability-based task assignment.  

> What do you mean exactly by "Dispatching drawing tasks"? What is a "drawing task"?

You define what is a "drawing task", it could be an arc, a line etc.

Each drawing task has a descriptor, and the dispatcher will only assign a task to a PE capable of fulfilling the requirement. 

You will immediately find that the algorithm of dispatcher is the key to improve performance. And it is the next stage. We should first finish such framework with a simple dispatcher algorithm, something like:

- Search a Free-PE list and check its capability
    * if it is capable to do a drawing-task, then assign the task to that PE

In the next stage, we might find a way to sort the Free-PE list, e.g. put the PE with less capability in the front part of the list etc.

I hope this answers your question.

kisvegabor, 2022-09-07:
> It appears that you only care about LVGL or FPS, but GUI is only part of a system

I might be biased :slightly_smiling_face:  However, it's worth mentioning the GUI is usually the most computational heavy task which requires quite fast response (30-60 FPS). 

If I understood correctly with your suggestion all we needed to do is implementing a (user extendable) dispatcher (probably in draw_ctx) which can decide e.g. in lv_draw_arc which PE to use. And the difficulty is how we construct the dispatchers to make the PEs as busy as possible.  Actually, such dispatcher already exists implicitly in NXP drivers where they need to choose from 2 GPUs and and a CPU. If so, my problem with that it doesn't really improves performance "just" lowers CPU usage. 

Just to cover a broader spectrum let's consider an MCU with 2 homogeneous cores (e.g. ESP32). So we have 2 PEs. To take advantage of them, they cannot work on the same area as one core will block the other. (Okay, one will be free for other tasks, but it'd be case with the current singe core architecture too.) 

So to make use of the 2 cores they should work on 2 different areas. Not considering the memory bandwidth and display refresh time (which are significant though) we get double FPS. 

Speaking about GPU+CPU, in the end the best would be to have areas which are GPU only and would work fully parallel with CPU. On a typical UI there is a lot of space that simply need to be filled, or an image needs to be copied (basic tasks for any GPU). 

Sooner or later we will need to "see into the future" to optimally assign the drawing tasks to the PEs. I mean seeing that now an arc needs to be drawn won't be enough. We will also need to know what comes after the arc. It assumes a list of render commands in a buffer that can be read freely. 

So as I see the gist of this improvement are:
1. Create a "command buffer" from the drawing tasks. Basically the draw functions and their parameters.  It seems the most complicated one because of the draw hooks (`LV_EVENT_DRAW_...`) where the user can do anything.
2. Subdivide the draw area to smaller areas. 
3. Select which areas are GPU only and start rendering them in the background. 
4. Start the others with CPU(s) in parallel tasks 

What do you think?

GorgonMeducer, 2022-09-07:
> So to make use of the 2 cores they should work on 2 different areas.
This is the idea condition but ... in theory, it is very hard. 
In my understanding, for any graphic user interface, e.g. a window, a dialog or a panel, the key data structure is something called the element tree, and this tree structure leads to the ultimate problem: it is highly lightly that for most of the time, more than one PEs are working on the same area...

Or maybe this is our answer:

- if drawing tasks come from different element trees or branches, then those tasks can be dispatched to different PEs. Otherwise, 
- drawing tasks from the same element tree or branch must be sent to the same PE (or waiting is unavoided).

So we do not randomly divide a window into different parts but use a tree-based (branch-based) dispatching algorithm.

nicusorcitu, 2022-09-08:
>  Actually, such dispatcher already exists implicitly in NXP drivers where they need to choose from 2 GPUs and and a CPU.
>
Currently is not a dispatcher but more like a fallback mechanism. It tries to do it by GPU first, otherwise it fallback to the CPU.
But I think it can be implemented a check and see if the GPU is busy or not. If busy then to use the CPU.

> And in terms of performance, for most of the time, HW accelerators are faster than CPU. So it does indeed improve the FPS. But the first priority of those HW accelerators is improving power efficiency rather than PURELY improving the FPS
>
Please keep in mind that most of the GPUs are effective while processing large areas. If you submit a small blit area to the GPU or a short draw line, the GPU can not be faster. The GPU driver APIs will also add some extra load to the CPU which might not be smaller that the GPU execution time. Some of the GPUs require also a separate task for processing commands so this task switching is also a CPU consumer. At the end neither the CPU usage might not be released. If you call the GPU just for a small piece, then it does not worth to do it.

Waiting for GPU, as was pointed, is inevitable. For example, you can not pass the whole frame buffer to the display until you assure every area is completed.
But we should not take advantage and use "wait for GPU" when there is not needed. For example, if you submit a blit cmd to the GPU and then right away the CPU comes and "waits for GPU completion" then the GPU did not really had enough time to do its job. So this wait will eventually put the CPU to sleep. It might free the CPU usage but the performance is not better.

With the multi-threading design we have to keep in mind to make multiple PEs work in parallel without waiting (as much as possible) one for each other.

GorgonMeducer, 2022-09-08:
@nicusorcitu agree with you.

kisvegabor, 2022-09-09:
Okay, my approach is a much larger rework so I'm open to take the simpler path especially if mine doesn't have that much advantage. 

Let me summarize what we know so far:
- The ideal case would be to buffer a lot of commands to GPU because it's faster this way 
- It's important to allow sleeping the CPU (yielding task) while we are waiting for a GPU to lower CPU usage
- We don't expect much performance boost, but mainly lower CPU usage
- Larger areas are preferred as bilting small areas is slow and there is more overhead
- Waiting for GPU is inevitable so we don't target finding GPU only areas
- We call CPU, GPU and other rendering resources as PE with capability descriptors
- Assuming that there are multiple PEs, we need a dispatcher which decides to which PE a given drawing task should be assigned to. 
- A drawing task is a part of an object. E.g. rectangle border, rectangle body, a letter, a transformed image. 
- It's a possible direction to branch on non overlapping widgets and start rendering them in parallel with more PEs. 
- With 2 CPU cores two areas can be rendered in parallel to.

GorgonMeducer, 2022-09-09:
@kisvegabor YES! Exactly!

kisvegabor, 2022-09-09:
Wao, nice :slightly_smiling_face: 

How to proceed? As you mentioned we should start with something simple. I see these tasks to do as a first step:
1. Implement the concept of PEs
2. Have a basic dispatcher 
3. Modify the architecture to allow yielding. 

After that:
1. Improve the dispatcher
2. Modify the architecture to allow multi thread rendering (no global states)
3. Figure out how to spawn threads 
4. Make experiments with "branched" rendering or multiple area parallel rendering.

nicusorcitu, 2022-09-09:
> The ideal case would be to buffer a lot of commands to GPU because it's faster this way
>
Usually, if GPU is doing a blit and next command is a draw, these two have different contexts and that context is loaded each time to the GPU. I am not sure if VGLite can save its previous context or not, I will ask around. But the case here is that, ideally it will be good if we can group a bunch of commands of the same type. So the context will not be loaded for each command. In this matter we can think on a solution to reorder the draw/blit commands. This is a new request that we can think about it. Reordering, batching multiple commands and if possible caching will be a separate topic.
>Waiting for GPU is inevitable so we don't target finding GPU only areas
>
I did not really mean that way. I say that waiting for GPU is inevitable because at the end the CPU will have to process the whole buffer and prepare it for display. But I mean to wait one time per frame. Whenever is possible if we have an area that can be processed by GPU there is no waiting reason here. Indeed, with multitasking support if we wait for the GPU even on small areas, the task scheduler will assign the CPU to another task. But we need to verify if this improves things at the end.
> A drawing task is a part of an object. E.g. rectangle border, rectangle body, a letter, a transformed image.
>
I imagine this will create a big overhead for a single CPU board. There will be hundreds of tasks created for each frame.

Parallelism is a good thing (especially on multi core systems), but too much parallelism will generate too many task creation or task switching.

GorgonMeducer, 2022-09-09:
> 1. Implement the concept of PEs
> 2. Have a basic dispatcher
> 3. Modify the architecture to allow yielding.

Yes. Agreed.

But before "implementing the concept of PEs", there is an important step:

0. define the descriptor of drawing-task
    - define the data structure of drawing-task, lv_draw_task_t
    - use lv_draw_task_t to describe existing low-level draw operations 
    - extract the key attributes (requirements) from those drawing-tasks and generate a list of capabilities
    - design a capability fast matching scheme

1. Implement the concept of PEs
    - define the descriptor of capabilities
    - define the descriptor of PEs
    - create the default PE for the local software renderer
    - create PEs for existing GPU drivers
    - create a temporary dispatcher with a simple policy: try GPU PE first, and if failed, use default local SW PE.

2. Have a basic dispatcher and allow yielding
    - create a framework of dispatching which separates the dispatching into two parts:
        * common dispatch part
        * algorithm (policy list) part
      here, the dispatch algorithm is a list of policies. New policies can be added at both compile-time and/or runtime. We all know that no one policy fits for all, so more and more policy will be introduced in the future and the priority of those policy can be defined at compile-time and updated at runtime. 
      ‘lv_policy_t’ is a low-level data structure to hold a dispatching decision-making agent. For given input (conditions), each agent can return a score, and once the sum of the score reaches a threshold, a decision is made and no further policy agents will be checked against. 

    - Define a simple yielding scheme
        * Yield when all PEs are busy
        * Yield when no drawing-tasks in the waiting list can find a free-capable PE

What do you think?

GorgonMeducer, 2022-09-09:
> I imagine this will create a big overhead for a single CPU board. There will be hundreds of tasks created for each frame.

That's not true. For a single CPU board, there are at most two PEs, one for Local SW PE and one for GPU. For this kind of simple condition, at most two tasks will be created (one for each PE), and the whole process is serialised as what the current LVGL does.

kisvegabor, 2022-09-12:
I agree in general. But let's clarify 2 terms as I see we might understand different things on them:
1. block a task: for me it means the task spins in a loop and eats up the CPU. I think yielding a task is more common to say the task lets other run.
2. drawing-task: not an OS task, but a todo task.

> For the early implementation stage, we won't first generate all drawing tasks for each frame and then buffer them in the task-FIFO. A more practical way is to create a POOL for lv_draw_task_t, and the default pool size would be the number of existing PEs plus + 1 (and the ideal size would be the number of existing PEs x2).

Why PEs x2 is the the ideal FIFO size? Isn't PE x 100 better? If we see more commands in the list an advanced policy can take better decisions. (It needs more memory for sure.)

Just an idea: optionally we can separate the generation of drawing tasks and SW rendering to 2 CPU cores. Or even drawing task generation can be done by multiple cores if each core is working on a separate widget tree. I'm not sure it's worth it, but getting the style parameters alone usually takes 10%-15% of rendering time.

GorgonMeducer, 2022-09-12:
> 1. block a task: for me it means the task spins in a loop and eats up the CPU. I think yielding a task is more common to say the task lets other run.
>2. drawing-task: not an OS task, but a todo task.

Yes, we share the same picture and I will use yielding instead of blocking a task in the future discussion.

> Why PEs x2 is the the ideal FIFO size? Isn't PE x 100 better?  If we see more commands in the list an advanced policy can take better decisions. (It needs more memory for sure.)

If our view is limited to a pipeline, then buffering as many commands as possible is a way to mitigate the "insufficient information" problem. But we have sufficient information and the information is well organized, called the element tree. 

Rather than developing and command-stream-agnostic algorithm, focusing on the element tree is much easier and more efficient. 

Based on this, PEs x2 is the ideal FIFO size it is essentially a double-buffer structure for each PE. 

> optionally we can separate the generation of drawing tasks and SW rendering to 2 CPU cores. 
Yes, that's a typical frontend-backend separation and you can call it SW-GPU.

kisvegabor, 2022-09-15:
Alright, all sounds good. 

Based on your [comment](https://github.com/lvgl/lvgl/issues/3643#issuecomment-1241783083) we can start with:
> 0. define the descriptor of drawing-task
>    - define the data structure of drawing-task, lv_draw_task_t
>    - use lv_draw_task_t to describe existing low-level draw operations
>    - extract the key [...]

I'm a little bit busy now, and I'd like to finalize other half ready features first (improve image handling and color formats + new driver architecture). 

Are you open to get started with it?

GorgonMeducer, 2022-09-15:
@kisvegabor How about starting a dedicated issue for this part first:

> 0. define the descriptor of drawing-task
    * define the data structure of drawing-task, lv_draw_task_t
    * use lv_draw_task_t to describe existing low-level draw operations
    * extract the key attributes (requirements) from those drawing tasks and generate a list of capabilities
    * design a capability-fast-matching scheme

In fact, there are some details to discuss.

kisvegabor, 2022-09-19:
> define the data structure of drawing-task, lv_draw_task_t

Okay, let's discuss this first. 

E.g. [lv_draw_rect_dsc_t](https://github.com/lvgl/lvgl/blob/d89e732634ee60073254aa4140dd92f5513f338f/src/draw/lv_draw_rect.h#L32-L69) can be a  good basis of the discussion. Probably we need to pack it into an other struct where the are to draw is saved too.

nicusorcitu, 2022-09-19:
I think the draw_task shall represent only a part of rectangle object: background, background img, border, outline, shadow.

For example, VG-Lite GPU can draw everything from the rectangle box except the shadow. It would be a pity to not use the GPU at all on drawing rectangles just because it does not have shadow. And not all rectangles are being used with shadow. Most of them are simple.

kisvegabor, 2022-09-20:
> I think the draw_task shall represent only a part of rectangle object: background, background img, border, outline, shadow.

Makes sense. Finally would be 3 "rectangle parts":
- "body": filled (rounded) rectangle with or without gradient
- border: used as border and outline too
- shadow 
- (bg image: it's a simple image)

Our designer keeps telling me that border and outline should also support gradient. Is it something you could manage in VG-Lite?

GorgonMeducer, 2022-09-20:
@kisvegabor 

> * "body": filled (rounded) rectangle with or without gradient
* border: used as border and outline too
* shadow
* (bg image: it's a simple image)

I think this is still too high level. Before we go even further, would you please start a dedicated issue for this discussion?

kisvegabor, 2022-09-21:
Sure, please open a new topic with your suggested approach.

kisvegabor, 2022-10-06:
Shall we keep this open?

GorgonMeducer, 2022-10-06:
@kisvegabor does not matter.

closed 2022-10-23 by kisvegabor