lvgl/lvgl #3645: gridnav_set_focused crashes on unix+SDL+micropython(memory access error)

daviel, 2022-09-01:
### Perform all steps below and tick them with [x]
- [X] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [X] Update lvgl to the latest version
- [-] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
gridnav_set_focused crashes on unix+SDL+micropython with memory access error:

"Memory Access Error (Dump Written)"

I can reproduce the issue on my main machine (x86) and a Raspberry Pi.

### To Reproduce

```python
import lvgl as lv
import usys as sys
lv.init()

import SDL
SDL.init(w=320,h=240)

# Register SDL display driver.
draw_buf = lv.disp_draw_buf_t()
buf1_1 = bytearray(480*10)
draw_buf.init(buf1_1, None, len(buf1_1)//4)
disp_drv = lv.disp_drv_t()
disp_drv.init()
disp_drv.draw_buf = draw_buf
disp_drv.flush_cb = SDL.monitor_flush
disp_drv.hor_res = 320
disp_drv.ver_res = 240
disp_drv.screen_transp = 1
disp_drv.register()

indev_drv1 = lv.indev_drv_t()
indev_drv1.init()
indev_drv1.type = lv.INDEV_TYPE.KEYPAD
indev_drv1.read_cb = SDL.keyboard_read
indev1 = indev_drv1.register()

scr = lv.obj()
lv.scr_load(scr)

container = lv.obj(lv.scr_act())
container.set_flex_flow(lv.FLEX_FLOW.ROW_WRAP)

button = lv.btn(container)
label = lv.label(button)
label.set_text("Test1")

button2 = lv.btn(container)
label2 = lv.label(button2)
label2.set_text("Test2")

group = lv.group_create()
group.add_obj(container)
indev1.set_group(group)

lv.gridnav_add(container, lv.GRIDNAV_CTRL.NONE)
# this line causes the crash - remove it and it works without problems
lv.gridnav_set_focused(container, button2, lv.ANIM.OFF)
```

### Expected behavior
In the simulator the button can be focused and this is what I would also expect using unix.

### Screenshots or video
-

I can't registrate in the forums. It says my IP address (German) is not allowed to. Github login is also not working... Therefore I am creating the issue here.

daviel, 2022-09-05:
I found the issue: You should add the gridnav to the container before you start adding childs to it. Then it works. But is this the expected behavior? In the simulator the order of gridnav creation does not lead to a crash.

kisvegabor, 2022-09-07:
Based on the comments in #3646 and some testing I came to the same conclusion. I should have checked this issue first :slightly_smiling_face: 

So the girdnav container's `LV_EVENT_CHILD_CREATED` event sets the `focused_obj`.  https://github.com/lvgl/lvgl/blob/7802d2006db193532e0712b48d2a6b533e051a1e/src/others/gridnav/lv_gridnav.c#L228-L239
If you call `gridnav_add` first, it adds the `LV_EVENT_CHILD_CREATED` event to the container and its first child will be the focused object. 

There is no problem if you call `gridnav_set_focused` before `gridnav_add` because `gridnav_set_focused` handles if it's being used on a non-gridnav object:
https://github.com/lvgl/lvgl/blob/7802d2006db193532e0712b48d2a6b533e051a1e/src/others/gridnav/lv_gridnav.c#L84-L91

The only killer combo is calling `gridnav_add` at the end (`focused_obj` won't be set automatically) and call `gridnav_set_focused` (`cont` is already a gridnav but has no `focused_obj`). 

Fortunately, the fix was very simple :relaxed:

commit f52fb6f4 (2022-09-07): fix(gridnav): fix crash with specific order of commands

fixes #3645
closed 2022-09-07 by kisvegabor