lvgl/lvgl #3646: gridnav can't open dropdown list

daviel, 2022-09-01:
### Perform all steps below and tick them with [x]
- [X] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [X] Update lvgl to the latest version
- [X] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

When adding a dropdown widget to an container with gridnav you can focus the dropdown but you can't open it to select from the values of the dropdown list. It is working when using just a group without the gridnav (by tabbing).

### To Reproduce

```python
##### startup script #####

#!/opt/bin/lv_micropython -i

import lvgl as lv
import SDL
import display_driver

indev_drv1 = lv.indev_drv_t()
indev_drv1.init()
indev_drv1.type = lv.INDEV_TYPE.KEYPAD
indev_drv1.read_cb = SDL.keyboard_read
indev1 = indev_drv1.register()

scr = lv.obj()
lv.scr_load(scr)

container = lv.obj(lv.scr_act())
container.set_flex_flow(lv.FLEX_FLOW.ROW_WRAP)
container.set_size(320, 240)

button = lv.btn(container)
label = lv.label(button)
label.set_text("Test1")

button2 = lv.btn(container)
label2 = lv.label(button2)
label2.set_text("Test2")

primarycolordd = lv.dropdown(container)
primarycolordd.set_options("\n".join(["White","Black"]))
primarycolordd.set_height(40)

group = lv.group_create()
group.add_obj(container)
indev1.set_group(group)
lv.gridnav_add(container, lv.GRIDNAV_CTRL.NONE)
lv.gridnav_set_focused(container, button, lv.ANIM.OFF)

```

### Expected behavior

I would expect that focusing the dropdown menu and pressing enter would open it to choose from its values.

### Screenshots or video

kisvegabor, 2022-09-04:
I've tried it on in C with ASAN enabled and didn't find any error. Maybe it's an issue with the  Micropython binding.

@amirgon, could you take a look? There are no callbacks for `lv_grdidnav` so we can eliminate a large number of potential issues.

kisvegabor, 2022-09-04:
What can you imagine as expected behavior? Let's say you can open the dropdown with ENTER. After that when you press the arrows to select an item, the gridnav will focus on a new widget. 

So the gridnav should somehow know whether an object should absorb the arrows or not?

amirgon, 2022-09-04:
@kisvegabor Tried this now and it crashes with null pointer exception.  
Because `gridnav_add` sets `focused_obj` to `NULL` here:

https://github.com/lvgl/lvgl/blob/7802d2006db193532e0712b48d2a6b533e051a1e/src/others/gridnav/lv_gridnav.c#L69

and `dsc->focused_obj` is still `NULL` when `lv_gridnav_set_focused` is called, so `lv_obj_clear_state` crashes:

https://github.com/lvgl/lvgl/blob/7802d2006db193532e0712b48d2a6b533e051a1e/src/others/gridnav/lv_gridnav.c#L98

Actually I don't understand how you run it without hitting a SIGSEGV.

daviel, 2022-09-05:
> What can you imagine as expected behavior? Let's say you can open the dropdown with ENTER. After that when you press the arrows to select an item, the gridnav will focus on a new widget.
> 
> So the gridnav should somehow know whether an object should absorb the arrows or not?

Yes I would expect exactly this. This is also how it works when you are using only a group and tabbing through the widgets as soon as you reach the dropdown you can open the menu and select an item.

daviel, 2022-09-05:
> @kisvegabor Tried this now and it crashes with null pointer exception. Because `gridnav_add` sets `focused_obj` to `NULL` here: [...]

Sorry for this. The focus was a copy-paste mistake from the problem described here: #3645. There it also crashes.

kisvegabor, 2022-09-07:
The difference is that with groups TAB is used to change focus and arrows are used to set the values. With gridnav arrows are used to change focus but we expect them to be used to change values too. 

ENTER could be used to change mode and we already group widgets by editable (has settable value) or non-editable:
https://github.com/lvgl/lvgl/blob/f52fb6f47dae111ca1ad627243d4828efe476ada/src/widgets/dropdown/lv_dropdown.c#L69

So we could put the widgets into editing state and forward the arrows to the widget instead of processing them in the gridnav. 

The only issue is that the editing state is known from the widget's group:
https://github.com/lvgl/lvgl/blob/f52fb6f47dae111ca1ad627243d4828efe476ada/src/widgets/dropdown/lv_dropdown.c#L662-L663

daviel, 2022-09-12:
I also found that this issue exists for the roller widget also. Maybe implementing a standard solution for all widgets would be optional here.

kisvegabor, 2022-09-13:
I can implement it, just wanted describe my idea to see if there are an objections.

daviel, 2022-09-28:
> I can implement it, just wanted describe my idea to see if there are an objections.
> 
> Probably it will be required to create a dummy group in the gridnav to handle the editing state.

Any idea how long it will take you to implement this? Can I support you in any way?

kisvegabor, 2022-10-15:
Hi,

Sorry for that it took so long.  I've looked into it and found that it's not worth to update gridnav at this moment as `groups` and keypad navigation in general will be updated in v9. So it'd be a very short term update.

Instead I create a patch for you which is modify the gridnav to handle drop-downs correctly:
```diff
diff --git a/examples/others/gridnav/lv_example_gridnav_1.c b/examples/others/gridnav/lv_example_gridnav_1.c
index 8270e0a3c..ce7ebdbb8 100644
--- a/examples/others/gridnav/lv_example_gridnav_1.c
+++ b/examples/others/gridnav/lv_example_gridnav_1.c
@@ -23,6 +23,9 @@ void lv_example_gridnav_1(void)
     lv_obj_t * label = lv_label_create(cont1);
     lv_label_set_text_fmt(label, "No rollover");
 
+    lv_obj_t *dd = lv_dropdown_create(cont1);
+    lv_group_remove_obj(dd);   /*Not needed, we use the gridnav instead*/
+
     uint32_t i;
     for(i = 0; i < 10; i++) {
         lv_obj_t * obj = lv_btn_create(cont1);
diff --git a/src/others/gridnav/lv_gridnav.c b/src/others/gridnav/lv_gridnav.c
index 37cf82363..02a0b61bc 100644
--- a/src/others/gridnav/lv_gridnav.c
+++ b/src/others/gridnav/lv_gridnav.c
@@ -12,6 +12,7 @@
 #include "../../misc/lv_assert.h"
 #include "../../misc/lv_math.h"
 #include "../../core/lv_indev.h"
+#include "../../widgets/dropdown/lv_dropdown.h"
 
 /*********************
  *      DEFINES
@@ -123,6 +124,11 @@ static void gridnav_event_cb(lv_event_t * e)
         if(dsc->focused_obj == NULL) return;
 
         uint32_t key = lv_event_get_key(e);
+        if(lv_obj_get_class(dsc->focused_obj) == &lv_dropdown_class && lv_dropdown_is_open(dsc->focused_obj)) {
+            lv_event_send(dsc->focused_obj, LV_EVENT_KEY, &key);
+            return;
+        }
+
         lv_obj_t * guess = NULL;
 
         if(key == LV_KEY_RIGHT) {
@@ -259,7 +265,9 @@ static void gridnav_event_cb(lv_event_t * e)
             /*Forward press/release related event too*/
             lv_indev_type_t t = lv_indev_get_type(lv_indev_get_act());
             if(t == LV_INDEV_TYPE_ENCODER || t == LV_INDEV_TYPE_KEYPAD) {
-                lv_event_send(dsc->focused_obj, code, lv_indev_get_act());
+                if(lv_obj_get_class(dsc->focused_obj) != &lv_dropdown_class) {
+                    lv_event_send(dsc->focused_obj, code, lv_indev_get_act());
+                }
             }
         }
     }
```

daviel, 2022-10-16:
Thank you for the patch. Can you give me some insights regarding v9? Will it solve my problem? Then it would be okay to wait till the release of v9. :)

kisvegabor, 2022-10-16:
> Will it solve my problem? 

I don't know yet. We got some feedback about `group`s, keypad's key press handling and focusing in general requires some rework, but I haven't started the planning yet. 

Anyway, this issue is an other thing that we should consider. The limiting factor here is that the object should know if it's in editing state from it's own data instead of from its group. This way objects could be controlled manually by gridnav as if they were in a group. 

I added a not about it in the roadmap to be sure it won't be forgotten: https://github.com/lvgl/lvgl/commit/bcf5d79cb4f868270d9f564a92a080e07280b67b

kisvegabor, 2024-09-05:
Fixing this seems quite complex and it's not on the [roadmap of v9.3](https://github.com/lvgl/lvgl/issues/6763). Closing it for now. 

If this issue comes up again we can add it to the roadmap of a new release.

closed 2024-09-05 by kisvegabor