lvgl/lvgl #3650: Oblique dashed lines are not dashed

StefanBabatie, 2022-09-02:
### Perform all steps below and tick them with [x]
- [ x ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ x ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x ] Update lvgl to the latest version
- [ x ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
When drawing oblique dashed lines with the CPU (no acceleration), the resulting lines will not be dashed, but simple lines. This happens on the up to date master branch, both on the board itself and in the simulator.

### To Reproduce
Create a shape consisting of multiple dashed, connected, oblique lines, and a horizontal dashed line below them. Both have the same style applied, which contains the dash properties, but only the horizontal line is dashed in the final result.

```c
	/*Create an array for the points of the lines*/
	static lv_point_t line1_points[] = { {5, 5}, {70, 70}, {120, 10}, {180, 60}, {240, 10} };
	static lv_point_t line2_points[] = { {5, 100}, {240, 100}};

	/*Create style for the lines*/
	static lv_style_t style_line;
	lv_style_init(&style_line);
	lv_style_set_line_width(&style_line, 10);
	lv_style_set_line_color(&style_line, lv_palette_main(LV_PALETTE_BLUE));
	lv_style_set_line_dash_width(&style_line, 20);
	lv_style_set_line_dash_gap(&style_line, 20);

	/*Create first line and apply the style*/
	lv_obj_t * line1;
	line1 = lv_line_create(lv_scr_act());
	lv_line_set_points(line1, line1_points, 5);     /*Set the points*/
	lv_obj_add_style(line1, &style_line, 0);
	lv_obj_center(line1);

	/*Create second line and apply the style*/
	lv_obj_t * line2;
	line2 = lv_line_create(lv_scr_act());
	lv_line_set_points(line2, line2_points, 2);     /*Set the points*/
	lv_obj_add_style(line2, &style_line, 0);
	lv_obj_center(line2);
```

### Expected behavior
Each of the four oblique lines should also be dashed in the same way in which the horizontal line beneath them is, as they have the exact same style applied to them as well.

### Screenshots or video
The expected result would be something along the lines of the next image: 
[image]
Meanwhile, the actual result is the following:
[image]

kisvegabor, 2022-09-05:
True, it's a limitation of the SW renderer. 

Theoretically it could be solved rendering multiple smaller skew lines but as line drawing has only 1 px resolution  the dashes wouldn't be evenly inlined. 

Does VGLite support dashed lines?

StefanBabatie, 2022-09-05:
Yes, VGLite does support them.

kisvegabor, 2022-09-05:
Unfortunately, I have no good idea at this moment resolve it. 

We can add a warning to SW line draw if dashed skew lines should be drawn.

Similarly to masks (as mentioned [here](https://github.com/lvgl/lvgl/issues/3651#issuecomment-1236611882)) I'm planning to improve line drawing to make it more precise, more versatile and faster. My idea is treating skew lines as rotated rectangles. The original rectangles can be easily dashed, round capped, etc.  All we need is a super efficient transformation algorithm which is easier said than done :) Fortunately rotating a one-color rectangle is a quite special case so there is some room for optimization.

kisvegabor, 2023-08-16:
Will be fixed by https://github.com/lvgl/lvgl/issues/4388
Let continue there.

closed 2023-08-16 by kisvegabor