lvgl/lvgl #3652: Dashed round cap lines are poorly drawn

StefanBabatie, 2022-09-02:
### Perform all steps below and tick them with [x]
- [ x ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ x ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x ] Update lvgl to the latest version
- [ x ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Drawing a line with the dashed style and round cap style at the same time results in a normally dashed (square capped) line with rounded beginning and ending points, instead of a line in which each dash is rounded. 
Aside from this, the circles appended at the ends of the dashed line don't seem to align with the dashes themselves.
This happens on the up to date master branch and has been reproduced both in the simulator and on a board.

### To Reproduce
Create a thick, blue, round capped and dashed line:
```c
  /*Create an array for the points of the lines*/
  static lv_point_t line_points[] = { {5, 100}, {240, 100} };

  /*Create style for the lines*/
  static lv_style_t style_line;
  lv_style_init(&style_line);
  lv_style_set_line_width(&style_line, 20);
  lv_style_set_line_color(&style_line, lv_palette_main(LV_PALETTE_BLUE));
  lv_style_set_line_rounded(&style_line, true);
  lv_style_set_line_dash_width(&style_line, 35);
  lv_style_set_line_dash_gap(&style_line, 40);

  /*Create line and apply the style*/
  lv_obj_t* line;
  line = lv_line_create(lv_scr_act());
  lv_line_set_points(line, line_points, 2);     /*Set the points*/
  lv_obj_add_style(line, &style_line, 0);
  lv_obj_center(line);
```

### Expected behavior
The rounded ends should apply to each dash individually, resulting in a dash element resembling a pill rather than a simple rectangle. Currently, only the first and last dashes are rounded, but only on one end and only when the circles happen to align with the dashes, which is not always the case.

### Screenshots or video
Using the exact same code as the one presented, but removing the rounded cap, a square capped dashed line is obtained. 
[image]

This should give an ideea on how the round cap version would look: just round the ends of each of the three dashes. Based on this, the expected result should be something along the lines of the next image:
[image]

Meanwhile, the actual result is the following:
[image]

kisvegabor, 2022-09-05:
Thanks for pointing out these issues.  VGLite's output can be a good reference for SW drawing too.

We can resolve it with the other related issues in v9 (#3651 and #3650)

kisvegabor, 2024-01-14:
v9 has a generic vector API and we have built in SW vector engine called ThorVG. It should be used for non trivial cases.

closed 2024-01-14 by kisvegabor

anaGrad, 2024-01-15:
I ran a dashed line test in v9 and it is currently not working anymore. The dashed line is drawn as a normal line:
[image]
Here is the code I ran:
```c
static void line_dashed_cb(void)
{
    lv_style_reset(&style_common);
    lv_style_set_line_opa(&style_common, scene_with_opa ? LV_OPA_20 : LV_OPA_COVER);
    lv_style_set_line_width(&style_common, LINE_WIDTH);
    lv_style_set_line_color(&style_common, lv_palette_main(LV_PALETTE_BLUE));
    lv_style_set_line_dash_width(&style_common, LINE_DASH);
    lv_style_set_line_dash_gap(&style_common, LINE_GAP);
    line_create(&style_common);
}

static void line_create(lv_style_t * style)
{
    lv_obj_t * obj = lv_line_create(scene_bg);
    static lv_point_precise_t line_points[] = { {0, 0}, {OBJ_WIDTH, OBJ_HEIGHT}};
    lv_obj_remove_style_all(obj);
    lv_obj_add_style(obj, style, 0);
    lv_obj_set_style_line_color(obj, lv_palette_main(LV_PALETTE_ORANGE), 0);
    lv_obj_set_size(obj, OBJ_WIDTH, OBJ_HEIGHT);
    lv_line_set_points(obj, line_points, 2);
    lv_obj_center(obj);
}
```

kisvegabor, 2024-01-15:
It was never working with SW rendered and skew lines.