lvgl/lvgl #3657: [buildroot] Need help to create LVGL package

legerborea, 2022-09-04:
### Introduce the problem
Hello everyone,
I'm currently trying to port LVGL to buildroot but I'm stuck, I'm not very experienced with buildroot packing, furthermore, very few are using kconfig build system. And, for packages which currently used KConfig build system... they are the big one : busybox, linux, zephyr, uclibc, etc... So I don't have any **simple** example to help me.

This issues is related to  [lv_port_linux_frame_buffer - Issue 17 - Packing for buildroot](https://github.com/lvgl/lv_port_linux_frame_buffer/issues/17). 

### Examples and cases
Current examples are : 
- Yocto meta-package : https://cgit.openembedded.org/meta-openembedded/tree/meta-oe/recipes-graphics/lvgl/lvgl_8.1.0.bb
- Busybox repository : https://github.com/mirror/busybox
- SWUpdate repository : https://github.com/sbabic/swupdate

### Suggested solution
So far, I have proper `Config.in`, `lvgl.config` (which is default config file generated with `kconfig-qconf` utility) and `lvgl.hash`.
Main troubles are in `lvgl.mk`.  
I'm joining my current patch to this issue (at `txt` format because github doesn't support `patch`.

When trying to run `lvgl-rebuild` or even `lvgl-menuconfig` in my buildroot directory, I got those errors : 
```shell
>>> lvgl 8.3.1 Patching
Using /home/<userpath>/bsp-output/build/lvgl-8.3.1/.config as base
#
# merged configuration written to /home/<userpath>/bsp-output/build/lvgl-8.3.1/.config (needs make)
#
make[4]: *** No rule to make target « oldconfig ». Stop.
make[3]: *** [package/lvgl/lvgl.mk:18 : /home/<userpath>/bsp-output/build/lvgl-8.3.1/.stamp_dotconfig] Error 2
make[2]: *** [Makefile:84 : _all] Error 2
```

If anyone have more experience with buildroot packages and can help me to add support for LVGL, that will be appreciated !
Thank you for your time
[0001-package-lvgl-new-package.patch.txt](https://github.com/lvgl/lvgl/files/9485008/0001-package-lvgl-new-package.patch.txt)

legerborea, 2022-09-19:
I'm commenting this issue to keep it open, I didn't have time to continue the support but I will, ASAP...

legerborea, 2022-10-08:
Hello to all,

So I tried with CMake build system but I'm still struggling, especially with `lv_conf.h`file (new patch can be found attached).
I'm using option `-DLV_CONF_PATH` to set custom LVGL config file path but I'm still getting those errors:
```shell
src/core/../lv_conf_internal.h:46:17: note: « #pragma message: Possible failure to include lv_conf.h, please read the comment in this file if you get errors »
   46 |         #pragma message("Possible failure to include lv_conf.h, please read the comment in this file if you get errors")
      |                 ^~~~~~~

```

I don't understand why specified file can't be found, my CMake output is correct:
```shell
/home/<user>/Documents/buildroot/bsp-output/host/bin/cmake /home/<user>/Documents/buildroot/bsp-output/build/lvgl-8.3.1/ -DCMAKE_TOOLCHAIN_FILE="/home/<user>/Documents/buildroot/bsp-output/host/share/buildroot/toolchainfile.cmake" -DCMAKE_INSTALL_PREFIX="/usr" -DCMAKE_COLOR_MAKEFILE=OFF -DBUILD_DOC=OFF -DBUILD_DOCS=OFF -DBUILD_EXAMPLE=OFF -DBUILD_EXAMPLES=OFF -DBUILD_TEST=OFF -DBUILD_TESTS=OFF -DBUILD_TESTING=OFF -DBUILD_SHARED_LIBS=ON  -DLV_CONF_PATH="/home/<user>/Documents/buildroot/bsp-external/lvgl/lvgl_conf.h"
```

Does anyone have any idea on this behaviour ? Did I missed a CMake option ? I don't see any in [custom.cmake](https://github.com/lvgl/lvgl/blob/v8.3.3/env_support/cmake/custom.cmake) file.

Bonus question : Can I disable example build ? CMake file doesn't seem to have such option.
[0001-package-lvgl-new-package.patch.txt](https://github.com/lvgl/lvgl/files/9739626/0001-package-lvgl-new-package.patch.txt)

kisvegabor, 2022-10-10:
```
-DLV_CONF_PATH="/home/<user>/Documents/buildroot/bsp-external/lvgl/lvgl_conf.h"
```

It's not `lvgl_conf.h` but `lv_conf.h`.

legerborea, 2022-10-11:
> It's not `lvgl_conf.h` but `lv_conf.h`.

Oh ! Thank you for your response, I though than since we can add our custom path for lvgl configuration, filename were not important... This fix the behaviour !
I push a pull request two days ago (https://github.com/lvgl/lvgl/pull/3733), since we found the issue, not sure if this is pertinent.

kisvegabor, 2022-10-12:
Happy to hear that it solved the issue!

> (I keep the issue open for now just in case, once buildroot patch will be submit, I'm gonna close it)

Is there anything we can do on LVGL's side?

legerborea, 2022-10-12:
> Is there anything we can do on LVGL's side?

Currently, I think I have all I need with submitted pull request (https://github.com/lvgl/lvgl/pull/3733 and https://github.com/lvgl/lvgl/pull/3738), I had some trouble to understand at first how LVGL configuration was included, but now this is good to go 😄

For this repository, I'am able to built it with buildroot, now I have to include [LVGL drivers](https://github.com/lvgl/lv_drivers) but I saw than configuration file is required too without option to use an external one, I think I'm gonna submit a pull request with same behaviour than this repository to use external config.

kisvegabor, 2022-10-13:
> Currently, I think I have all I need with submitted pull request (https://github.com/lvgl/lvgl/pull/3733 and https://github.com/lvgl/lvgl/pull/3738)

Great, both are merged now. 

> now I have to include [LVGL drivers](https://github.com/lvgl/lv_drivers)

Please wait a little bit with it. I'm almost ready with a [new driver architecture](https://github.com/lvgl/lvgl/tree/arch/drivers). With this all the drivers will be hosted in LVGL.

legerborea, 2022-10-13:
> Great, both are merged now.

Perfect, thank you !

> Please wait a little bit with it. I'm almost ready with a [new driver architecture](https://github.com/lvgl/lvgl/tree/arch/drivers). With this all the drivers will be hosted in LVGL.

Oh, that's great news ! I didn't know than driver restructuration was in pending, I'm gonna take a look at it.
Is this restructuration planned for v9.x ?

closed 2022-11-05 by lvgl-bot

yangkkokk, 2024-06-29:
> 由于这里有一段时间没有活动，我们关闭了这个问题。但别担心，对话仍然在这里，你可以随时回到它。
> 
> 所以
> ，如果你对这个话题有意见或想法，请随时发表评论。

where is patch