lvgl/lvgl #3659: Compilation warning on lv_example_file_explorer_1.c

uraich, 2022-09-04:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Compilation warning on lv_example_file_explorer_1.c prevent compilation of lv_micropython
### To Reproduce

in line 11 and 50 I had to replace:
char * cur_path =  lv_file_explorer_get_current_path(obj);
by
const char * cur_path =  lv_file_explorer_get_current_path(obj);
### Expected behavior

### Screenshots or video

kisvegabor, 2022-09-05:
Hi,

Line 50 seems unrelated:
https://github.com/lvgl/lvgl/blob/master/examples/others/file_explorer/lv_example_file_explorer_1.c#L50

But I added `const` in all three file explorer examples. I wonder there was no CI error.

uraich, 2022-09-05:
You are right about line 50. This was certainly in one of the other examples.
I have another problem however:
I only get 
[User]	(21.180, +0)	 show_dir: Open dir error 3! 	(in lv_file_explorer.c line #547)
I am using a Linux system and there the library paths are wrong:
e.g. /home/videos should be /home/user:name/Videos

kisvegabor, 2022-09-07:
Am I correct to assume that it's being handled in #3671?

uraich, 2022-09-08:
yes
My intention for this PR was to add the MP examples for file_explorer. However, since the C examples did not work on my Ubuntu system I modified them and I decided to include my modifications into this PR

kisvegabor, 2022-09-23:
@uraich 
As all MP related PRs are merged can we close this issue?

closed 2022-09-23 by uraich