lvgl/lvgl #3661: lot of ime needed to read an image from flash memory

rmantovani68, 2022-09-05:
Hello, I am using lvgl on an NXP RT1062 , this sbc has 1MB RAM and 16MB Flash memory
The Flash is used to store all the images (rasterized).
The speed of the flash (empirically calculated) is about 6MB/S
The screen dimensions are 320X240
This means that a screen wide image is taking 320 * 240 * 2 / 6  - 24ms - to be read and put in the framebuffer memory to be displayed
The problem is that all the images contained in the screen are read from flash memory and put in the framebuffer memory **every** refresh cycle.
Our refresh cycle is 50ms and this means that with 2 or more big images the system is taking more than 50ms (90ms) to refresh the screen.
The only thing that I can think to avoid the need of reading the images from flash is that lvgl should check if there's the need to redraw the image or not.

kisvegabor, 2022-09-07:
> The speed of the flash (empirically calculated) is about 6MB/S

It seems a little bit slow for an 500 MHz MCU. Is it an SPI flash?

> This means that a screen wide image is taking 320 * 240 * 2 / 6 - 24ms - to be read and put in the framebuffer memory to be displayed

Assuming that the frame buffer is in internal RAM.

> Our refresh cycle is 50ms and this means that with 2 or more big images the system is taking more than 50ms (90ms) to refresh the screen.

Why do you have 2 screen sized images on one screen? One has opacity?

> The only thing that I can think to avoid the need of reading the images from flash is that lvgl should check if there's the need to redraw the image or not.

LVGL redraws only what has been changed on the screen, except if you have enabled `disp_drv.full_refresh`.

rmantovani68, 2022-09-09:
> > The speed of the flash (empirically calculated) is about 6MB/S
> 
> It seems a little bit slow for an 500 MHz MCU. Is it an SPI flash?
Yes 
> 
> > This means that a screen wide image is taking 320 * 240 * 2 / 6 - 24ms - to be read and put in the framebuffer memory to be displayed
> 
> Assuming that the frame buffer is in internal RAM.
Yes
> 
> > Our refresh cycle is 50ms and this means that with 2 or more big images the system is taking more than 50ms (90ms) to refresh the screen.
> 
> Why do you have 2 screen sized images on one screen? One has opacity?
> 
Because the images are overlapped i.e. the underlying image is 320x240 and there is another image on top of it of 220x220 and another one of 180x180 , the images can have or not transparency
> > The only thing that I can think to avoid the need of reading the images from flash is that lvgl should check if there's the need to redraw the image or not.
> 
> LVGL redraws only what has been changed on the screen, except if you have enabled `disp_drv.full_refresh`.
disp_drv.full_refresh is enabled

kisvegabor, 2022-09-09:
Just be sure I understand the issue correctly. The question is that why LVGL reads the whole 320x240 image once most of its area is covered by a 220x220 image, right?

rmantovani68, 2022-09-12:
> Just to be sure I understand the issue correctly. The question is that why LVGL reads the whole 320x240 image once most of its area is covered by a 220x220 image, right?  

Yes, but if we are using transparency this is necessary. 
The idea would be to not load images if the area has no differences from the previous screen...

kisvegabor, 2022-09-14:
> The idea would be to not load images if the area has no differences from the previous screen...

LVGL redraws only the changes areas. But if the 320x240 image changes and the 220x220 doesn't, LVGL needs to get both:
1. the 320x240 because it has been changed
2. the 220x220 to draw above the 320x240 one

closed 2022-10-07 by lvgl-bot