lvgl/lvgl #3664: Lightweight TTF LVGL Integration

codewitch-honey-crisis, 2022-09-06:
### Integrating my lightweight TTF/OTF orchestration into LVGL

I wrote [htcw_gfx](https://honeythecodewitch.com/gfx) in part because LVGL's TTF support via Freetype didn't fit my needs - it had to be compiled on a linux machine (at least at the time i looked into it) and it basically needed an ESP32 or better to run realistically, with restrictions on the size of the font file size. All of this is me spitballing from memory so forgive me if I got some of the details incorrect regarding Freetype.

I supported truetype by adapting this public domain code: https://github.com/nothings/stb/blob/master/stb_truetype.h and changing it around so it could stream from storage instead of requiring the font to be loaded into RAM. It also assumed gobs of memory was available in other ways, like in its internal custom heap, so I modified that as well.

I'd like to integrate it into LVGL, and I started poking around on the forums which led me here:

https://docs.lvgl.io/master/overview/font.html#add-a-new-font-engine

The trouble is, I don't quite understand it. I can't produce traditional glyph bitmaps because TTF requires glyphs to be able to overlap and to be antialiased. Is the glyph bitmap for basically an alpha bitmap? Meaning, is it "grayscale" but goes between transparent and solid? If so, I can work with that.

### Examples and cases

My library renders glyphs via a callback method so it doesn't have to allocate unnecessary memory. I can adapt it to render to a glyph pretty easily, but it demands an alpha bitmap as above

### Suggested solution

I need help here. I need some clarification on the above points. I'm reaching out here because Gabor Kiss-Vamosi said I should open an issue here regarding this. I didn't do it right away because I wanted to do some legwork first so I didn't fire this off half cocked.

kisvegabor, 2022-09-07:
> Does LVGL include any kind of stream interface I should be using? What is the best practice in a cross platform manner to connect my engine to a TTF file or program flash?

We have a [file system interface](https://docs.lvgl.io/master/overview/file-system.html)  with optional modules for some exiting file APIs. There is no flash interface in it but I can easily imagine a simple one. 

> The other thing is, I have a question about memory allocation. A lot of the LVGL code returns pointers, which is troubling to me. For starters it means whipping the heap, and also it means what about PSRAM? what about DMA capable ram that must be specially allocated? Is there a facility for custom allocators in LVGL, and a memory allocation scheme i should be using that goes with it?

We are considering to implement a global cache/buffer pool from which capped memory can be malloc-ed. It's a little bit complicated and still there a lot of questions. However, as an alternative a simple platform specific `lv_malloc_caps(size, caps)` also could be useful. By default it could be mapped to the normal `lv_malloc`.

codewitch-honey-crisis, 2022-09-07:
Cool. I'll look into using lv_malloc instead of malloc at least. I don't know that platform specific is the best route here. This library works on a ton of platforms as long as they're fast enough, with the SAMD21 being the "weakest" device I've tried it on. The ESP32 struggles at larger font sizes, but memory use should be good for a lot of platforms.

If you were doing font scaling how would you go about it? Spin off another lv_font_t for a font at each different scale? That's what I'm thinking, but it would be nice to have input from someone more familiar.

If it was my lib, I'd have made you pass in the pointer to an lv_font_t and filled it in the routine for each different scale, but when in rome as they say. Either way, does generating a new lv_font_t instance for the font at each scale make the most sense to you or can you think of something more appropriate?

Thanks!

kisvegabor, 2022-09-07:
> I don't know that platform specific is the best route here. 

If we want to let people allocating into DMA-able, PSRAM, flash, etc we can't avoid being platform dependent. But on the top level it's only a `#define lv_malloc_caps my_malloc`. And the user can malloc whatever he/she wants and fallback to `lv_malloc` for other cases. 

> If you were doing font scaling how would you go about it? Spin off another lv_font_t for a font at each different scale? That's what I'm thinking, but it would be nice to have input from someone more familiar.

We have no good solution for it at this moment. A few ideas:
1. zoom the label, which zooms the rastered text, so it won't have a really good quality 
2- keep changing the font of the label as you described, but it's far from optimal
3. if it's a FreeType (or other vector font) the label should have a dedicated font descriptor (`lv_font_t`) and it should be reinitialized with a new size.

> If it was my lib, I'd have made you pass in the pointer to an lv_font_t and filled it in the routine for each different scale, but when in rome as they say. 

Romans would be happy to hear the opinion of people form different cultures :slightly_smiling_face: 

> Either way, does generating a new lv_font_t instance for the font at each scale make the most sense to you or can you think of something more appropriate?

It's doable, but seems very wasteful and cumbersome. On web (HTML/CSS) there is no such a thing as a "font descriptor". Or it seems like all elements have its own descriptor which can be configured freely. It's similar to my 3rd point above.

codewitch-honey-crisis, 2022-09-07:
> If we want to let people allocating into DMA-able, PSRAM, flash, etc we can't avoid being platform dependent. But on the top level it's only a #define lv_malloc_caps my_malloc. And the user can malloc whatever he/she wants and fallback to lv_malloc for other cases.

That's good to know. What I did in htcw_gfx was I took allocator and deallocator arguments that pointed to functions - defaulting to malloc and free, and I would use those to do allocations. It worked because that way you could create your own platform specific stuff and plug in your own allocators. So like with the ESP32 under Arduino you could pass in ps_malloc instead of defaulting to malloc. I'm not saying you should change LVGL, but just food for thought.

> if it's a FreeType (or other vector font) the label should have a dedicated font descriptor (lv_font_t) and it should be reinitialized with a new size.

I'm not entirely clear on this, but it gave me an idea (and might be what you're saying?)

When I fill in the callbacks and all that I suppose I could use user_data to hold the scale, and then provide a custom routine to change it. Hadn't occurred to me until now, but maybe that makes the most sense. Do you see that causing problems? Like, what happens if I assign a font to a label, and then change the size? My assumption would be on next render the text would be resized, but if there's caching or any funny business then no dice. What do you think?

kisvegabor, 2022-09-07:
> I'm not entirely clear on this, but it gave me an idea (and might be what you're saying?)

Let's say you have `lv_font_t font1` and you used `font1` eith 10 label widgets. If you want to scale e.g. `label3` you have to scale `font1` which will scale all 10 labels. So if you want to scale only one label it should have a dedicated font. I.e.. `label1-font1`, `label2-font2`, `label3-font3` etc.

> I suppose I could use user_data to hold the scale

We could add a `scale` field to `lv_font_t`, not problem with that. But do we really want to scale (e.g. make 30% larger) or set a new size (e.g. make it 28pt high)?

codewitch-honey-crisis, 2022-09-07:
I have an internal scale value that is the percentage of the original font size I think? (the code came from the STB stuff I linked to initially), but it has a function to convert the line height to a scale, so if you want a font that's 50px high, you give it a float of 50 (yeah they can be fractional because of anti-aliasing), and it computes the actual scale.

codewitch-honey-crisis, 2022-09-08:
Ouch! the lv_font_glyph_dsc_t.adv_width member is unsigned, but my TTF engine can report a negative advance in some cases due to kerning on certain fonts. As weird as it is, it has bit me before.

How would I compensate for that? Can I set ofs_x? I'm not 100% sure if that's copasetic because I don't know exactly what it does, only what I *think* it does.

Any insight would be appreciated!

codewitch-honey-crisis, 2022-09-08:
In addition to my previous ask, I've run into the strangest thing.
[https://github.com/codewitch-honey-crisis/ttgo_lvgl](https://github.com/codewitch-honey-crisis/ttgo_lvgl)

I set the font for a label, it's calling my get_glyph_dsc callback, but not my render callback, and the font remains default.

Please see main.c and let me know what I'm doing wrong? if it's not wrong there, lib/lv_ttf/lv_ttf.c contains the relevant code, at the end of the file.

Thanks again!

codewitch-honey-crisis, 2022-09-09:
I'm still having a problem with lvgl calling my glyph dsc callback but not my render glyph callback

I'm reporting each of the outputs here. I've included them below. 

Forgive the long post. I kept the code as close to the demo as possible. I don't think this font renders commas, hence the one weird dsc but it shouldn't stop the code from rendering. 

Here's the code that should render the font. Keep in mind, the cbfs_register creates a "filesystem" over the Ubuntu_data array that's embedded as a header. I've tested this fs emulator and I'm completely confident that it works.

```cpp
cbfs_register(0,'Q',Ubuntu_data,sizeof(Ubuntu_data));
lv_fs_file_t file;
lv_fs_open(&file,"Q:/",LV_FS_MODE_RD);
lv_style_t style;
lv_font_t font;
lv_ttf_t ttf;
lv_ttf_open(&ttf,&file);
// create a font with a 50 line_height
lv_ttf_create_font(&font,&ttf,50,NULL);
lv_style_init(&style);
lv_style_set_text_font(&style, &font);  /*Set a larger font*/
/* Create simple label */
lv_obj_t *label = lv_label_create( lv_scr_act() );
lv_style_selector_t selector=0;

lv_obj_set_style_text_font(label,&font,selector);

lv_label_set_text( label, LVGL_Arduino.c_str() );
lv_obj_align( label, LV_ALIGN_CENTER, 0, 0 );
```

What the heck am I doing wrong here?

The output of my glyph desc callback is below: again sorry for the length.
```
ch: T, k: 0, adv_w: 25, box_w: 24, box_h: 32, ofs_x: 1, ofs_y: 1
ch: e, k: 0, adv_w: 25, box_w: 24, box_h: 27, ofs_x: 1, ofs_y: 2
ch: x, k: 0, adv_w: 25, box_w: 25, box_h: 25, ofs_x: 1, ofs_y: 1
ch: t, k: 0, adv_w: 25, box_w: 24, box_h: 33, ofs_x: 1, ofs_y: 2
ch: H, k: 0, adv_w: 25, box_w: 22, box_h: 32, ofs_x: 2, ofs_y: 1
ch: e, k: 0, adv_w: 25, box_w: 24, box_h: 27, ofs_x: 1, ofs_y: 2
ch: l, k: 0, adv_w: 25, box_w: 24, box_h: 37, ofs_x: 1, ofs_y: 2
ch: l, k: 0, adv_w: 25, box_w: 24, box_h: 37, ofs_x: 1, ofs_y: 2
ch: o, k: 0, adv_w: 25, box_w: 24, box_h: 27, ofs_x: 1, ofs_y: 2
ch:  , k: 0, adv_w: 25, box_w: 1, box_h: 1, ofs_x: 0, ofs_y: 1
ch: A, k: 0, adv_w: 25, box_w: 26, box_h: 32, ofs_x: 0, ofs_y: 1
ch: r, k: 0, adv_w: 25, box_w: 20, box_h: 25, ofs_x: 4, ofs_y: 1
ch: d, k: 0, adv_w: 25, box_w: 22, box_h: 37, ofs_x: 1, ofs_y: 2
ch: u, k: 0, adv_w: 25, box_w: 21, box_h: 26, ofs_x: 2, ofs_y: 2
ch: i, k: 0, adv_w: 25, box_w: 24, box_h: 37, ofs_x: 1, ofs_y: 2
ch: n, k: 0, adv_w: 25, box_w: 21, box_h: 25, ofs_x: 3, ofs_y: 1
ch: o, k: 0, adv_w: 25, box_w: 24, box_h: 27, ofs_x: 1, ofs_y: 2
ch: !, k: 0, adv_w: 25, box_w: 10, box_h: 33, ofs_x: 8, ofs_y: 2
ch:  , k: 0, adv_w: 25, box_w: 1, box_h: 1, ofs_x: 0, ofs_y: 1
ch: V, k: 0, adv_w: 25, box_w: 26, box_h: 32, ofs_x: 0, ofs_y: 1
ch: 8, k: 0, adv_w: 25, box_w: 24, box_h: 34, ofs_x: 1, ofs_y: 2
ch: ., k: 0, adv_w: 25, box_w: 10, box_h: 11, ofs_x: 8, ofs_y: 2
ch: 3, k: 0, adv_w: 25, box_w: 22, box_h: 34, ofs_x: 2, ofs_y: 2
ch: ., k: 0, adv_w: 25, box_w: 10, box_h: 11, ofs_x: 8, ofs_y: 2
ch: 1, k: 0, adv_w: 25, box_w: 22, box_h: 32, ofs_x: 2, ofs_y: 1
ch: H, k: 0, adv_w: 25, box_w: 22, box_h: 32, ofs_x: 2, ofs_y: 1
ch: e, k: 0, adv_w: 25, box_w: 24, box_h: 27, ofs_x: 1, ofs_y: 2
ch: l, k: 0, adv_w: 25, box_w: 24, box_h: 37, ofs_x: 1, ofs_y: 2
ch: l, k: 0, adv_w: 25, box_w: 24, box_h: 37, ofs_x: 1, ofs_y: 2
ch: o, k: 0, adv_w: 25, box_w: 24, box_h: 27, ofs_x: 1, ofs_y: 2
ch:  , k: 0, adv_w: 25, box_w: 1, box_h: 1, ofs_x: 0, ofs_y: 1
ch: A, k: 0, adv_w: 25, box_w: 26, box_h: 32, ofs_x: 0, ofs_y: 1
ch: r, k: 0, adv_w: 25, box_w: 20, box_h: 25, ofs_x: 4, ofs_y: 1
ch: d, k: 0, adv_w: 25, box_w: 22, box_h: 37, ofs_x: 1, ofs_y: 2
ch: u, k: 0, adv_w: 25, box_w: 21, box_h: 26, ofs_x: 2, ofs_y: 2
ch: i, k: 0, adv_w: 25, box_w: 24, box_h: 37, ofs_x: 1, ofs_y: 2
[... 28 lines cut ...]
ch: 3, k: 0, adv_w: 25, box_w: 22, box_h: 34, ofs_x: 2, ofs_y: 2
ch: ., k: 0, adv_w: 25, box_w: 10, box_h: 11, ofs_x: 8, ofs_y: 2
ch: 1, k: 0, adv_w: 25, box_w: 22, box_h: 32, ofs_x: 2, ofs_y: 1
ch: H, k: 0, adv_w: 25, box_w: 22, box_h: 32, ofs_x: 2, ofs_y: 1
ch: e, k: 0, adv_w: 25, box_w: 24, box_h: 27, ofs_x: 1, ofs_y: 2
ch: l, k: 0, adv_w: 25, box_w: 24, box_h: 37, ofs_x: 1, ofs_y: 2
ch: l, k: 0, adv_w: 25, box_w: 24, box_h: 37, ofs_x: 1, ofs_y: 2
ch: o, k: 0, adv_w: 25, box_w: 24, box_h: 27, ofs_x: 1, ofs_y: 2
ch:  , k: 0, adv_w: 25, box_w: 1, box_h: 1, ofs_x: 0, ofs_y: 1
ch: A, k: 0, adv_w: 25, box_w: 26, box_h: 32, ofs_x: 0, ofs_y: 1
ch: r, k: 0, adv_w: 25, box_w: 20, box_h: 25, ofs_x: 4, ofs_y: 1
ch: d, k: 0, adv_w: 25, box_w: 22, box_h: 37, ofs_x: 1, ofs_y: 2
ch: u, k: 0, adv_w: 25, box_w: 21, box_h: 26, ofs_x: 2, ofs_y: 2
ch: i, k: 0, adv_w: 25, box_w: 24, box_h: 37, ofs_x: 1, ofs_y: 2
ch: n, k: 0, adv_w: 25, box_w: 21, box_h: 25, ofs_x: 3, ofs_y: 1
ch: o, k: 0, adv_w: 25, box_w: 24, box_h: 27, ofs_x: 1, ofs_y: 2
ch: !, k: 0, adv_w: 25, box_w: 10, box_h: 33, ofs_x: 8, ofs_y: 2
ch:  , k: 0, adv_w: 25, box_w: 1, box_h: 1, ofs_x: 0, ofs_y: 1
ch: V, k: 0, adv_w: 25, box_w: 26, box_h: 32, ofs_x: 0, ofs_y: 1
ch: 8, k: 0, adv_w: 25, box_w: 24, box_h: 34, ofs_x: 1, ofs_y: 2
ch: ., k: 0, adv_w: 25, box_w: 10, box_h: 11, ofs_x: 8, ofs_y: 2
ch: 3, k: 0, adv_w: 25, box_w: 22, box_h: 34, ofs_x: 2, ofs_y: 2
ch: ., k: 0, adv_w: 25, box_w: 10, box_h: 11, ofs_x: 8, ofs_y: 2
ch: 1, k: 0, adv_w: 25, box_w: 22, box_h: 32, ofs_x: 2, ofs_y: 1
```

codewitch-honey-crisis, 2022-09-09:
Almost!
[image]

Sorry it's a bit blurry. I couldn't get my camera to focus on the text at that angle. Anyway, as you can see it's rendering, but the font isn't lined up along the descent line. I'm working on that, and as soon as I have it figured out I'll have a deliverable here: [https://github.com/codewitch-honey-crisis/ttgo_lvgl](https://github.com/codewitch-honey-crisis/ttgo_lvgl)

Then we can begin to integrate fully.

kisvegabor, 2022-09-09:
You are progressing rapidly. :slightly_smiling_face: 

To add something to it I debugged the y offset issue.

You need to modify these:
```c
dsc_out->ofs_y = -y2;
```
```c
out_font->base_line = line_height - (lv_coord_t)(fi->ascent*scale);
```

Resulting in
[image]

from:
```c
    lv_label_set_text( label, "I'm very happy with this!\nCongrats @codewitch-honey-crisis :) ^^ ~" );
```

The only issue I see now is `\n` should have a special treatment. We solved like this in the built in font engine in  `lv_font_get_glyph_dsc`:
```c
    if(letter < 0x20 ||
       letter == 0xf8ff || /*LV_SYMBOL_DUMMY*/
       letter == 0x200c) { /*ZERO WIDTH NON-JOINER*/
        dsc_out->box_w = 0;
        dsc_out->adv_w = 0;
    }
```

codewitch-honey-crisis, 2022-09-09:
Thank you! I'm still stuck on that issue, but I'll poke around some more. =)

kisvegabor, 2022-09-10:
Cool :sunglasses: 

For the sake of simplicity we don't use submodules in LVGL. 3rd party libs are either should be added separately to the project, or added directly into LVGL.

As your TTF lib is only one C file and two H files (is it correct?) we can copy it right into LVGL into the [libs folder](https://github.com/lvgl/lvgl/tree/master/src/libs).

BTW, were you able to convert it C? I see only stb_truetype_htcw.h, stb_rect_pack.h and lv_ttf.c.

codewitch-honey-crisis, 2022-09-10:
I think it's C, but I'm so used to C++ but I can't be certain I nailed it. I did my best to remove any C++isms, and I'm pretty sure it's C now, in part because the original library I adapted and modified was C.

One thing that's funny, is one of the headers - the htcw one - since it's a "private" header only used by a C file, I stick it in with the source rather than the includes when I put it in my projects. Do what you like, but keep in mind that header is "private"

kisvegabor, 2022-09-10:
Okay, we will see if there are any C++/C issues once it's integrated into LVGL. 

Are you interested in sending a PR?

codewitch-honey-crisis, 2022-09-10:
Sure. I'll go ahead and do that now.

codewitch-honey-crisis, 2022-09-10:
Which branch do I submit to? I'm making a separate project with just the lib, and an example underneath.

The issue is the const buffer filesystem for reading from headers. Should I PR that too?

kisvegabor, 2022-09-10:
> Which branch do I submit to?

`master` is fine.

> The issue is the const buffer filesystem for reading from headers. Should I PR that too?

Yes please include the const buffer file system support too. It should be added to [src/libs/fsdrv](https://github.com/lvgl/lvgl/tree/master/src/libs/fsdrv).

codewitch-honey-crisis, 2022-09-10:
Okay. I'm going to clean up the const buffer fs to make the api a little cleaner and more consistent.

Then I'll post

kisvegabor, 2022-09-10:
> I may need to brainstorm to figure out a convenience method for instantiating it. I've got engineering tunnel vision at the moment.

What's the problem?

codewitch-honey-crisis, 2022-09-10:
Well, for starters in lv_fsdrv.h you have init functions that take no parameters. Doing so with CBFS is impossible since it needs an array and a drive letter to give it. So you must init manually, I think. (unless you have any ideas?) 

```cpp
const uint8_t* data = Ubuntu_data;
const size_t data_size = sizeof(Ubuntu_data);
lv_fs_cbfs_register(0,'Q',data,data_size);
lv_fs_open(&file,"Q:/",LV_FS_MODE_RD);
lv_ttf_open(&ttf,&file);
// create a font with a line height of 40
lv_ttf_create_font(&lfont,&ttf,40,NULL);
```
It was noted that this code is very counterintuitive. I guess part of that is unavoidable, because making a filesystem over a header is weird. I really wish y'all would have made a stream API and used that rather than requiring an entire FS to stream, but what's done is done. So now I need to figure out a way to make that sequence of code simpler and more intuitive**. Like, this is terrible because there are only N slots for a CBFS drive (currently 3 via a #define), and to register one every time you create a font is not great. So I'm at something of a loss.

**Like being able to take a const buffer array as a parameter when creating the font

codewitch-honey-crisis, 2022-09-10:
I've done the necessary source updates but I don't have permission to create a pull request in master

kisvegabor, 2022-09-10:
My idea about whole cbfs was something like:
```c
lv_cbfs_init(); //Called once

...

lv_cbfs_add_file("filename", buffer, buffer_size); //save this 3 parameters in a list

...

lv_ttf_open(&ttf, "Q:/filename"); //Have a file descriptor in ttf
```

Even better would be to have something like:
```c
char fn[64];
lv_cbfs_get_filename(fn, sizeof(fn), drv_letter, buf, buf_size); //E.g. fn = Q:0x12345678-32768 (dv_letter:address-size)
lv_fs_open(&f, fn, LV_FS_MODE_RD);
```
This way to user don't need to create cbfs files manually, free them when they are not used anymore, or avoid name collisions. 

I wouldn't worry too much about memory fragmentation for 2 reasons:
1. LVGL already uses a lot of malloc/free and it works
2. As there are a lot of  malloc/free the holes are filled

codewitch-honey-crisis, 2022-09-10:
I'll change cbfs around. I also still need to be able to put a pull request into master, or I need alternative branch to use where I have permission.

embeddedt, 2022-09-10:
Anyone on GitHub should have permission to open a PR to `master`. Are you going to https://github.com/lvgl/lvgl/pulls?

[image]

codewitch-honey-crisis, 2022-09-10:
Yeah when I do, the create button is grayed out.

codewitch-honey-crisis, 2022-09-10:
Adding, how do I assign a new drive letter?

In lv_fs_cbfs_init() i need to create an lv_fs_drv_t instance, so I malloc it (unfortunately, gosh i hate all these mallocs)

I go to fill it, and it wants a drive.

I don't have a drive letter, and I can't take a drive letter parameter from the init function.

Your move.

codewitch-honey-crisis, 2022-09-10:
Sorry, I've done a pull request once or twice, a long time ago.

My screen looks different than yours. It says "base:" instead of "base repository:" and lvgl/lvgl isn't in my list.
[image]

codewitch-honey-crisis, 2022-09-10:
Is there any way LVGL can traverse its list of active lv_fs_drv_t pointers and give me a free drive letter?

codewitch-honey-crisis, 2022-09-10:
I submitted a pull request, with caveats (listed in the checklist for the pull request)

Particularly the TTF code is not sanitized, and the code reformatter linked to in the pull request template is bad.

codewitch-honey-crisis, 2022-09-10:
The pull request failed the build, and I'm not sure why. I've only partially replicated your build environment so that might be the problem. I could use some help here. And it's failing on code formatting, but where is the code formatting script? The link given in the template is dead like I said.

embeddedt, 2022-09-10:
Which link? The template just says to run `scripts/code-format.py`; it should be in your local copy. The link in that checklist entry works for me.

[image]

codewitch-honey-crisis, 2022-09-10:
The link to astyle is what's broken.

embeddedt, 2022-09-10:
I think that is on your end. I can browse to this without issues. http://astyle.sourceforge.net/install.html

codewitch-honey-crisis, 2022-09-10:
weird. anyway, I googled and it came up. I'll work on the formatting. I need to figure out the build error too.

embeddedt, 2022-09-10:
The current set of errors can be fixed by running `scripts/lv_conf_internal_gen.py` to regenerate `lv_conf_internal.h` with a default value for `LV_USE_FS_CBFS`.

codewitch-honey-crisis, 2022-09-10:
I'm about ready to pack it in. The script does nothing. Therefore, I copied lv_conf_template.h and created lv_conf.h from it. I've put it in the root of the project, and one directory above it.

Still no dice.

[build] ../src/misc/../lv_conf_internal.h:46:17: note: '#pragma message: Possible failure to include lv_conf.h, please read the comment in this file if you get errors'
[build]    46 |         #pragma message("Possible failure to include lv_conf.h, please read the comment in this file if you get errors")
[build]       |                 ^~~~~~~
[build] [107/430  20% :: 10.000] Building C object CMakeFiles/lvgl.dir/src/font/lv_font_montserrat_46.c.obj
[build] In file included from ../src/misc/lv_log.h:16,
[build]                  from ../lvgl.h:25,
[build]                  from ../src/font/lv_font_montserrat_46.c:8:
[build] ../src/misc/../lv_conf_internal.h:46:17: note: '#pragma message: Possible failure to include lv_conf.h, please read the comment in this file if you get errors'

Only over 300 times.

htcw_gfx builds without requiring a ton of prereqs. This config is way too complicated /rant

embeddedt, 2022-09-10:
> The script does nothing. 

This indicates something is wrong with your build setup, as the script should silently regenerate `lv_conf_internal.h`.

> Therefore, I copied lv_conf_template.h and created lv_conf.h from it. I've put it in the root of the project, and one directory above it.

This should not be necessary to get the tests to build, if that's what you are trying to build.

codewitch-honey-crisis, 2022-09-10:
tests isn't an option in the cmake provided.

i've lvgl. I've lvgl_demos. I've got lvgl_examples

Nothing builds.

what do I do? Do I have to install make into my build environment and run "make tests"?

Is there a readme somewhere? FFS this is crazy/

embeddedt, 2022-09-10:
```
cd tests
./main.py build
```

Be sure you have the necessary dependencies from [here](https://github.com/lvgl/lvgl/blob/master/scripts/install-prerequisites.sh).

codewitch-honey-crisis, 2022-09-10:
Ruby? really?

I guess I'm installing linux then.

Ruby? Seriously? *headdesk*

embeddedt, 2022-09-10:
It's because our test framework uses Ruby.

codewitch-honey-crisis, 2022-09-10:
I see that. I wish I didn't see that, but I see that.

embeddedt, 2022-09-10:
Do you want me to take a stab at the CI issues and push the fixes to your branch?

codewitch-honey-crisis, 2022-09-10:
If you think you can, that would be awesome, if only because I don't know how long it will take me to get a build env setup for this project.

And thanks. Sorry I'm frustrated. I've been banging on this all day. It actually took me less time to get the initial TTF engine into LVGL than it is for me to push a repo.

codewitch-honey-crisis, 2022-09-10:
I'm leaving this up to you. After linux and the depencies your scripts still fail

/usr/bin/env: ‘python3\r’: No such file or directory

embeddedt, 2022-09-10:
Something is off about your system; that looks like some type of line-ending issue. In any case, I am just about done fixing the errors.

codewitch-honey-crisis, 2022-09-10:
I'm using WSL under windows. I can build other things.

embeddedt, 2022-09-10:
Admittedly neither @kisvegabor nor I use WSL as far as I know, so it's possible that there is an issue with the CI scripts there.

Is the last condition here a typo?

https://github.com/codewitch-honey-crisis/lvgl/blob/cd2dbed9f2f56121c1a60fcac5c6d4d665e93b0a/src/libs/ttf/lv_ttf.c#L4482

codewitch-honey-crisis, 2022-09-10:
Indeed it is. I don't know it got in there. It's usually how I check floats for NAN since isnan() isn't always defined.

codewitch-honey-crisis, 2022-09-10:
You're amazing! Thank you so much!

kisvegabor, 2022-09-25:
The discussion continued in https://github.com/lvgl/lvgl/pull/3680

closed 2022-09-25 by kisvegabor