lvgl/lvgl #3665: Local alpha for objects

Makskuk, 2022-09-06:
### Introduce the problem
Hello! I`m developing UI for video camera and I need to draw menu on top of video scene. 

On lvgl v5.3 I have drawn partial-transparent UI with function **lv_obj_set_style_local_bg_opa()**.
But in version 8.3 there is no this function and **lv_obj_set_style_bg_opa()** work in other way: 
it apply opacity but with blcak background of object instead of video.

Can I draw objects with loca alpha on v8.3?

### Examples and cases
```
// init display 
/*Transparent background*/
    lv_disp_set_bg_opa(disp, LV_OPA_TRANSP);
    static lv_style_t style;
    lv_style_init(&style);
    lv_style_set_bg_opa(&style, LV_OPA_TRANSP);
    lv_obj_remove_style_all(lv_scr_act()); /*Remove the styles coming from the theme*/
    lv_obj_add_style(lv_scr_act(), &style, 0);

// draw object
    lv_obj_t *rect = lv_obj_create(lv_scr_act());
    lv_obj_set_size(rect, 200, 100);
    lv_obj_set_pos(rect, 200, 30);
    lv_obj_set_style_bg_color(rect, lv_palette_main(LV_PALETTE_RED), 0);
    lv_obj_set_style_bg_opa(rect, LV_OPA_40, 0);
```

### Suggested solution

Makskuk, 2022-09-07:
If in function **lv_draw_sw_blend_basic()** set **disp->driver->screen_transp = 1;** alpha channel works as expected. Else, after checking **disp->driver->screen_transp** renderer go to **fill_argb** for solid objects and to **fill_normal** for transparent objects.

I have no idea, why this flag differs for different objects and same display driver...

embeddedt, 2022-09-07:
If the display needs to have transparency (which seems to be true for your case) you should set `disp_drv.screen_transp = 1` at the time that you create and register the display.

Makskuk, 2022-09-07:
Thanks! It works :)

Docs not very clear about this. https://docs.lvgl.io/8.3/overview/display.html#transparent-screens not point to it. (only in Porting gide in optional fields)

closed 2022-09-07 by Makskuk