lvgl/lvgl #3667: [Error] (19234.671, +0)  _lv_inv_area: detected modifying dirty areas in render         (in lv_refr.c line #212)

anghd, 2022-09-06:
### Introduce the problem
Hi @kisvegabor 
I want to fresh the content of first column of table when draging scroll bar，however, 
there will report error " detected modifying dirty areas in render " when draging the scroll bar down and up.

1.                  static  lv_obj_t* setting_icon[TABLE_ROW_MAX] ={NULL};
2. 
3.                 if (setting_icon[row] == NULL)
4.                 {
5.                     setting_icon[row] = lv_img_create(obj);
6. 
7.                     lv_img_set_src(setting_icon[row], LV_SYMBOL_SETTINGS);
8.                 }
9. 
10.                 lv_obj_set_pos(setting_icon[row], dsc->draw_area->x1, dsc->draw_area->y1);

there'll be OK if I comment line 3 to 10.  I think this problem may result from static pointer array setting_icon.
But this event callback function can be called by different threads? the array setting_icon can only be called in this event cb function.
 
I'm using PC simulator and lvgl8.x

thank you in advance.

### Examples and cases
[image]

### Suggested solution

kisvegabor, 2022-09-08:
What's the problem here?

closed 2022-09-30 by lvgl-bot

codegrue, 2023-12-15:
What does this error mean. I get them all time also.

XuNeo, 2023-12-15:
> What does this error mean. I get them all time also.

It normally means widget's properties have been modified during rendering(such as DRAW_MAIN event callback). It will cause rendering issues thus are forbidden.

codegrue, 2023-12-16:
Using FreeRTOS where the UI is being altered in a different thread than the LVGL refresh loop, what is the proper pattern to prevent this kind of conflict?

XuNeo, 2023-12-16:
LVGL is not thread safe. You can only call LVGL APIs from one thread.

kisvegabor, 2023-12-16:
@XuNeo 
I'm thinking about something like this:
```c
lv_lock();  /*LVGL is not thread safe, so lock it before using any `lv_...` functions in any threads. */
lv_timer_handler(); /*/*Process timers, events, animations, display refreshed, and input devices*/*/
lv_unlock(); 
```
`lv_lock` and `lv_unlock` are using an `lv_mutex` internally if `LV_USE_OS` is set. Without OS they set/clear a bool variable for which we can ASSERT on some critical places. In case of `LV_OS_CUSTOM` the user can implement these functions.

This practice can be shown all the places so people will intuitively know that is locking is required.  

What do you think?

codegrue, 2023-12-16:
Take for example a screen that shows data from an api call. The main loop is calling lv_timer_handler() and another thread calls the API and then updates screen labels when the data cones back. It seems like it's not possible to have this on the same thread without blocking the UI during the API wait period.

XuNeo, 2023-12-17:
Adding a global lock to `lvgl` will work. I'm not sure if it's a best practice. 
Operation to `lvgl` normally should be done in signal thread and never block UI thread. Move block operations outside of `lvgl`'s thread like using `libuv` etc.

kisvegabor, 2023-12-18:
Once people have an OS, they will use it for sure and will attempt to adjust the widgets from an other thread as well.

kisvegabor, 2023-12-20:
`lv_lock` is not added yet, we are just talking about it. 

If you are calling LVGL function from multiple threads take look at [this](https://docs.lvgl.io/8.3/porting/os.html).

codegrue, 2023-12-21:
Thank you. Wanted to share for FreeRTOS the code looks like this:

```
SemaphoreHandle_t xLVGLMutexHandle;
TaskHandle_t tUIHandle;
TaskHandle_t tLVGLHandle;

xLVGLMutexHandle = xSemaphoreCreateMutex();
xTaskCreate(uiCallback, "tUIHandle", 3000, (void *)0, tskIDLE_PRIORITY, &tUIHandle);
xTaskCreate(lvglCallback, "tLVGLHandle", RTOS_LVGL_STACK_SIZE, (void *)0, RTOS_LVGL_PRIORITY, &tLVGLHandle);

void lvglCallback(void *p)
{
    while (1)
    {
        // use mutex to insure nothing writes to LVGL during the render
        if (xSemaphoreTake(xLVGLMutexHandle, (TickType_t)10) == pdTRUE)
        {
            lv_timer_handler();

            xSemaphoreGive(xLVGLMutexHandle);
        }

        vTaskDelay(5 / portTICK_PERIOD_MS);
    }
}

void uiCallback(void *p)
{
    while (1)
    {
        // use mutex to lock LVGL before updating UI
        if (xSemaphoreTake(xLVGLMutexHandle, (TickType_t)10) == pdTRUE)
        {
            /* update LVGL UI here */

            xSemaphoreGive(xLVGLMutexHandle);
        }

        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}
```