lvgl/lvgl #3673: Bordered polygons are poorly drawn

StefanBabatie, 2022-09-07:
### Perform all steps below and tick them with [x]
- [ x ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ x ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x ] Update lvgl to the latest version
- [ x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Polygons that have borders are not properly drawn. Because the drawing is based on masking a rectangle on which the polygon style is applied, the polygon rarely comes out properly styled. This causes a problem not only with borders, but any style property applied, such as gradients. Only the simplest fill-only polygons give correct results. 

### To Reproduce
```c
  static lv_color_t cbuf[LV_CANVAS_BUF_SIZE_TRUE_COLOR(384, 384)];
  const lv_point_t points[] = { {45, 55}, {295, 305}, {50,300} };
  lv_draw_rect_dsc_t rect_dsc;
  lv_draw_rect_dsc_init(&rect_dsc);

  rect_dsc.border_width = 8;
  rect_dsc.border_color = lv_color_hex(0xffffff);
  rect_dsc.border_opa = LV_OPA_80;
  rect_dsc.bg_color = lv_color_hex(0xff0000);
  rect_dsc.bg_opa = LV_OPA_MAX;

  lv_obj_t* canvas = lv_canvas_create(lv_scr_act());
  lv_canvas_set_buffer(canvas, cbuf, CANVAS_WIDTH, CANVAS_HEIGHT, LV_IMG_CF_TRUE_COLOR);
  lv_obj_center(canvas);
  lv_canvas_fill_bg(canvas, lv_color_hex(0x000000), LV_OPA_COVER);
  lv_canvas_draw_polygon(canvas, points, 3, &rect_dsc);
```

### Expected behavior
Instead of the resulting polygon being a cutout from a rectangle styled with the given polygon style, it should be a polygon styled with the given polygon style.

### Screenshots or video
The expected result would be something along the lines of the next picture, obtained with VGLite:
[image]

Meanwhile, the actual result is the following:
[image]

I do understand this is also a limitation of the current implementation of software renderer, that will be reworked in the future, like [https://github.com/lvgl/lvgl/issues/3652](url) , [https://github.com/lvgl/lvgl/issues/3651](url) and [https://github.com/lvgl/lvgl/issues/3650](url), but I reported it in hopes it would be useful to know about this one as well.

kisvegabor, 2022-09-07:
A lot of week points of the current SW renderer were came to light in the past few days. :sweat_smile: It's really an other one. :slightly_frowning_face:

kisvegabor, 2023-08-16:
Polygon drawing is removed in v9 and will be replaced by a generic vector  API. See https://github.com/lvgl/lvgl/issues/4388

Let continue there.

closed 2023-08-16 by kisvegabor