lvgl/lvgl #3676: Accelerating canvas drawing and adding a polygon widget

StefanBabatie, 2022-09-08:
### Introduce the problem
There are two things: first, the current version of LVGL always initializes the draw context of a canvas using the software version of the draw context initializer, the result being that all the drawing callbacks attached to canvases are software renderer versions, even if accelerators are present and enabled.  See the init_fake_disp function in lv_canvas.c:

```
...
LV_ASSERT_MALLOC(draw_ctx);  
if(draw_ctx == NULL)  return;
lv_draw_sw_init_ctx(drv, draw_ctx);  // Can be replaced with  disp->driver->draw_ctx_init(drv, draw_ctx);
disp->driver->draw_ctx = draw_ctx;
draw_ctx->clip_area = clip_area;
...
```

I understand that this is in place of a limitation of the SDL renderer, but maybe some checks can be implemented such that when this accelerator or the cpu are desired, the software drawing is used, and when not, the appropiate accelerator callbacks are used.

Second, drawing polygons within the current version can not be done outside of a canvas. Maybe it would be a good ideea to also add a polygon widget so that polygons can be drawn without the need of a canvas.

I've talked about these issues with the maintainer via e-mail but I opened a ticket so that we can get a discussion going and have more input.

### Examples and cases
As some accelerators do support polygon drawing and some users might desire this feature, it might be a good ideea to not only accelerate canvas polygon drawing, but also allow users to draw polygons outside of canvases, with different styles applied.

### Suggested solution
As the maintainer suggested, for the canvas acceleration, a possible  change is to add a configuration parameter (e.g. LV_ACCELERATE_CANVAS) that can control what draw methods are attached to canvases: software for CPU and cases like the SDL renderer or accelerated for the other cases.

For the polygon widget, maybe the structure of a possible polygon descriptor type can be discussed here. This descriptor can allow using multiple vector graphics features such as dashed lines, bezier curves, gradients and so on.

StefanBabatie, 2022-09-08:
A mention regarding VGLite and polygons: accelerating polygons with borders and outlines would be difficult and inefficient to implement. Because of this, polygon acceleration using vglite should work like the current rectangle acceleration works: use the CPU to draw elements like borders, outlines and shadows and only use VGLite to fill with solid color/gradients.

kisvegabor, 2022-09-10:
> As the maintainer suggested, for the canvas acceleration, a possible change is to add a configuration parameter (e.g. LV_ACCELERATE_CANVAS) that can control what draw methods are attached to canvases: software for CPU and cases like the SDL renderer or accelerated for the other cases.

I agree. I'd call it `LV_CANVAS_SW_RENDER` but it's not that important. 

> A mention regarding VGLite and polygons: accelerating polygons with borders and outlines would be difficult and inefficient to implement. 

Why? Isn't a border just a line using the points of the polygon? 

I'm thinking about completely removing the SW polygon rendering because it's too complex but still very limited. Using a 3rd party SVG library would be better instead. E.g. [nanosvg](https://github.com/memononen/nanosvg) or [thorvg](https://github.com/Samsung/thorvg). Maybe we could have an `lv_svg` widget instead of polygon, and configure it to render with an SVG lib or VGLite.

StefanBabatie, 2022-09-12:
> Why? Isn't a border just a line using the points of the polygon?

VGlite can, at most, within one call to the gpu, create a given outline (a stroke path) and fill the space contained by said outline with a solid color or a gradient. If both an outline and a border are needed, drawing two stroke paths is needed, and thus one of them needs to be made separately, in another call to the GPU. 

Moreover, because the outline is outside of the polygon itself, the points that make up its path can not be the same as the ones of the polygon path, but they need to look like a slightly magnified polygon compared to the initial one. This zoom would also need a pivot... These operations bring in additional overhead.

Stroke support for VGLite is pure software, and because of this, the performance of stroking is not very good. Especially because areas that are stroked are pretty small, I think that the CPU will be faster than VGLite in those cases and that's why I think keeping borders and outlines CPU only is a valid option.

From what we deduced with the help of the benchmark, I think avoiding strokes everywhere it's possible is a good ideea to maintain performance.

> I'm thinking about completely removing the SW polygon rendering because it's too complex but still very limited. Using a 3rd party SVG library would be better instead. E.g. [...]

I think this could work, looks like nanosvg is more of a lightweight thing so that one might be the winner.

StefanBabatie, 2022-09-12:
I've looked into accelerating all the style elements from the rectangle descriptor. It seems that the shadows can not be accelerated by vglite, it's possible maybe only when there is a sharp shadow and no gradual color change. The rest of the elements can be accelerated: the background and border can be drawn in a single vglite call, while the outline will need an additional call. The shadow should remain software drawn.
Doing this for rectangles is trivial as opposed to random convex polygons, because the path for the outline can easily be obtained from the four original rectangle points by adding some offsets. 
I'm currently working on implementing the acceleration of these additional elements.

kisvegabor, 2022-09-13:
Note that neither the SW render nor the current style system support outline and shadow on polygons.

I think it's not critical so for these very complex cases we can expect to use images.

StefanBabatie, 2023-02-22:
Hello! Any upate on this discussion? Are you planing on settling it in v9? Now that I've had a bit more time to spend with LVGL I see that non rectangle polygons are indeed uncommon. Maybe it would be best to leave them in the canvas, and just permit canvas acceleration as stated in the ticket, after all.

kisvegabor, 2023-02-23:
We have just merged an [important update](https://github.com/lvgl/lvgl/issues/4011) into master which blocked further improvements. I believe things will speed up a little bit now. 

The next large related thing is the [multi thread rendering](https://github.com/lvgl/lvgl/issues/3700) in which we will rework a the drawing related parts of LVGL. 

My current idea about polygon, and vector drawing in general is that LVGL should provide an API for these, but we should have a 3rd parts library for software rendering, which can be replaced by GPUs is needed.

To add it, we needed some (actually a lot) work on the multi-thread rendering part. It will take 2-3 month to have a stable ground on which we can start build new things.

kisvegabor, 2023-08-16:
Closing it as complete.

The canvas drawing is already updated in `master` (v9-dev). It uses the the very same draw functions as other widgets does, so the GPUs can be utilized too.

See the examples here: https://docs.lvgl.io/master/widgets/canvas.html

(I've just noticed that image and arc drawing is broken, but these are just minor issues not affecting the concept in general)

closed 2023-10-26 by kisvegabor