lvgl/lvgl #3685: Wrong behavior of overflow of lv_span

FASTSHIFT, 2022-09-13:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
When the font placeholder is turned off, and `lv_span` is used to display the mixed text of Chinese + emoji, displaying ellipses is not expected behavior.

### To Reproduce
in lv_conf.h:
`#define LV_USE_FONT_PLACEHOLDER 0`

```c
void span_test(void)
{
    lv_obj_t* spans = lv_spangroup_create(lv_scr_act());
    lv_spangroup_set_align(spans, LV_TEXT_ALIGN_LEFT);
    lv_spangroup_set_overflow(spans, LV_SPAN_OVERFLOW_ELLIPSIS);
    lv_spangroup_set_mode(spans, LV_SPAN_MODE_EXPAND);
    lv_obj_set_style_border_width(spans, 1, 0);
    lv_obj_set_style_border_color(spans, lv_color_black(), 0);
    lv_obj_set_style_text_font(spans, &lv_font_simsun_16_cjk, 0);

    lv_span_t* span = lv_spangroup_new_span(spans);
    const char txt[] = "a写\ua060"; // ‘\ua060’ is an emoji that does not support display
    lv_span_set_text(span, txt);
}

```

### Expected behavior
[image]

### Screenshots or video
[image]

FASTSHIFT, 2022-09-13:
It can be displayed normally when the text order is modified:
```c
lv_span_set_text(span, "写a\ua060");
```

[image]

I tried to compare the drawing process of the two texts and found that `_lv_txt_get_next_line()` in `lv_txt_get_snippet()` returned different values.

For text: "a写\ua060", `ofs = 4`
[image]

For text: "写a\ua060", `ofs = 7`
[image]

`ofs` will be passed to `last_snippet->bytes` here:
[image]

This leads to an error in the text overflow judgment of the function `lv_draw_span()`, and finally makes the span mistaken for a truncated text, and finally draws an ellipsis.

FASTSHIFT, 2022-09-13:
The reason why the return value of `_lv_txt_get_next_line()` is different is caused by word truncation.
For text: "a写\ua060"
'a' occupies 1 byte, '写' occupies 3 bytes, because Chinese is truncated text and the `max_width` passed in by span is exactly the sum of the pixel widths of 'a' and 'write' (\ua060' glyphs cannot be drawn glyph, pixel width is 0), and finally returns 4.

For text: "写a\ua060"
"a\ua060" is regarded as a completed word by `lv_txt`, the byte occupied by this word is `1+3=4`, plus 3 bytes of 'write', so return 7.

FASTSHIFT, 2022-09-13:
There is a problem now that `lv_span` will calculate the total width of the text and pass it to the `max_width` of `_lv_txt_get_next_line()`, but since the placeholder is closed, it will encounter the case where the glyph width is 0, and `max_width` cannot contain this information, resulting in problems with word matching.

kisvegabor, 2022-09-28:
https://github.com/lvgl/lvgl/pull/3687 is merged so I close it.

closed 2022-09-28 by kisvegabor