lvgl/lvgl #3686: Arduino Library Example does not compile

Ghostrider-1911, 2022-09-14:
### Perform all steps below and tick them with [x]
- [ x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x] Update lvgl to the latest version
- [ x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

I am not able to compile the Arduino Example, after following all Steps described here:
https://docs.lvgl.io/master/get-started/platforms/arduino.html#get-the-lvgl-arduino-library

Erros like lv_demo.h No such file or directory, maybe dependencies changed? 

### To Reproduce

Using LVGL_Arduino, with Arduino 1.8.19, esp32 2.0.4

### Expected behavior

Example should compile without errors 

### Screenshots or video

[image]

kisvegabor, 2022-09-15:
It's `lv_demos.h`. (plural).

Did we mistype it somewhere in the docs?

Ghostrider-1911, 2022-09-19:
> It's `lv_demos.h`. (plural).
> 
> Did we mistype it somewhere in the docs?

Thank you for the Answer, i changed #include <lv_demo.h> to #include <lv_demos.h> in the LVGL_Arduino.ino, i also changed #define LV_USE_DEMO_WIDGETS 0 to #define LV_USE_DEMO_WIDGETS 1. Now i have new Erros while trying to compile. 

> C:\Users\Ghost\Documents\Arduino\libraries\lvgl-master\src\core\lv_obj.c: In function 'lv_init':
> C:\Users\Ghost\Documents\Arduino\libraries\lvgl-master\src\core\lv_obj.c:156:18: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
>      char * txt = "Á";
>                   ^~~~
> C:\Users\Ghost\Documents\Arduino\libraries\lvgl-master\src\widgets\checkbox\lv_checkbox.c: In function 'lv_checkbox_constructor':
> C:\Users\Ghost\Documents\Arduino\libraries\lvgl-master\src\widgets\checkbox\lv_checkbox.c:125:13: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
>      cb->txt = "Check box";
>              ^
> C:\Users\Ghost\Documents\Arduino\libraries\lvgl-master\src\widgets\tabview\lv_tabview.c: In function 'lv_tabview_add_tab':
> C:\Users\Ghost\Documents\Arduino\libraries\lvgl-master\src\widgets\tabview\lv_tabview.c:88:25: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
>          new_map[tab_id] = "";
>                          ^
> C:\Users\Ghost\Documents\Arduino\libraries\lvgl-master\src\widgets\tabview\lv_tabview.c:97:24: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
>              new_map[1] = "";
>                         ^
> C:\Users\Ghost\Documents\Arduino\libraries\lvgl-master\src\widgets\tabview\lv_tabview.c:100:37: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
>              new_map[tab_id * 2 - 3] = "\n";
>                                      ^
> C:\Users\Ghost\Documents\Arduino\libraries\lvgl-master\src\widgets\tabview\lv_tabview.c:102:37: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
>              new_map[tab_id * 2 - 1] = "";
>                                      ^
> C:\Users\Ghost\Documents\Arduino\libraries\lvgl-master\src\widgets\tabview\lv_tabview.c: In function 'lv_tabview_constructor':
> C:\Users\Ghost\Documents\Arduino\libraries\lvgl-master\src\widgets\tabview\lv_tabview.c:227:21: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
>      tabview->map[0] = "";
>                      ^
> c:/users/ghost/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: libraries\lvgl-master\demos\benchmark\assets\objs.a(img_benchmark_cogwheel_rgb565a8.c.o):C:\Users\Ghost\Documents\Arduino\libraries\lvgl-master\src\demos\benchmark\assets/img_benchmark_cogwheel_rgb565a8.c:216: multiple definition of `img_benchmark_cogwheel_rgb565a8'; libraries\lvgl-master\benchmark\assets\objs.a(img_benchmark_cogwheel_rgb565a8.c.o):C:\Users\Ghost\Documents\Arduino\libraries\lvgl-master\src\benchmark\assets/img_benchmark_cogwheel_rgb565a8.c:216: first defined here
> c:/users/ghost/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: libraries\lvgl-master\demos\benchmark\assets\objs.a(img_benchmark_cogwheel_rgb565a8.c.o):C:\Users\Ghost\Documents\Arduino\libraries\lvgl-master\src\demos\benchmark\assets/img_benchmark_cogwheel_rgb565a8.c:11: multiple definition of `img_benchmark_cogwheel_rgb565a8_map'; libraries\lvgl-master\benchmark\assets\objs.a(img_benchmark_cogwheel_rgb565a8.c.o):C:\Users\Ghost\Documents\Arduino\libraries\lvgl-master\src\benchmark\assets/img_benchmark_cogwheel_rgb565a8.c:11: first defined here
> c:/users/ghost/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: libraries\lvgl-master\widgets\objs.a(lv_demo_widgets.c.o): in function `lv_demo_widgets':
> C:\Users\Ghost\Documents\Arduino\libraries\lvgl-master\src\widgets/lv_demo_widgets.c:98: multiple definition of `lv_demo_widgets'; libraries\lvgl-master\demos\widgets\objs.a(lv_demo_widgets.c.o):C:\Users\Ghost\Documents\Arduino\libraries\lvgl-master\src\demos\widgets/lv_demo_widgets.c:98: first defined here
> c:/users/ghost/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: libraries\lvgl-master\widgets\assets\objs.a(img_clothes.c.o):C:\Users\Ghost\Documents\Arduino\libraries\lvgl-master\src\widgets\assets/img_clothes.c:193: multiple definition of `img_clothes'; libraries\lvgl-master\demos\widgets\assets\objs.a(img_clothes.c.o):C:\Users\Ghost\Documents\Arduino\libraries\lvgl-master\src\demos\widgets\assets/img_clothes.c:193: first defined here
> c:/users/ghost/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: libraries\lvgl-master\widgets\assets\objs.a(img_clothes.c.o):C:\Users\Ghost\Documents\Arduino\libraries\lvgl-master\src\widgets\assets/img_clothes.c:13: multiple definition of `img_clothes_map'; libraries\lv [... cut ...]
> c:/users/ghost/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: DRAM segment data does not fit.
> c:/users/ghost/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `dram0_0_seg' overflowed by 34664 bytes
> collect2.exe: error: ld returned 1 exit status
> exit status 1
> Fehler beim Kompilieren für das Board ESP32 Dev Module.

embeddedt, 2022-09-19:
> I find it very strange why these obvious warnings was not spotted by our CI. @embeddedt do you have any ideas?

It seems we [compile with this warning disabled](https://github.com/lvgl/lvgl/blob/24893c37ae18618089eff8a9bf38f604ae23b96e/tests/CMakeLists.txt#L301).

kisvegabor, 2022-09-19:
True, I removed it here and fixed some related issues: d78e92725

Ghostrider-1911, 2022-09-19:
Thank you for quick Answers!
I have to apologize, i am an Beginner. 

Can you maybe change the #include <lv_demo.h> to #include <lv_demos.h> in the .ino?
I think other beginners would be glad ;) 

If i copy the files like you stated in the arduino.md, the Arduino IDE is not able to find lv_demos.h, so i copied all Files from demo and example directly to src. 

Maybe Arduino IDE is not able to find the .h File in Subfolders? I dont know.

kisvegabor, 2022-10-06:
I've updated the INO file. It should solve the mentioned issues.

> If i copy the files like you stated in the arduino.md, the Arduino IDE is not able to find lv_demos.h, so i copied all Files from demo and example directly to src.

[The docs](https://docs.lvgl.io/master/get-started/platforms/arduino.html#use-the-examples-and-demos) says:
> IMPORTANT Due to some the limitations of Arduino's build system you need to copy lvgl/examples to lvgl/src/examples. Similarly for the demos lvgl/demos to lvgl/src/demos.

So it also says the `src` folder.

commit 66a36b7b (2022-10-06): example(arduino): fix the example INO file.

fixes: #3686