lvgl/lvgl #3688: Draw line and circle issue on monochrome LCD 128x64

pnt325, 2022-09-15:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

I implement the button has border and draw line with thinness set to 1pixel but draw on screen always 2

### To Reproduce

```c
	static lv_style_t style_btn;

    lv_style_init(&style_btn);
    lv_style_set_radius(&style_btn, 5);
    lv_style_set_border_color(&style_btn, lv_color_black());
    lv_style_set_border_width(&style_btn, 1);

#if 1
    lv_obj_t * label1 = lv_label_create(lv_scr_act());
    //lv_label_set_long_mode(label1, LV_LABEL_LONG_WRAP);     /*Break the long lines*/
    // lv_obj_set_width(label1, 127);  /*Set smaller width to make the lines wrap*/
    //lv_obj_set_style_text_align(label1, LV_TEXT_ALIGN_CENTER, 0);
    lv_label_set_text(label1, "ABC abc xyz 123456 ght ijk");
    lv_obj_set_pos(label1, 0, 0);

    // Create button
    lv_obj_t* btn = lv_btn_create(lv_scr_act());
    lv_obj_set_pos(btn, 20, 20);
    lv_obj_set_size(btn, 50, 20);
    lv_obj_t* label = lv_label_create(btn);
    lv_label_set_text(label, "Button");
    lv_obj_center(label);
    lv_obj_add_style(btn, &style_btn, 0);
#endif

    static lv_point_t line_points[] = { {5, 5}, {15, 15}, {20, 20}, {40, 40}, {120, 0} };

    static lv_style_t style_line;

    lv_style_init(&style_line);
    lv_style_set_line_width(&style_line, 1);
    lv_style_set_line_color(&style_line, lv_color_black());
    lv_style_set_line_rounded(&style_line, true);

    /*Create a line and apply the new style*/
    lv_obj_t * line1;
    line1 = lv_line_create(lv_scr_act());
    lv_line_set_points(line1, line_points, 5);     /*Set the points*/
    lv_obj_add_style(line1, &style_line, 0);
    lv_obj_center(line1);
```

### Expected behavior

### Screenshots or video

[image]

kisvegabor, 2022-09-16:
Did you set `disp_drv.antialiasing = 0;`?

pnt325, 2022-09-16:
Thank fast response.

I try to add `disp_drv.antialiasing = 0;` int to init functon like bellow:

```c
	// lvgl
	static lv_color_t disp_buf1[GUI_HOR_RES*GUI_VER_RES/8];
	static lv_disp_draw_buf_t buf;

	lv_init();
	lv_disp_draw_buf_init(&buf, disp_buf1, NULL, GUI_HOR_RES*GUI_VER_RES);
	lv_disp_drv_init(&disp_drv);
	disp_drv.draw_buf = &buf;
	disp_drv.flush_cb = gui_flush_cb;
	disp_drv.rounder_cb = gui_rounder_cb;
	disp_drv.set_px_cb = gui_set_px_cb;
	disp_drv.hor_res = GUI_HOR_RES;
	disp_drv.ver_res = GUI_VER_RES;
	disp_drv.antialiasing = 0;
	lv_disp_drv_register(&disp_drv);

//	lv_theme_t* theme = lv_theme_mono_init(disp, false, &font_lcd);
//	lv_disp_set_theme(disp, theme);

	lv_example_label_1();
```

Result nothing change.

kisvegabor, 2022-09-19:
It looks like this for me with the mono theme and LV_COLOR_DEPTH = 1
[image]

pnt325, 2022-09-19:
So, your test also drawn double of pixel when setting width = 1.

kisvegabor, 2022-09-20:
It's 1px here. I added an 1x1px grid in GIMP:
[image]

pnt325, 2022-09-21:
Can you share me the code init mono theme and LVGL configure for mono theme. Because I have configure `LV_COLOR_DEPTH = 1` but when init mono theme LVGL stuck.

kisvegabor, 2022-09-22:
It's the same as yours:
```c
	lv_theme_t* theme = lv_theme_mono_init(disp, false, &some_font);
	lv_disp_set_theme(disp, theme);
```

What does "LVGL stuck" mean exactly?

pnt325, 2022-09-23:
`What does "LVGL stuck" mean exactly?`

That is reason I comment the theme init.
I try it and let you know result.

Thank you very much!

closed 2022-10-15 by lvgl-bot

pnt325, 2023-03-09:
> `What does "LVGL stuck" mean exactly?`
> 
> That is reason I comment the theme init. I try it and let you know result.
> 
> Thank you very much!

[image]

Here is fixed issue. 
It's look verify nice. 

Thank for your support