lvgl/lvgl #3689: In lv_api_map undefined reference to lv_timer_handler()

PaulskPt, 2022-09-15:
For an Espressif ESP32-S3-BOX, using Espressif repo 'esp-box', example image_display,
the ```idf.py flash monitor``` command results in the following error:

[image]

where as the call to ```lv_timer_handler```, in file ```lv_api_map.h```  is here:
[image]

The definition is in lv_timer.h:

[image]

kisvegabor, 2022-09-16:
It's very strange. Which LVGL version do you use?

PaulskPt, 2022-09-16:
Thank you for your reply!
I am using the lvgl which is a componentent in the Espressif esp-box repo. Below a screenshot of the lvgl/library.properties file.

[image]

PaulskPt, 2022-09-16:
Here a view of the location of lvgl component in the esp-box repo

[image]

kisvegabor, 2022-09-20:
`lv_timer_handler` is available in v8.1 for sure:
https://github.com/lvgl/lvgl/blob/release/v8.1/src/misc/lv_timer.c#L67

I can't see what could be the problem. :slightly_frowning_face: 

At worst case, just delete the problematic part from `lv_api_map.h`.

PaulskPt, 2022-09-28:
OK, thank you

closed 2022-09-28 by kisvegabor