lvgl/lvgl #3691: pitch number isn't fixed in the output of font convertor

GorgonMeducer, 2022-09-15:
### Introduce the problem

According to FreeType:

> Pitch:  its absolute value is the number of bytes per bitmap line
...
To speed up memory access, pitch is in most cases a multiple of 16bit, 32bit, or even 64bit. It often happens that the pitch is thus larger than the necessary bits (or bytes) for a bitmap or pixmap row; in such cases, unused bits (or bytes) are at the very right (i.e., the end) of a row.

When I study the output of the font convertor, I notice that the pitch is not a fixed value when the width is not an integer multiple of 2 (bpp==4) or an integer multiple of 4 (bpp==2). 

This arrangement makes the low-level acceleration hard. Can we at least introduce an option to use a fixed pitch (attach unused bits to the end of a row)?

### Examples and cases

### Suggested solution

kisvegabor, 2022-09-19:
To make HW acceleration simpler for v9 I'm planning to use only 8 bit bitmaps on the "draw level". The font's bitmap get function should handle converting the bpp < 8 bitmaps to bpp=8. 

What do you think about it?

GorgonMeducer, 2022-09-19:
@kisvegabor  Oh my god... I just finished adding A2 and A4 support to arm-2d... and Helium is capable of accelerating them.

I know some 2D GPUs support A2 and A4 also. So maybe you can remove the A3 support from LVGL, which is not acceleration friendly and is hard to see any A3 in other places except LVGL.

A8 is of cause the fastest. If you want to only support A8, I strongly support you!

Anyway, the non-fixed-pitch-number problem could be fixed first. 

Thank you.

kisvegabor, 2022-09-19:
A3 is not seen directly by LVGL. It used only if the the bitmaps are compressed and LVGL decompresses it to A4. 

> Anyway, the non-fixed-pitch-number problem could be fixed first.

We had long discussions with @puzrin (who created the current font converter) about this topic and we decided to choose the "continuous format" to save some space in case of smaller fonts.  (If I recall correctly)

Anyway, if we update the font engine to produce A8 output it doesn't really matter. The conversion from Ax to A8 can be very fast. Probably much than the current solution. E.g. for A4 we need an `uint16_t a4_to_a8[256]` array where for all a4 pairs we can easily find the two A8 counterparts.

GorgonMeducer, 2022-09-19:
@kisvegabor The key benefit of A2 and A4 is size. A4 has almost the identical visual effects as A8 but only takes half of the ROM. 

But on the other hand, I don't think people who decide to include a full FONT library would care about this size saving. That's why I strongly support you in abandoning A2 and A4 support in V9.

GorgonMeducer, 2022-09-19:
> choose the "continuous format" to save some space in case of smaller fonts.

This makes A2 and A4 hardware acceleration very hard, and to be honest, the bytes saved by this "continuous" mode are negligible.

puzrin, 2022-09-19:
A3 has the best balance of size/quality. It can be removed for some kinds of hw drivers, but it's not good idea to remove it at all.

Also, for best result, decompressor need to be refactored.

kisvegabor, 2022-09-19:
> A fixed pitch number can give GPU a chance to accelerate A2 and A4.

I'd like to entirely remove A2/A4 from the rendering part. The [`get_bitmap` callback](https://github.com/lvgl/lvgl/blob/master/src/font/lv_font_fmt_txt.c#L79) of the font should return A8 in all cases so the GPU will never never see pitched or continuous A2/A4 only A8.

So up to the font engine how to produce A8 format. Probably the built-in engine will convert the A2/A3/A4 bitmaps into a temporary buffer (maybe using the lookup table I described above).

GorgonMeducer, 2022-09-19:
> Probably the built-in engine will convert the A2/A3/A4 bitmaps into a temporary buffer (maybe using the lookup table I described above).

That's the current implementation:
https://github.com/lvgl/lvgl/blob/389d15be3d950d702b86c5a90b5bead862ea0f5e/src/draw/sw/lv_draw_sw_letter.c#L49-L78

kisvegabor, 2022-09-20:
Yes, but it involves a lot of shifting and masking on A2/A4 bitmaps.

closed 2022-10-12 by lvgl-bot