lvgl/lvgl #3695: Latest doc do not point to v8

Sh3Rm4n, 2022-09-19:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

The `latest` link to the lvgl documentation points to version `v7.11` instead `v8`.

### To Reproduce

https://github.com/lvgl/lvgl/blob/d78e9272526935068eb867327edd4d0358c4e27a/docs/get-started/index.md?plain=1#L5

points to https://docs.lvgl.io/latest/en/html/intro/index.html while latest shows `v7.11`. 

https://github.com/lvgl/docs/tree/c956b6d963b031cb7719094cbdb70cb9140bf74c/latest/en confirms that the symlink is currently pointing to the `v7.11` version.

which is confirmed by 
### Expected behavior

https://docs.lvgl.io/latest/en/html/intro/index.html should redirect to https://docs.lvgl.io/8/intro/index.html or at least show this version. 

### Screenshots or video

[image]

embeddedt, 2022-09-19:
It's intended. We had a lot of links pointing to `latest` back in the v7 days, and if we changed that symlink to v8 it would confuse v7 users. However, now that v7 is no longer supported maybe it is worth retiring this symlink altogether.

Sh3Rm4n, 2022-09-19:
Thanks for the quick answer. This is a reasonable explanation.

kisvegabor, 2022-09-19:
We have dropped the name `latest` ans started to use `master` from v8.

I think it's fine to keep `latest` in v7 docs and use `master` in v8 part. Only few places might have left to fix (like this).

kisvegabor, 2022-09-19:
I've fixed this particular issue in 389d15be3d950d702b86c5a90b5bead862ea0f5e

closed 2022-10-11 by lvgl-bot