lvgl/lvgl #3700: [multi-threaded-drawing]Discussion about drawing tasks

GorgonMeducer, 2022-09-22:
### Introduce the problem
This issue is used as a dedicated discussion about the drawing-task of the multi-threaded-drawing. The original idea comes from #3643.

This discussion covers the following content. 
> 0. define the descriptor of drawing-task
    - define the data structure of drawing-task, `lv_draw_task_t`
    - use `lv_draw_task_t` to describe existing low-level draw operations 
    - extract the key attributes (requirements) from those drawing tasks and generate a list of capabilities
    - design a capability-fast-matching scheme

GorgonMeducer, 2022-09-22:
## Proposal 1:  task and task-group ##
### Background ###
The capability of PE (i.e. 2D GPU, 2D capable DMA, CPU etc) are different: some as powerful as 2D GPU can do all kinds of 2D operations, and some can only do simple operations such as 2D-copy with/without alpha-blending etc. 

Tasks are descriptors of low-level 2D operations. Suppose we have defined a set of simple operations, such as:
- fill colour
- fill colour with a mask
- fill colour with a mask and a given opa
- copy image
- copy image with a mask
- - copy image with opa
- copy image with a mask and a given opa
- copy image with transform 
- copy image with transform and mask
- copy image with transform, mask and opa
...

I won't exhaust this list but you might already notice that: there are building blocks in the list, for example,   fill colour,  copy image, mask, transform, opa etc.  For example, "copy image with transform, mask and opa" is obviously a complex operation, it is a combination of the building blocks aforementioned. 

### Problems ###
If we want to do "copy image with transform, mask and opa", immediately we can comes with an idea about the design of the drawing-task:

- we design the data structure of the drawing-task and let it be able to represent the meaning of "copy image with transform, mask and opa". To achieve our goal, one instance of this drawing-task will be created in the frontend and passed to the backend. 

At the backend, on receiving such **task**,  based on our capability-based-dispatching scheme, ALL PEs will be checked and see whether they are capable to do such a operation, i.e. "copy image with transform, mask and opa":

- If a powerful 2D GPU can do all those "building-blocks", then it is the ideal scenario - the PE takes the task as long as it is not busy.
- If a less powerful 2D GPU that can only do "copy image with transform and mask" but no "opa" as its hardware design can only do transform with mask together (but no extra opacity), then obviously the PE's capability mismatch what the "task" wants and it is considered as "incapable". 

Obviously, for the later scenario, we see there is a workaround to take advantage of the less powerful 2D GPU:
1. Let the 2D GPU do the "transform with mask and opa" and save the result to a temporary frame-buffer
2. use the temporary FB as image and perform a "copy image with opa" operation

YEAH, problem solved, but here comes a question: **How to elegantly makes such a workaround happen in the backend**?

Let me rephrase this question in tech term:
In backend, when receiving a task and no PE (except the default Local software PE) can do all the works described in that task, how can the backend:

1. Find the most capable PEs 
2. Divide the original task into sub-tasks?
3. how to handle the sub-tasks dependency issue (because they almost always draw the same area)?

Well, there must be a solution in the backend only - I have no doubt about this.
But I think this problem can be solved more easily when we have a different design for the drawing-task.

### Solution ###

kisvegabor, 2022-09-25:
**How the drawing tasks are built up?**
It's very important to clarify what "mask" means. I'm planning to rework masking for v9 so it's a great time to figure what are the goals with it.

There are 2 options:
1. The mask is A8 bitmap: A higher level layer  produces the mask bitmap and passes it to renderer. In this case we can make mask-generation modular as well to hook in GPUs or use SW masks. 
2. The masks are operations: E.g. mask out this rectangle with this radius and apply this fading effect. 
 
If we choose 1)  we can optimize out the case of `opa AND mask` because the mask bitmap can be pre-multiplied with `opa`.

**Examples for draw tasks**
There are some cases that I cannot see how fit into your list:
- fill with gradient: Is it a special fill, or something needs to calculate the gradient and pass it as small rectangles or an image? 
- fill rounded rectangle: Is it fill with mask or the radius is a parameter of fill and the PE can handle as it wants?
- border: Is it like fill, or fill with masks?

To have a more complete list I see these tasks types (all can have opa and/or mask):
- fill: coords, color/gradient, radius
- border: coords, width, color/gradient, radius
- shadow: coords, rectangle radius, blur radius, spread, offset 
- arc: coords, width, color(/gradient?), start/end angle, line cap, (bg image?) 
- image: coords, src
- image transform: coords, src, scale, rotate, (other transforms?)
- image recolor?
- line: p1, p2, (point list?), width, color, dash pattern, line cap
- polygon: point list, color/gradient

Note that, we can't always say SW render is a fallback.We can assume that it will  support everything in the future what **any** GPU can do. In other words we can add features to LVGL because we assume there is a GPU (and only the GPU) which can draw it.

**How to break complex tasks into simpler ones?**
> 1. Let the 2D GPU do the "transform with mask and opa" and save the result to a temporary frame-buffer
2. use the temporary FB as image and perform a "copy image with opa" operation

I think it should be handled by the given PE. The capability descriptor can tell "I can do it but in a suboptimal way".

**Tasks and task groups**
If I see it correctly the need of task groups is derived from the issue mentioned in the previous point (handle complex operation in a suboptimal way) and handle "area dependencies".

E.g. ST's ChromaArt cannot fill with opacity, only copy images with opacity. So it needs to fill a temporary buffer first with the color then blend it as an image with opa.  What should we do here? 
1. What if only a small area needs to be filled with opacity? Malloc a temporary framebuffer, fill it and blend it, yield and resume? I think it's better to leave it to SW.
2. What if a screens sized 50% opa  black rectangle needs to be drawn for a modal's background? Malloc a screen sized temporary buffer? It doesn't sound like a good idea. Break it into smaller parts? E.g. malloc 1/10 screen sized buffer and blend it 10 times? Sounds better to me.  

Anyway, it's not trivial. The takeaway for me is that a simple `pe->caps & task->cap` test is not enough because there are a lot of factors to consider. Probably the PE capability check should be rather a callback. Finally, the PE can tell how good it is for a certain operation. If we blindly assigned "fill buffer + blend buffer with opa" for a simple "fill with opa" to ChromeArt probably we'd lose more than leaving it for SW. Not mentioning out of memory issues for the many buffer allocations. 

For the dependency question: I wonder if there could be a point when we malloc a temporary frame buffer to "restart" the dependency chain on an area. E.g. while a GPU is filling the background we can't do anything. However, if we said here is this chart, very complex, independent from other widgets, why can't we draw into a separate buffer in parallel? This way while the SW rendering is waiting for the GPU on other parts of the UI we can make it do some tasks for the chart. 

The most extreme case would be to draw everything into its own temporary buffer. This way there were no dependencies but we wouldn't have enough PE to work on all these areas. Besides the lot of extra copy would reduce the performance too. 

In v8.3 in some cases a new "layer" is created. E.g. to additively blend, transform, blend with opa a group of widget. A layer is basically just a new buffer where LVGL draws and blend that buffer with some blend_mode, transformation, opa, etc. It can be a good starting point to mitigate dependency issues. 

As I imagine  a dependency "graph" would look like this:
```
- screen (fill color)
  - button1 (fill color with radius)
     - icon1 (copy image)
     - label1 (many fill with mask for the letters)  
  - button2 (fill color with radius) 
     - icon2 (copy image)
     - label2 (many fill with mask for the letters)
  - chart on a separate layer (a lot of tasks)
```
1. Can fill the screen and start to work on the chart in parallel
2. Buttons needs to wait as they depend on the screen
3. When the screen is read the buttons can be handled in parallel
4. The icons and labels need to wait for the buttons
5. When the buttons are filled the icons and labels can be drawn independently. 
6. Even the letters can be handled independently.

GorgonMeducer, 2022-09-25:
> How the drawing tasks are built up?
It's very important to clarify what "mask" means. I'm planning to rework masking for v9 so it's a great time to figure what are the goals with it.
>
> There are 2 options:
>
> The mask is A8 bitmap: A higher level layer produces the mask bitmap and passes it to renderer. In this case we can make mask-generation modular as well to hook in GPUs or use SW masks.
The masks are operations: E.g. mask out this rectangle with this radius and apply this fading effect.

I prefer 1)

> If we choose 1) we can optimize out the case of opa AND mask because the mask bitmap can be pre-multiplied with opa.

Well, about this, I would like to say, it depends. At least I can name a scenario:
- An RGB565-A8 image
- We want to copy with mask and opa
Some 2D GPU (including Arm-2D) can process the rgb-array and mask array directly (no buffer is required).

If we "the mask bitmap can be pre-multiplied with opa" then:
- We need to dedicated step to process mask with opa first (a buffer for mask is required)
- Then do the "copy image with mask" later. 

So in total: 2 steps with an extra mask buffer

GorgonMeducer, 2022-09-25:
> fill with gradient: Is it a special fill, or something needs to calculate the gradient and pass it as small rectangles or an image?

See what I said: 
> I won't exhaust this list but ...

So the list I provided is just an example. It is not a complete list. We can add a dedicated draw task or (combination of tasks) to achieve this. For example: 
- Fill with gradient
Then with "mask" and "opa", we can get some useful combinations, such as
- Fill with gradient and opa
- Fill with gradient and mask
- Fill with gradient, mask and opa

> fill rounded rectangle: Is it fill with mask or the radius is a parameter of fill and the PE can handle as it wants?

It is "fill with mask". And we need an algorithm to generate a rounded corner with a given radius (or get a pre-stored mask).
My personal opinion is, some PE is capable of generating a quarter of a circle and fill it with specified colour (8bit grayscale), if not, local SW PE can do it. So this can be a task, i.e. generate a circle with a given radius and colour to fill.

GorgonMeducer, 2022-09-25:
> To have a more complete list I see these tasks types (all can have opa and/or mask):

We will definitely finish this list, but it is not now. I would like to first finish the discussion about task and task group. It is the key schema, once we get agreed on it, we can move to the discussion about the list.

> Note that, we can't always say SW render is a fallback.We can assume that it will support everything in the future what any GPU can do. In other words we can add features to LVGL because we assume there is a GPU (and only the GPU) which can draw it.

No problem

GorgonMeducer, 2022-09-25:
> How to break complex tasks into simpler ones?
> 1. Let the 2D GPU do the "transform with mask and opa" and save the result to a temporary frame-buffer
> 2. use the temporary FB as image and perform a "copy image with opa" operation

> I think it should be handled by the given PE. The capability descriptor can tell "I can do it but in a suboptimal way".

The capability descriptor can of course tell "I can do it but in a suboptimal way", but the dispatcher in practice need to find a way to divide the original task into sub-tasks, one of which can be sent to the PE. 

My proposal: task + task group is a solution for the dispatcher to separate task-group into separate tasks.

GorgonMeducer, 2022-09-25:
> If I see it correctly the need of task groups is derived from the issue mentioned in the previous point (handle complex operation in a suboptimal way) and handle "area dependencies".

YES, exactly.

> E.g. ST's ChromaArt cannot fill with opacity, only copy images with opacity. So it needs to fill a temporary buffer first with the color then blend it as an image with opa. What should we do here?

In general the original task group "fill with opa" will be destroyed and the tasks "fill colour" and "opa (alpha-blending)" inside the task group will be used instead. 

> What if only a small area needs to be filled with opacity? Malloc a temporary framebuffer, fill it and blend it, yield and resume? I think it's better to leave it to SW.

Dispatcher will have list a policies, I believe one of which could be check the area size:
- If it is smaller than a threshold, use SW PE, (e.g. Local SW PE)
- If it is bigger than the threshold, find a faster PE.

For "how to choose PE", I have a lot of ideas, but at the current stage of discussion, we should solve things one by one.  I prefer to start a dedicated new issue for dispatching policies. 

> What if a screens sized 50% opa black rectangle needs to be drawn for a modal's background? Malloc a screen sized temporary buffer? It doesn't sound like a good idea. Break it into smaller parts? E.g. malloc 1/10 screen sized buffer and blend it 10 times? Sounds better to me.

I agree that backend should have the ability to generate tasks, so " Break it into smaller parts? E.g. malloc 1/10 screen sized buffer and blend it 10 times? Sounds better to me." is doable.
A policy agent can evaluate this and make the decision to generate several small tasks.

GorgonMeducer, 2022-09-25:
> Anyway, it's not trivial. The takeaway for me is that a simple pe->caps & task->cap test is not enough because there are a lot of factors to consider. Probably the PE capability check should be [...]

I agree with you. And I have been designing a PE capability ranking algorithm. I have the confidence that it is capable to solve what you said. It takes cost, performance, capability and fast-matching into consideration. I prefer to discuss this in a separate issue.

GorgonMeducer, 2022-09-25:
> For the dependency question: I wonder if there could be a point when we malloc a temporary frame buffer to "restart" the dependency chain on an area. E.g. while a GPU is filling the background we [...]

Sure, the frontend is capable to do that.

> The most extreme case would be to draw everything into its own temporary buffer. This way there were no dependencies but we wouldn't have enough PE to work on all these areas. Besides the lot of extra copy would reduce the performance too.

That's what win32 does: each control has its own framebuffer, and a container(a special control) use z-order to blending all controls' framebuffer together, a window is the top container. 

Embedded systems cannot afford this. But, this doesn't stop us from finding some trade-offs. And I believe the layer cache is the trade-off. 

>As I imagine a dependency "graph" would look like this:
> 
> - screen (fill color)
>  - button1 (fill color with radius)
>     - icon1 (copy image)
>     - label1 (many fill with mask for the letters)  
>  - button2 (fill color with radius) 
>     - icon2 (copy image)
>     - label2 (many fill with mask for the letters)
>  - chart on a separate layer (a lot of tasks)
>Can fill the screen and start to work on the chart in parallel
>Buttons needs to wait as they depend on the screen
>When the screen is read the buttons can be handled in parallel
>The icons and labels need to wait for the buttons
>When the buttons are filled the icons and labels can be drawn independently.
>Even the letters can be handled independently.

I am so glad we are sharing the same picture ! Cheers.
You call it the dependency "graph"，I call it the "element tree". It is the same thing and works exactly the same way you described. So with this dependency graph (i.e. element tree), the frontend can generate parallel tasks at any time when it is possible.

kisvegabor, 2022-09-26:
**Masks**
Ok, let's go with A8 bitmap only and we can figure out a flexible mask generator interface later. 
However, I think rounded rectangle should not be "mask + fill" as some hardware are capable of drawing rounded rectangles directly. The hardware still can fallback to "generate mask + fill" if it wants too.

**Element tree**
It's just theoretical but might important to clarify to be sure we are on the same page.  To make optimal decisions and parallelism we need to see an element tree in this order ([Breadth-first search](https://en.wikipedia.org/wiki/Breadth-first_search)):
```
level 0: screen
level 1: button1, button2, chart
level 2: icon1, label1, icon2,  label2
```
 
So traverse each level first. It's required to start drawing `button1` and `button2` in parallel.

However, LVGL applies [Depth-first search](https://en.wikipedia.org/wiki/Depth-first_search) during rendering. I.e. explore each branch first:
```
- screen (fill color)
   - button1 -> icon1 -> label1
   - button2 -> icon2 -> label2
   - chart
```

To get the first tree from the second we need to buffer some (many) drawing tasks. I think it's not a big issue as the tasks are usually simple and small. In addition with a larger buffer the exploration of the tree can happen independently of the drawing and it doesn't have to wait for having free space in the buffer (or at least not that much with a smaller buffer)

GorgonMeducer, 2022-09-26:
> However, I think rounded rectangle should not be "mask + fill" as some hardware are capable of drawing rounded rectangles directly. The hardware still can fallback to "generate mask + fill" if it wants too.

OK. We can add a dedicated task element for this. 

> To get the first tree from the second we need to buffer some (many) drawing tasks. I think it's not a big issue as the tasks are usually simple and small. In addition with a larger buffer the [...]

Agree with you. We are in the same picture. Confirmed.

kisvegabor, 2022-09-28:
> If we all agree with the task-task group scheme

I'm still not sure about it. Imagine that there is a complex task group, e.g. `A + B + C + D`. There is no PE which can do it as it is so we break it into `A`, `B`, `C`, `D` and assign them one-by-one to the best possible PE. However, there can be a PE which can do `ABC` at once. What do you think about this case?

GorgonMeducer, 2022-09-28:
> I'm still not sure about it. Imagine that there is a complex task group, e.g. A + B + C + D. There is no PE which can do it as it is so we break it into A, B, C, D and assign them one-by-one to the [...]

No. That's not the case. When A + B + C + D is not possible, dispatcher will find a less capable possibility, such as A + B + C. So a new group will be generated by the dispatcher.  Hence we get A + B + C and D. 

In brief, One-by-one is just the worst case. And the capability matching algorithm is capable to find a less optimal solution.

kisvegabor, 2022-09-29:
So if ABCD doesn't work dispatcher will try:
- ABC D  -> A BC D or AB C D
- A BCD -> A B CD or A BC D (was test above too)

If both  ABC D and A BCD can work how can is decide which is the better option? I don't want to see a whole algorithm, just understand the concept.

GorgonMeducer, 2022-09-29:
@kisvegabor 
As we mentioned before the PE will provide a capability description. And in the capability matching, PE will provides a score together with a less optimal solution.

For example, Suppose there are two PEs, PE1 and PE2. When given a capability request "A+B+C+D":
- PE1 reply: A+B+C and a score, for example, 768 (the number is just an example)
- PE2 reply: A+B and a score, for example, 1024
- Default SW PE0 surely support C and D, and each give a score 128 as baseline. 

Hence, PE2(A+B)+PE0(C)+PE0(D) provides the highest score, so this would be chosen as the optimal solution. 

About the performance ranking (here is an proposal):
For each element, such as A, B, C... the performance of the C implementation will be used as the baseline, and marked as 128. If hardware acceleration is two times faster than the baseline, 256 will be given and etc.

These numbers are provided by the GPU driver at compile time. So the GPU driver provide responsible to evaluate the performance and generate the number. If a driver provider don't want to provide this information (or don't have time to measure the performance boost), they can use 128 as the default value).

About the performance ranking system, I put it here as an example and it only shows an plausible algorithm the dispatcher can use. There should be other better solutions, or people can improve this ranking system in the future. I don't want to distract our discussion with this kind of algorithm. When we move to the right stage, we can discuss the policy system and hence discuss similar performance ranking policies.

kisvegabor, 2022-10-02:
Got it, and it seems like a working direction. There can be one catch tough: the relation is usually not linear for GPUs as there is  setup time + execution time. But your suggestion can really give an idea which PE combination to choose.

>> If I see it correctly the need of task groups is derived from the issue mentioned in the previous point (handle complex operation in a suboptimal way) and handle "area dependencies".

> YES, exactly.

I wonder if really need the concept of task groups. So there are two things here: complex operations and area dependency. Can't we solve it on task level?  Do we really really care if some tasks are related to each other? These are just primitives to be drawn somewhere and there are better or worse options to utilize the hardware for them. E.g. if there are some tasks in a buffer we can get any tasks in an order. In fact, probably we are better picking unrelated tasks as they are independent.

Anyway, task groups seems an unnecessary layer to me. Or can you mention an example task group? Maybe I just misunderstood the concept.

GorgonMeducer, 2022-10-02:
> can you mention an example task group?

The 1st floor provided an example. If without a task group, I have no idea how to handle the capability matching in a fast way.

I believe this proposal has limited me.  Let's work in the other way, i.e. starting the next step without the concept of the task group, and when we see any issue, we can re-evaluate it.

kisvegabor, 2022-10-02:
I agree, let's keep in mind and if needed let's get back to it. 
I'm not saying it's a bad idea, just not 100% convinced it's necessary. A little bit seems like premature optimization but I'm not sure.

So the next next step is
> have a brainstorming about the list of task elements.

In [a comment](https://github.com/lvgl/lvgl/issues/3700#issuecomment-1257197900) I collected this list:
> fill: coords, color/gradient, radius
border: coords, width, color/gradient, radius
shadow: coords, rectangle radius, blur radius, spread, offset
arc: coords, width, color(/gradient?), start/end angle, line cap, (bg image?)
image: coords, src
image transform: coords, src, scale, rotate, (other transforms?)
image recolor?
line: p1, p2, (point list?), width, color, dash pattern, line cap
polygon: point list, color/gradient

We can complete it or approach the topic from a fully different angle.

GorgonMeducer, 2022-10-02:
> > fill: coords, color/gradient, radius
border: coords, width, color/gradient, radius
shadow: coords, rectangle radius, blur radius, spread, offset
arc: coords, width, color(/gradient?), start/end angle, line cap, (bg image?)
image: coords, src
image transform: coords, src, scale, rotate, (other transforms?)
image recolor?
line: p1, p2, (point list?), width, color, dash pattern, line cap
polygon: point list, color/gradient

Can you provide some simple explanation of each of those terms? For example, what is coords? 

I can think two more transforms: perspective transform and four points stretch.

kisvegabor, 2022-10-11:
God, I just came here to see if there are any updates and noticed that my reply wasn't sent. Sorry...

So, `coords` is for `coordinates`. I think the rest is quite self-explanatory. E.g. `fill` is a `struct` with 
- coordinate
- gradient
- radius
- mask (see blow)
- opa
fields.

> I can think two more transforms: perspective transform and four points stretch.

True.

> Where are those masks?

All draw_task should have a mask an opa parameter. (I haven't repeated it on each task)

For masks I think there should be 
- a generic interface for simple and very common masks: in the first iteration it can be only a rounded rectangle mask and bitmap mask. The draw tasks need to support these masks in some form. 
- an interface to set up a mask engine in a modular way which can produce bitmap masks.

kisvegabor, 2022-10-11:
> And what is the radius here about?

Eg. NXP's VGLite can draw rounded rectangle on its own. So instead of hard coding that rounded rectangles need to be drawn with masks we should allow to the GPU's to handle this very common case as they can. Of course, a GPU can still use the "mask engine" (let's call it like this for now) to generate a rounded rectangle bitmap mask.

GorgonMeducer, 2022-10-12:
@kisvegabor So radius means rounded rectangle?

kisvegabor, 2022-10-12:
Oh, yes. I didn't understand what was not clear about it. Yes, it means the radius of rounded corners. Just like `style_radius` now. It's a minor things but probably it should be `radius[4]` to describe all the corners one by one.

GorgonMeducer, 2022-10-12:
@kisvegabor OK, I understand it now. Thank you.
I think this is a typical communication problem: we are talking about the same words but have different background knowledge. Before you explaining it, radius just mean: "a line segment extending from the center of a circle or sphere to the circumference or bounding surface" for me.

kisvegabor, 2022-10-13:
I just put the types we have already discussed into the `enum`. But one thing is not clear. If we had `LV_DRAW_TASK_IMAGE_RECOLOR`, `LV_DRAW_TASK_IMAGE_TRANSFORM_ROTATE`, `LV_DRAW_TASK_IMAGE_PERSPECTIVE` how to describe a recolored, rotated, perspective distorted image as one task? Or for these complex cases we need more steps and intermediate buffers? I don't think so because this way we would make it impossible for the GPU to some these tasks at once. As a placeholder I added `LV_DRAW_TASK_IMAGE_COMPLEX`. 

```c
enum lv_draw_task_types_t {
    LV_DRAW_TASK_NONE     = 0x0000,
    LV_DRAW_TASK_FILL     = 0x001,
    LV_DRAW_TASK_BORDER   = 0x002,
    LV_DRAW_TASK_SHADOW   = 0x004,
    LV_DRAW_TASK_ARC      = 0x0008,
    LV_DRAW_TASK_IMAGE    = 0x010,
    LV_DRAW_TASK_IMAGE_COMPLEX = 0x020, /*Recolor, transformations, etc*/
    LV_DRAW_TASK_LINE     = 0x040,
    LV_DRAW_TASK_POLYGON  = 0x080,
};
```

GorgonMeducer, 2022-10-13:
> how to describe a recolored, rotated, perspective distorted image as one task? 

That's why I said : Each item should occupy one bit.

So there will be a member called CapabilityReq:

```c
typedef struct lv_draw_task_t {
    ...
    uint32_t CapabilityReq;
    ...
} lv_draw_task_t;
...

lv_draw_task_t.CapabilityReq 
    = LV_DRAW_TASK_IMAGE_RECOLOR
    |   LV_DRAW_TASK_IMAGE_TRANSFORM_ROTATE
    |  LV_DRAW_TASK_IMAGE_PERSPECTIVE;
```

For these kinds of complex tasks, it is described as above. i.e. we don't need `LV_DRAW_TASK_IMAGE_COMPLEX`.

kisvegabor, 2022-10-15:
If we do it this way the the other option should be more verbose too:
```c
    LV_DRAW_TASK_NONE,
    LV_DRAW_TASK_RECTANGLE_FILL,
    LV_DRAW_TASK_RECTANGLE_GRADIENT,
    LV_DRAW_TASK_RECTANGLE_BORDER,
    LV_DRAW_TASK_RECTANGLE_SHADOW,
    LV_DRAW_TASK_ARC_SQUARE_CAP,
    LV_DRAW_TASK_ARC_ROUNDED_CAP,
    LV_DRAW_TASK_IMAGE_COPY,
    LV_DRAW_TASK_IMAGE_RECOLOR,
    LV_DRAW_TASK_IMAGE_SCALE,
    LV_DRAW_TASK_IMAGE_ROTATE,
    LV_DRAW_TASK_IMAGE_PERSPECTIVE,
    LV_DRAW_TASK_IMAGE_4_POINT_STRETCH,
    LV_DRAW_TASK_LINE_SQUARE_JOINT,
    LV_DRAW_TASK_LINE_ROUND_JOINT,
    LV_DRAW_TASK_LINE_MITER_JOINT,
    LV_DRAW_TASK_LINE_BEVEL_JOINT,
    LV_DRAW_TASK_LINE_DASH,
    LV_DRAW_TASK_POLYGON_FILL,
    LV_DRAW_TASK_POLYGON_GRADIENT,
    LV_DRAW_TASK_POLYGON_SHADOW,
```

And from the actual draw task we can extract the required capabilities.

GorgonMeducer, 2022-10-15:
> If we do it this way the the other option should be more verbose too:

Well, this is what I want at the very beginning...

kisvegabor, 2022-10-15:
:smile: Okay, views are getting closer. 

> Maybe you can review the task-group idea and might have a different understanding.

E.g. a `rectangle fill + rectangle border + rectangle shadow` is task or task group?

GorgonMeducer, 2022-10-16:
> rectangle fill + rectangle border + rectangle shadow

It is a task group with a structure like this:

```c
typedef struct lv_draw_task_t lv_draw_task_t 

struct lv_draw_task_t {
    ...
       lv_draw_task_t *ptNext;
        uint32_t CapabilityReq;
        ...
        uint8_t  bIsGroup    : 1;
        ...
       lv_draw_task_t *ptSubTasks;
       ...
    
    ...
};

lv_draw_task_t tTasks[] = {
[0] = {
    .ptNext = &tTasks[1],
    .CapabilityReq = LV_DRAW_TASK_RECTANGLE_FILL,
    ...
},
[1] = {
    .ptNext = &tTasks[2],
    .CapabilityReq = LV_DRAW_TASK_RECTANGLE_BORDER,
    ...
},
[2] = {
    .ptNext = NULL,
    .CapabilityReq = LV_DRAW_TASK_POLYGON_SHADOW,
    ...
},
};

lv_draw_task_t  tTaskGroup = {
    .CapabilityReq 
        = LV_DRAW_TASK_RECTANGLE_FILL
        | LV_DRAW_TASK_RECTANGLE_BORDER
        | LV_DRAW_TASK_POLYGON_SHADOW,
    .bIsGroup = true,
   ptSubTask = tTasks,
    ...
};

```

The backend dispatcher will first use TaskGroup.CapabilityReg to check against all PEs:
- If Any of the PE can fully match the capability, then the PE will take the whole task group
- If no PE can fully match the capability, then the dispatcher can use the bitmap (and some ranking system) to find the sub-optimal solutions, for example, PE1 and takes LV_DRAW_TASK_RECTANGLE_FILL + LV_DRAW_TASK_RECTANGLE_BORDER and PE2 can take LV_DRAW_TASK_POLYGON_SHADOW, then
    a. A new task group will be created for PE1, i.e.

```c
lv_draw_task_t  tPE1TaskGroup = {
    .CapabilityReq 
        = LV_DRAW_TASK_RECTANGLE_FILL
        | LV_DRAW_TASK_RECTANGLE_BORDER,
    .bIsGroup = true,
   ptSubTask = &tTasks[0],
    ...
};

tTasks[1].ptNext = NULL;

```
    b. `tTasks[2]` will be sent to PE2.

kisvegabor, 2022-10-17:
I see, thanks. One piece of the puzzle still doesn't fit in my mind. 

Let's stay there is the rectangle with `gradient + outline + shadow`. We have types for each, we can OR them and handle as a task group. Is no one can handle it we break it into 3 parts and `gradient`, `outline`, `shadow` will be drawn one after another. So these are independent operations grouped together.

Now let's say we have a rotated and a zoomed image. If no PE can do it we can't break it into 2 tasks and draw a rotated and a zoomed image. So these are not independent operations. 

Same with line dashing. We can't say for `line round joint + dash` if no one take the two to draw a line and a separate "dash".

GorgonMeducer, 2022-10-18:
> Now let's say we have a rotated and a zoomed image. If no PE can do it we can't break it into 2 tasks and draw a rotated and a zoomed image. So these are not independent operations.

I don't think this is a good example. Because:
a. Default PE should always be able to achieve the tasks we list in the `lv_draw_task_types_t`
b. We can of course break it into 2 tasks, i.e. drawing a rotated image and then zooming it as long as there is a frame buffer.

Although this is now a good example, I think I got your point. Let's explain more:

All operations listed in the `lv_draw_task_types_t` can be combined as long as the combination has a practical meaning. 
So,  if `line round joint` + `dash` doesn't mean 'line dashing` in practice, then this combination isn't valid, or at least it should not be integrated as `line dashing`. 
Instead, we should have a dedicated operation called `line dashing`. 

Do you still have questions?

GorgonMeducer, 2022-10-19:
> If we can't combine the options in a modular way we will end up with something like this

Most of the operations can be combined in a modular way, but we have to admit that there are always some exceptions. Adding dedicated operations for those special cases is unavoidable. 

As you have listed, I am OK will them. 

If we are able to get an initial version of `lv_draw_task_types_t`, maybe you can use it to try existing drawing tasks and see whether there are cases that cannot be covered:

- If there are some, then we can discuss here
- If you think it is OK (even temporarily), we can move to the next stage.

kisvegabor, 2022-10-20:
I agree to don't overthink it because probably after 1 day of coding we will find problems that we never thought of. :smile: 

All my doubt is about whether we can describe the operations and capabilities with a simple bit field or not. I think very soon it will turn out that we need to use callbacks on the PEs to ask them how they can handle certain tasks. We can use the bit fields to some extent for sure: simple rectangle building block and images. It should be enough refactor the drawing part and we can figure out the capability matching for more complex cases later.

> So, how about halting the doubts you have (for now) and moving to next stage?

Ok, here are my thought:

>  Finish an initial version of lv_draw_task_types_t

So I suggest to go with these only for first:
```c
typedef enum {
   LV_DRAW_TASK_TYPE_FILL,
   LV_DRAW_TASK_TYPE_BORDER,
   LV_DRAW_TASK_TYPE_IMAGE,
} lv_draw_task_types_t;
```

> define the initial version of lv_draw_task_t

I can't see now what are the generic parts of the tasks. This is sure:
```c
typedef struct {
   lv_draw_task_types_t type;
} lv_draw_task_t;
```

>  for each operation listed in lv_draw_task_types_t, derive a class from lv_draw_task_t and define the necessary members to describe the target operation.

```c
typedef struct {
  lv_draw_task_t base;
  lv_area_t coords;
  lv_color_t color;
  lv_coord_t radius;
  lv_opa_t opa;
  lv_opa_t * mask;
} lv_draw_task_rect_fill_t;

typedef struct {
  lv_draw_task_t base;
  lv_area_t coords;
  lv_color_t color;
  lv_coord_t radius;
  lv_coord_t width;
  lv_opa_t opa;
  lv_opa_t * mask;
} lv_draw_task_rect_fill_t;

/*It's a simple blend*/
typedef struct {
   lv_draw_task_t base;
   lv_area_t dest_coords;
   lv_color_t * src;
   lv_coord_t width;
   lv_coord_t stride;
   lv_coord_t height;
   lv_opa_t opa;
   lv_opa_t * mask;
} lv_draw_task_image_t;
```

What do you think?

GorgonMeducer, 2022-10-20:
Looks OK, except for this:

>  typedef struct {
       lv_draw_task_types_t type;
} lv_draw_task_t;

instead of lv_draw_task_types_t, it should be an integer for bitmap, so I propose:

```c
typedef struct {
       uint32_t  type;
} lv_draw_task_t;
```

kisvegabor, 2022-10-21:
> instead of lv_draw_task_types_t, it should be an integer for bitmap, so I propose:

Okay.

> And I guess, there is a typo about lv_draw_task_rect_boarder_t, you put lv_draw_task_rect_fill_t

Yes, it was a typo, I fixed it.

---

Hopefully I can merge the new driver architecture in this weekend. So next week we can start to work on it. I think the most important would be see all the layers working together each with very minimal features. So we need to implement:
- the mechanism of collecting/buffering drawing tasks
- the level 0 dispatcher which assign everything to SW renderer 
- the new draw API
- see something about the multi-thread architecture ((RT)OS level).  

What do you think?

GorgonMeducer, 2022-10-21:
> What do you think?

Yes, having a working prototype is a good start. 

> - the mechanism of collecting/buffering drawing tasks
> - the level 0 dispatcher which assign everything to SW renderer

We can start from this and implement an empty dispatcher that only receives tasks and frees them.

kisvegabor, 2022-10-23:
> And to make it work with existing infrastructure, I propose to implement the new structure (frontend and backend) as a 2D-GPU, so :

At this moment I don't see if we can make the old and new approach live together as there are system-wide changes to buffer draw tasks, flush them, work on multi threads, etc. 

What I really want to see even only with rectangles + images is that all my CPU cores are spinning at 100% to render some basic UI at 10000 FPS. To do this we need to rework a lot of things. But we can do it in a separate branch, so I see no problem with breaking the current rendering for a some months in a "sandbox".

What do you think?

GorgonMeducer, 2022-10-23:
> At this moment I don't see if we can make the old and new approach live together as there are system-wide changes to buffer draw tasks, flush them, work on multi threads, etc.

Based on my experience of creating Arm-2D GPU, It is possible, and it is easy.

kisvegabor, 2022-10-24:
IMO it's not important to keep LVGL (or the branch in which we are working) functional. So I think it's a little bit counterproductive to spend time on maintaining the old and the new system's compatibility. 

> For example, we implement lv_draw_async_blend and use it as the wrapper of the frontend, so we can pick the drawing work we are interested in and create draw tasks in the frontend and send it to the backend etc...

Sorry, but I still can't see it in the practice. 

What is your your attitude about the development of this feature? I can implement some parts of it. Do you also planning to contribute with coding as well? If so, probably I could understand your idea better if you shown an example in a PR.

GorgonMeducer, 2022-10-25:
@kisvegabor I'd like to contribute some of the code but to be honest, I am not able to guarantee the time. 

> IMO it's not important to keep LVGL (or the branch in which we are working) functional. So I think it's a little bit counterproductive to spend time on maintaining the old and the new system's compatibility.

In fact, I don't know how to start if we have to create a new thing from scratch. It is a know-how issue of mine. I don't know the LVGL as well as you are. The only way to start from my side is to implement a 2D GPU first and gradually add more features to this 2D GPU and eventually create a good enough prototype. Why I choose this way because:

- At any stage, I always have a working LVGL to validate its functional correctness.
- At any time, I can focus on interface-oriented-programming, i.e. treat other parts of LVGL as black-box 
- At any time, I can add new features gradually and choose a different interface (and enlarge the working area) when necessary.

That is the method I am familiar with. So maybe I can make it more clear:

- Why do I choose the 2D-GPU interface? Because it is an existing interface that I have been working on.
- With this interface, I can treat other parts of LVGL as a black box.
- This interface provides sufficient information for me to extract drawing tasks, hence I know how to create drawing tasks and test them. 

On the other hand, if you can start a framework and leave some blanks, then I can contribute some modules. 

You can choose a path, I am OK with both.

kisvegabor, 2022-10-26:
I'd give it try to see if we can make a large change without breaking everything.

My idea is to convert the current `draw_dsc`s to `draw_task`s in `lv_draw_...()` functions, buffer them, and in a later stage make the actual drawing based on the buffered `draw_task`s. To keep LVGL functional temporarily we can have very complex `draw_task`s too (copy of `draw_dsc`s) and gradually simplify them. 

Two thing came to my mind:
- Now before drawing it's possible to create a text into a local variable and draw it. If the drawing will be async the texts needs to be buffered. Or if we handle text drawing as many A8 images the bitmaps needs to be buffered and freed later. 
- If we handle letters as A8 images, then for a longer texts there will be a lot of `draw_task`s.

GorgonMeducer, 2022-10-26:
> - Now before drawing it's possible to create a text into a local variable and draw it. If the drawing will be async the texts needs to be buffered. Or if we handle text drawing as many A8 images the [...]

Sure. Why not?

kisvegabor, 2022-10-26:
> Maybe A8 text display is a good example that we should keep it for local SW PE.

I though it could be a good example for parallel rendering :smile: Here we have many independent images that we could blend in parallel. Just create simple draw tasks like crazy and let the PEs work on them in parallel. E.g. VGLite could take 8 letters at once to work on a batch.

GorgonMeducer, 2022-10-26:
OK, I got it.

kisvegabor, 2022-10-29:
I was thinking about it more I think the safest step would be to fully consider `draw_dsc`s as `draw_task`s at this moment and rework only the high level architecture. Collect draw tasks and dispatch them in parallel. This way LVGL will keep working and we can gradually change the `draw_dsc`s to simpler `draw_task`s

GorgonMeducer, 2022-10-31:
@kisvegabor OK. I follow you and see any chances that I can contribute.

kisvegabor, 2022-11-05:
I started to think more about the details and this what I come up with. It has some changes compared to what we have discussed earlier, but  this made the most sense me. It seems to it can be integrated without too much break. 

I'm very curious about your opinion. I organized my thought into a (very long) bullet list to make it easier to comment on it.

### Basic idea
- Draw API similar to the current  with `lv_draw_dsc`s
- `lv_draw_dsc`s are more or less become `draw_task`s  
- In `lv_draw_...()` functions `draw_tasks` are created based on the `draw_dsc`s 
- `draw_task`s are stored in `draw_ctx` to allow rendering only with a `draw_ctx` e.g. in `lv_canvas`
- When a `draw_task` is added `dispatch()` is called (see later)
- When there are no more objects wait for draw ready and call `disp_flush()`.

### Dispatching
- When a `draw_task` is assigned to a PE it's marked as `in_progress`
- When a `draw_task` is finished it's marked as `finished`. `dispatch()` will clean up the finished tasks.  
- There are 2 cases when we `dispatch()` should run:
    1. when a `draw_task` is added
    2. when a `draw_task` is finished
- `draw task`s can be added only in the main thread. After that `dispatch()` is called immediately.
- `draw task`s can be finished in the 
    - main thread: call `dispatch()` on finish 
    - other threads: send `dispatch_request()`
    - interrupts: send `dispatch_request()`
- There are extra `dispatch_if_requested()` calls as the main thread traverses the tree to process async requests frequently
- `dispatch_request()` makes the currently running `dispatch()` restart, to re-evaluate with the new state    
- `dispatch_request()` is required (instead if calling `dispatch()` directly) to avoid starting e.g. SW render from an interrupt 
- Strategies and consideration: 
    - If 2 PEs are equally good for a task, choose the dumber to not use a more flexible for simple tasks
    - Optional: assign  `draw_task`s only if there is at least e.g. 10 buffered `draw_task`s to allow making a better decision based on future `draw_task`s   
    -  Let PEs pick more `draw_task`s if they want for batch execution
- Implementation idea: 
    - Store the PEs in a list: dumbest (most preferred) first
    - Pass the entire `draw_task list/tree` to the PE and let it pick whatever it wants
    - This way the PE can do the capability checking and we could have more complex `draw_task`s (i.e. not 10+ `draw_task` types for line drawing only)

### Draw task types
- Separate the current `rect_draw` into `bg`, `border` (also for outline) and `shadow`
- Keep the others (image, line, etc) as they are 
- If a mask is used a new draw layer is created and the mask will be applied on it resulting in an RGB + alpha mask blending (ARGB image?)
- Blending: remove the blend function for the `draw_ctx`. It's an internal SW draw utility, and it should be implemented in rectangle in image drawing. I.e. if blending is needed call `lv_draw_sw_rect_bg()` or `lv_draw_sw_img()` instead         
- `label`: the `draw_task` contains the text (and not a lot of A8 image blending for letters).  We provide a function to get the letters' positions from the text. This way the PE can handle the A8 belnding as it wants (e.g. batching)   
- Drop the idea of breaking e.g. complex image transformation into atomic operation (blend (SW) + zoom (GPU1) + rotate(GPU2)) as it's not lifelike to have 2 semi advanced GPU and have a better result by combining the two. Plus it adds a huge complexity. So I'd postpone until we are sure it has practical benefits 

### main thread:
- Traverses the widget tree and calls `lv_draw_...()` functions to create `draw_task`s
- On `draw_task` creation `dispatch()` will be called from here too 
- `display_flush()` will be called from here too

### SW render
- Each core can be considered as an independent SW render engine
- If there is only one core,
    - It creates `draw_task`s, `dispatch` and SW renders too. 
    - Similar as now, the only difference is that the `draw_task`s can be buffered while the GPU is working
- If there are  > 1 cores
    - One can be the main thread and the other can be SW render engines.      
    - The SW render cores are new tasks

### GPUs
- They work in the background
- They can maintain their own internal task list (e.g. break radius==0 border into 4 rectangles) and iterate execute the list
- Probably will be finished in a interrupt from where `dispatch_request()` is called

### Example hardware
- 1 core (any simple MCU)
    - When a `draw_task is added it's immediately dispatched to the SW render running on the same core. 
    - Same as now
    - No OS is required
- 1 core + 1 GPU (STM32)
    - The 1 core is the main thread which can assign `draw_task`s either to itself or to the GPU
    - `draw_task`s can be buffered while the GPU is running
[... 26 lines cut ...]
1. Draw `bg`
    1.1. main thread: process -> buffer `draw_task` -> `dispatch()` -> assign to the GPU
2. Draw `blue_cont`: 
    2.1. main thread: process -> buffer `draw_task` -> `dispatch() `-> the new `draw_task` depends on `bg` -> do not assign
3. Draw  `blue_cont_border`: 
    3.1. main thread: process... 
    3.2. GPU interrupt: `bg` is ready -> `dispatch_request()`
    3.3. main thread: buffer `blue cont border`'s `draw_task` -> `dispatch()` depends on `blue_cont` -> do not assign and clear the dispatch request
4. Draw `blue_cont_image`: 
[... cut ...]
    6.3. main_thread: `draw_outline` -> `buffer draw_task` -> `dispatch()`: 
          6.3.1. Assign `blue_cont_image` to GPU (make GPU busy first)
          6.3.2. Assign `red_cont_border` to SW render -> render now
7. Draw `red_cont_text`:
    7.1. GPU interrupt: -> `blue_cont_image` is ready -> `dispatch_request()`
    7.2. process -> `buffer draw_task` -> `dispatch()` ->  If the GPU can do A8 blending assign to it else to the SW render and render now
8. Wait for ready:
    8.1 If the GPU is ready call `flush_cb`

GorgonMeducer, 2022-11-05:
Looks like you have put many details into this solution and I see no problems so far. 
Can't wait to see a working prototype.

kisvegabor, 2022-11-05:
Cool! 
The new driver architecture (#2720) is still in progress, but in the meantime I'll make some experiments.

kisvegabor, 2022-11-06:
Aaand it's working :slightly_smiling_face: 

Here is what I did:
- Commented out all draw functions except `lv_draw_rect`
- Disabled all the widget
- Created some SW render threads
- Buffered `draw_dsc`s into `draw_task`s
- Dispatched them to render cores
- Cried in pain for few hours while it wasn't working :smile:  

I tested it on my 8 core, 16 thread AMD Ryzen 5800H (quite new and powerful) notebook.

| Render thread count | Render time [ms] |
|-----------------------------|------------------------|
| 1 | 56 |
| 2 | 40 |
| 4 | 35 |
| 8 | 31 |
| 15 | 30 |
| 16 | 30 |
| 32 | 30 |

1 render thread:
[image]
In `lv_draw.c` form the 31%, 28% (almost all) is spent with waiting for the draw thread to be ready before flushing. We should block the task here. 

16 render thread:
[image]

`lv_area.c` and `lv_draw.c` (40% together) is the overhead of the dispatcher. Of course, it's no ideal. It's just the first day version.

I pushed it to https://github.com/lvgl/lvgl/tree/feat/parallel-rendering

kisvegabor, 2022-11-06:
It was the UI tested with (below the CPU usage with 4 render thread + 1 main thread)
[image]

kisvegabor, 2022-11-07:
Meanwhile I investigated what the main blocking parts of parallelism are. When the full render time is 30 ms, drawing the big bg rectangle is 11 ms which blocks all the other items as all are depending on it. 

I wonder it might make sense to break some large-area tasks into smaller ones to make them render in parallel and resolve dependencies earlier.

I can see more or less the whole system working in my mind but the management of masks still not clear to me.

GorgonMeducer, 2022-11-07:
> When the full render time is 30 ms, drawing the big bg rectangle is 11 ms which blocks all the other items as all are depending on it.

Maybe we can introduce a `dependency counter` in each draw_task to indicate how many other tasks depend on it. Once the counter exceeds a given threshold, we can call an algorithm to divide the area of this drat task into several parts and distribute them to PEs for parallelism.

kisvegabor, 2022-11-07:
> Maybe we can introduce a dependency counter in each draw_task to indicate how many other tasks depend on it.

I think it'd make sense to divide even if there is no dependency assuming there are PEs to draw the tiles in parallel. This way the original 10ms rendering time could e.g. 5 ms with 1 CPU and 1 GPU.  Actually we could check how many free PEs we have at a moment and how many of them could draw that large area and dived the area among them. 

> About the algorithm to divide the area of the blocking draw_task, one immediate idea is to divide the area based on the draw_tasks that depend on the blocking one. 

It turned out that dealing with dependencies could be complicated. On the UI I've shown above there are 400 `draw_task`s and it's hard to avoid O(N^2) processing. Maybe we could build a tree/graph instead of list but I can't imagine it now. 

Although probably we should limit the size of the `draw_task` buffer. The 400 tasks assuming 100 byte for each needs 40 kB of RAM. If the rectangle drawing were split into bg, border and shadow it would be better as shadow, border and outline are rarely used. So maybe 15kB. 

Anyway, the gist is that we see some solutions and we can leave the details for a later stage.

GorgonMeducer, 2022-11-08:
> Maybe we could build a tree/graph instead of list but I can't imagine it now.

Any tree/graph will be translated into a list because this list is just a FIFO. 

> It turned out that dealing with dependencies could be complicated.
In fact, we only need to consider 1 level of dependency.

kisvegabor, 2022-11-08:
There are cases when the children can be out of their parent. E.g if `LV_OBJ_FLAG_OVERFLOW_VISIBLE` is enabled.  

With having a lot of buffered ares it's more likely that we can find an independent area to draw. Imagine a screen with 2 large and slow-to-draw containers. Each has 30 children. If we buffer only 10 tasks, we need to wait for 3 full buffering until we "see" the second container. With a buffer size of 50, we will immediately (means as LVGL get there is processing) see the second container and an other PE can start to work on it in parallel.

kisvegabor, 2022-11-09:
> This is definitely not friendly for MCUs that have less SRAM compared to those MPUs.

With buffer size of 50 and rectangle drawing split (bg, border shadow), assuming, 50 byte/task it's only 2.5kB. I'd happily "pay"  2.5kB for better parallelism. But should be configurable. 

> Think about introducing a limitation that children will never be out of their parent, then we immediately see a lot of optimisation opportunity.

`LV_OBJ_FLAG_OVERFLOW_VISIBLE` is very rarely used and we can add a flag the normal and overflow cases. 
 
It's also wort mentioning that there are some non-obvious cases:
1. Let say there is a rectangle at `(100;200) (100;200)` but it has 10 px shadow resulting in `(90;210)(90;210)` draw area but children are still clipped to the smaller `(100;200) (100;200)` area  
2. If there is an 5 px border at `(100;200) (100;200)` tasks in `(105;195) (105;195)` are independent
3. If there is a rotated/zoomed object the draw area is completly different than the original area on which the children are clipped. There are 2 cases here: `lv_img` transformation which transforms the image directly not caring with the children, and `style_transform_...` which creates a snapshot with all children in a new layer and rotates it as an image.

GorgonMeducer, 2022-11-09:
For 1 and 2, we can enlarge the area to include shadows and borders.

For 3, I didn't see any problems. Maybe you can explain more?

kisvegabor, 2022-11-10:
> For 1 and 2, we can enlarge the area to include shadows and borders.

I meant something different. There are 2 different areas eg. for a border:
1. the bounding box should be used to check if we can start to draw the border (i.e. check the dependencies of the border)
2. the smaller inner area shall be used by the children during dependency checking. (i.e. the the border and the children can be drawn at the same time, assuming the children ar not overlapping the border.) 

> For 3, I didn't see any problems. Maybe you can explain more?

Nothing special, just we need to save as the transformed area as `draw_area` which is not related to the actual coordinates of the object.

[27 more replies]

closed 2023-02-25 by kisvegabor