lvgl/lvgl #3701: Very low FPS and high CPU during benchmark - ESP32 + ILI9488 + LVGL V8.3 + Arduino IDE

dills-b, 2022-09-24:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

### Expected behavior

Thanks for taking a look at this. My guess is that there is something I have not configured properly but I have been unable to make any progress.

C47D, 2022-09-24:
It's been a while since I used a similar setup, the limitation is the SPI interface and the color depth. Using a parallel interface showed bigger FPS.

Saying that maybe there's something else others can add so you could improve the FPS.

closed 2022-10-16 by lvgl-bot

r-karthik-r, 2024-05-07:
hi @dills-b, struck with the same issue. the load speed is satisfactory while directly controlling with TFT_ESPI library. so confused. Found any resolution?

kisvegabor, 2024-05-09:
Yes, we have found something. The drawing pipeline on v9 is more complex than v8 and probably due the higher number of nested functions the ESP chip is out of flash cache. We can mitigate it by introducing a few shortcuts for typical cases, e.g. immediately blend an image without any transformation or recoloring.

Exactly which use case was slower or you in v9?