lvgl/lvgl #3705: LV_IMG_CF_TRUE_COLOR_CHROMA_KEYED is not treated as having alpha channel

yiliu-leo, 2022-09-27:
### LV_IMG_CF_TRUE_COLOR_CHROMA_KEYED is not treated as having alpha channel
This maybe a bug or this customized feature should be maintained by users.
we are using LV_COLOR_CHROMA_KEY to draw partial transparent images with GPU_NXP_PXP.

### To Reproduce
When two images overlapping, if the top image is hidden, the LV should draw the cover area of the bottom image. But if the bottom image uses LV_IMG_CF_TRUE_COLOR_CHROMA_KEYED, the LV doesn't draw the cover area but draw it from the left-top of the bottom image.
The problem happens in lv_gpu_nxp.c at line 260. The function lv_img_cf_has_alpha(cf).

```
#if LV_USE_GPU_NXP_PXP
    if(!mask_any && !scale && !need_argb8565_support()
#if LV_COLOR_DEPTH!=32
       && !lv_img_cf_has_alpha(cf)
#endif
      ) {
        done = (lv_gpu_nxp_pxp_blit_transform(dest_buf, &blend_area, dest_stride, src_buf, coords,
                                              dsc, cf) == LV_RES_OK);
        if(!done)
            PXP_LOG_TRACE("PXP blit transform failed. Fallback.");
    }
#endif
```

### Expected behavior

```
/**
 * Check if a color format has alpha channel or not
 * @param cf a color format (`LV_IMG_CF_...`)
 * @return true: has alpha channel; false: doesn't have alpha channel
 */
bool lv_img_cf_has_alpha(lv_img_cf_t cf)
{
    bool has_alpha = false;

    switch(cf) {
        case LV_IMG_CF_TRUE_COLOR_ALPHA:
        case LV_IMG_CF_RAW_ALPHA:
        case LV_IMG_CF_INDEXED_1BIT:
        case LV_IMG_CF_INDEXED_2BIT:
        case LV_IMG_CF_INDEXED_4BIT:
        case LV_IMG_CF_INDEXED_8BIT:
        case LV_IMG_CF_ALPHA_1BIT:
        case LV_IMG_CF_ALPHA_2BIT:
        case LV_IMG_CF_ALPHA_4BIT:
        case LV_IMG_CF_ALPHA_8BIT:
//--------------------------------------------------------------------------------------------------------------------
        case LV_IMG_CF_TRUE_COLOR_CHROMA_KEYED: // maybe need to add this
//--------------------------------------------------------------------------------------------------------------------
            has_alpha = true;
            break;
        default:
            has_alpha = false;
            break;
    }

    return has_alpha;
}
```

### Screenshots or video
Try to represent it as below:
[image]

kisvegabor, 2022-09-28:
It seems `lv_img_cf_has_alpha` is used incorrectly here. It return `true` if the color format has an dedicated alpha channel. 

I think it can be easily fixed by adding a `cf != LV_IMG_CF_TRUE_COLOR_CHROMA_KEYED` check too into the if. 

cc @nicusorcitu

nicusorcitu, 2022-09-29:
PXP does support chroma key (on both 16 bits and 32 bits color depth) and alpha channel only for 32 bits color depth.
I do not see the problem with that check. If I add "cf != LV_IMG_CF_TRUE_COLOR_CHROMA_KEYED" on that check, the PXP will not process anylonger the chroma key on 16 bits.

closed 2022-10-22 by lvgl-bot