lvgl/lvgl #3706: LCD driver framework, some gotchas

codewitch-honey-crisis, 2022-09-28:
### Some pitfalls with creating a driver framework that works across multiple platforms

I remember @kisvegabor mentioning the future inclusion of a driver framework, and I didn't get much chance to look at it yet, but I wanted to drop something important that seriously impacts any potential attempt to make it generic where it applies to the ESP32S3.

The new i8080/SPI/I2C LCD controller framework in the ESP32S3 is necessary to take full advantage of the ESP32S3 hardware, and it doesn't work *anything* like the other platforms. Your generic framework may not interface with this. That's the problem I ran into, and I suspect Bodmer ran into, and the author of LoyanGFX will run into, should they ever try to take advantage.

I say this because it blew a hole in my arduino graphics drivers for htcw_gfx which can't take advantage of the S3 (it runs but not optimally)
I say this because TFT_eSPI doesn't take advantage of the S3
I say this because LoyvanGFX suffers the very same issue.

I wanted to head it off here.

Before you make a generic driver framework, study the i8080 hardware accelerated facilities of the s3.
It includes but is not necessarily limited to the include files here.

Why does it matter?
Traditional i8080 bitbanging on the S3 yielded 37FPS running LVGL benchmarks on an ILI9488
The same thing using the LCD facilities got 56FPS conservatively.

It's also designed with LVGL in mind, so it should be easy to integrate aside from the above gotcha

But like I said, it doesn't work anything like other platforms do.

```C
#include "esp_lcd_panel_interface.h"
#include "esp_lcd_panel_io.h"
#include "esp_lcd_panel_vendor.h"
#include "esp_lcd_panel_ops.h"
#include "esp_lcd_panel_commands.h"
```
I've not found the actual documentation for it yet. I wrote a driver though you can use for reference which I pieced together after scanning messageboards for info. https://github.com/codewitch-honey-crisis/EspDisplayS3 it's under /lib/lcd_controller

### Examples and cases

See my link to my project above

### Suggested solution

This is above my paygrade as I haven't studied your proposed driver framework yet. I'm just trying to head off a potential problem.

kisvegabor, 2022-09-28:
Thank you for bringing this to our attention. There are three things that makes me feel that there is hope :slightly_smiling_face: 
1. As you mentioned they designed it with LVGL in mind. By coincidence today I needed to look into [ESP LCD](https://github.com/espressif/esp-idf/tree/master/components/esp_lcd) to fix a client's issue and I found a quite LVGL-ish driver and interface.  
2. I made some progress (i.e. more refactoring) on the new drivers an I'm very happy with the result. Now it has a super simple API allowing to hide anything under it. Here you can see the 
    - [TFT_eSPI driver](https://github.com/lvgl/lvgl/blob/arch/drivers/src/dev/disp/tft_espi/lv_tft_espi.cpp) (Only 84 lines)
    - [SDL driver ](https://github.com/lvgl/lvgl/blob/arch/drivers/src/dev/sdl/lv_sdl_window.c)(supporting multiple windows and window resize)
    -  [Linux frame buffer driver](https://github.com/lvgl/lvgl/blob/arch/drivers/src/dev/disp/fb/lv_linux_fbdev.c)

To be sure v9 will be a real major release I completely changed the display and indev API. :smile:  
Now it's much more simpler yet more flexible.

You can see how they work in the links above. 

In the driver there will be two layers. So far I thought they need to be designed together but I've already realized they can be separated even in the design phase.  Now I'm working on the higher level a don't think about low level platform dependent drivers. E.g. SDL, Linux framebuffer or (to tell an embedded example) ILI9341. So I still don't know how to connect e.g. low level SPI drivers. On high level it can use `lv_ili9341_disp_create()` and it can know all the commands and API for ili9341 but I can't see how to describe the SPI level. 
- Shall we care about how the SPI needs to be initialized or we need only a callback to send data over SPI?
- Same question for IO: shall we have an API to set RESET pin to output?
- Shall we handle to CS pin?
- Or shall we approach it from a higher level? E.g. have only two callbacks: `send_image` and `send_cmd` and they can do whatever they want on any interface they want. And we only provide some examples for some boards.  

@amirgon Now I used `display->user_data` to store some driver specific data but realized that it might not be the best idea for MicroPython. I can switch to the same schema that we use for widgets. E.g.
```c
typedef struct {
   lv_disp_t disp;
   ...SDL data...
} lv_sd_window_t
```

codewitch-honey-crisis, 2022-09-28:
I'm drawing from my experience writing htcw_gfx. The way I see it, you have two potentially competing design goals here - which is factoring out the driver from the bus to save maintenance and just code, versus actually making the thing work with every platform, since some platforms like Teensy, and the ESP32S3 are a bit funny. The ESP32S3 already has its own decoupled bus framework, which is actually kind of a bad thing from an integration standpoint. Look at the way TFT_eSPI is coded vs the lcd_controller files i linked you to for example. It's entirely different for the ESP32S3 than anything he has in there. 

I think that on reflection, you should couple the driver to the bus(es) it can use. For example, the SSD1306 typically comes in SPI or I2C varieties (although in theory it supports i8080 parallel I think? I've never seen it in practice for those displays), so your SSD1306 driver would have the bus code for each of those embedded in it.

It's more work and potentially more maintenance, but it's also more cross platform.

The other option is to make specialized drivers for the weird platforms, which is what I started to do with the Teensy 4.1.

kisvegabor, 2022-09-29:
> which is factoring out the driver from the bus to save maintenance and just code,

That is to not handle SPI1, SPI1, etc but just an abstract communication channel, right? It suggests having a higher level abstraction.

> actually making the thing work with every platform, since some platforms like Teensy, and the ESP32S3 are a bit funny 

And it suggests having lower level abstraction to save the drive config pain for the user. 

> I think you should approach it at the highest level possible though, for maximum platform portability. Send image and send command would directly integrate with the ESP32S3's LCD api

So the user configs the SPI pins as he can/wants, puts together some `send_image` and `send_command` functions, and passes them to the display driver? 

It could work but this way it's the user's responsibility e.g. to handle DMA properly. This way we can have example projects where we show reference driver implementations. E.g. this how we created a SPI-DMA driver on ESP32, or this how we configured SRAM, DMA and LTDC on STM32. 

What do you think?

codewitch-honey-crisis, 2022-09-29:
> That is to not handle SPI1, SPI1, etc but just an abstract communication channel, right? It suggests having a higher level abstraction.

Yeah, it's a higher level abstraction. If you did go that route, I would suggest abstracting such that SPI, I2C and i8080 parallel are represented by an abstract bus type. However, this route is plagued with pitfalls, like not working well with the ESP32S3 and possibly the Teensy 4.1

> And it suggests having lower level abstraction to save the drive config pain for the user.

I'm not sure how it would save pain. Consider the declaration process in GFX

```cpp
using bus_type = tft_spi_ex<VSPI,PIN_NUM_CS,PIN_NUM_MOSI,PIN_NUM_MOSI,PIN_NUM_CLK,SPI_MODE0,
false
#ifdef OPTIMIZE_DMA
,LCD_WIDTH*LCD_HEIGHT*2+8
#endif
>;

using lcd_type = st7789<LCD_WIDTH,LCD_HEIGHT,PIN_NUM_DC,PIN_NUM_RST,PIN_NUM_BCKL,bus_type,3>;
```
There's not a lot of extra typing there. You'd have had to have put the arguments of tft_spi_ex into st7789 anyway, somehow. And to change to parallel i8080 you'd just change the bus_type declaration line

That said, LVGL is C, so of course, those pins would have to be set in lv_conf? as #defines, not as a template, but I'm just trying to illustrate a principle here.

I'm not sure how different the configuration would be for the user regardless of how this is abstracted.

With all of the above said, I'm not real clear on what you mean when you say pain. At least more pain than otherwise.

> So the user configs the SPI pins as he can/wants, puts together some send_image and send_command functions, and passes them to the display driver?

I'm thinking the user #defines pins and the driver chip the same way they do with TFT_eSPI, and then the driver sets up your callbacks - the ones the user has to set up currently. We keep that, because for some platforms, a driver will work pretty much that way anyway, like the ESP32S3s framework wants you to have those callbacks. It's designed for them. So keep them.

The driver fills them in for you, if you select one. Otherwise you fill it in yourself for a custom driver - and that looks EXACTLY how setting up a driver for LVGL looks now.

> It could work but this way it's the user's responsibility e.g. to handle DMA properly.

As long as we keep your current callback system (with the "flush ready" callback or whatever) that shouldn't be much of a problem. The only issue i see if someone wants DMA they aren't using your driver framework at all.

amirgon, 2022-09-29:
> @amirgon Now I used `display->user_data` to store some driver specific data but realized that it might not be the best idea for MicroPython. I can switch to the same schema that we use for widgets. E.g.

@kisvegabor - It's best not to abuse `display->user_data` for anything, and not only because of Micropython.  
There may be other use cases either for a user to attach some context, or for a framework such as C++ to attach the related object to it.  
The Widget scheme you suggested sounds good to me.

In general, I would like to see a more realistic example of the new display driver framework.  
In your examples the user can set `hor_res` and `ver_res` but not much more than that.  

Today Micropython display drivers recieve many other parameters, take [ili9341](https://github.com/lvgl/lv_binding_micropython/blob/a69f4b514e426a129fc3e891e0bf22a5f88845a9/driver/esp32/ili9XXX.py#L541-L545) for example.  
In your driver framework, where do you set the SPI pins? modes? freq? color mode? etc.  
Remember we need all these to be set on runtime in Micropython, so they can't be preprocessor defines.

kisvegabor, 2022-10-02:
@codewitch-honey-crisis 
> Yeah, it's a higher level abstraction. If you did go that route, I would suggest abstracting such that SPI, I2C and i8080 parallel are represented by an abstract bus type. However, this route is plagued with pitfalls, like not working well with the ESP32S3 and possibly the Teensy 4.1

Why isn't is enough to heave and abstract "channel" and the user can use any bus inside?

> I'm not sure how different the configuration would be for the user regardless of how this is abstracted.

You are right. We'd redo what is already done in BSP or vendor driver. We should look it from a higher level.

> The driver fills them in for you, if you select one. Otherwise you fill it in yourself for a custom driver - and that looks EXACTLY how setting up a driver for LVGL looks now.

It's basically supporting some selected (popular) platform (ESP, STM, NXP, ?) and have very lean an targeted SRAM, SPI, I2C, DMA and LCD drivers for them. It might work. And it's modular, so we can add this layer at any time.

@amirgon 
> The Widget scheme you suggested sounds good to me.

Okay, I'll update it accordingly. 

> Today Micropython display drivers recieve many other parameters, take [ili9341](https://github.com/lvgl/lv_binding_micropython/blob/a69f4b514e426a129fc3e891e0bf22a5f88845a9/driver/esp32/ili9XXX.py#L541-L545) for example.
In your driver framework, where do you set the SPI pins? modes? freq? color mode? etc.

It's an other layer. We are just talking about it with @codewitch-honey-crisis. The ILI9341 display "class" should have any abstract interface to connect to the hardware. In the simplest form it means a `send_image` and a `send_command` callback. As I wrote above in this comment we can add direct support for some platforms as a next step.

codewitch-honey-crisis, 2022-10-02:
> Why isn't is enough to heave and abstract "channel" and the user can use any bus inside?

I *think* it can be. It's a lot of thinking around corners to try to adapt this to concrete platforms in my head, but earlier I think I was misunderstanding some of your code. It's probably okay.

> It's basically supporting some selected (popular) platform (ESP, STM, NXP, ?) and have very lean an targeted SRAM, SPI, I2C, DMA and LCD drivers for them. It might work. And it's modular, so we can add this layer at any time.

I agree with that. I think this is the right track.

Sorry if I bugged you all about this for nothing. I was just worried because the ESP32S3 ESP-IDF LCD framework threw a wrench in so many other libraries.

codewitch-honey-crisis, 2022-10-02:
As an aside, this isn't a huge deal, because it's not even supported by things like the S3's LCD framework, but one thing that surprised me that your driver interface doesn't have (nor need) is a fast fill mechanism where you take a single color value and repeat it width*height times across a rectangular window.

The reason it surprised me was 
A) There's some significant optimizations that can be had in certain situations just repeating the same word on a bus over and over
B) It doesn't require a memory buffer to update a region of the display.

TFT_eSPI, and htcw_gfx both have it. I'm not necessarily saying you should, as many platforms don't have it. It just kind of shocked me at first because you still get really good framerates.

kisvegabor, 2022-10-02:
Great, we are on the same page. Thanks for the discussion, now I feel my head cleaner :relaxed: 

> As an aside, this isn't a huge deal, because it's not even supported by things like the S3's LCD framework, but one thing that surprised me that your driver interface doesn't have (nor need) is a [...]

It's a little bit different in LVGL. We have a [complex interface for GPUs](https://docs.lvgl.io/master/porting/gpu.html). E.g. STM and NXP are already supported, but the whole SW rendering engine can replaced by SDL render which will use the PC's or MPU's video card. So the  "fast fill" can hook in on rendering level. In the end LVGL only passes the rendered image to the driver.

> There's some significant optimizations that can be had in certain situations just repeating the same word on a bus over and over

I think it works only for 16 bit parallel port without DMA. In this case you can do this
```c
for(i = 0; i < len; i++) {
  if(*color != *(color + 1)) {  //Very cheap `if` which can also acts as a delay after  wr_strobe
     update_data_pins();
  }
  wr_strobe();
}
```

Besides probably it takes much more performance the test where there will be mono colored areas in the end than flushing the colors one by one.  

> It doesn't require a memory buffer to update a region of the display.

But we have a buffer anyway. :slightly_smiling_face:

codewitch-honey-crisis, 2022-10-02:
You're right, in most cases it only works on 16-bit, but it can work on 8-bit for certain color values - black/white, etc where both of the bytes are the same, or on 8-bit grayscale oled displays for example. That's all just theoretical, of course. As I said, you get great framerates as it is, it was more just a curiosity for me when I encountered it.

kisvegabor, 2022-10-19:
I made some progress with the new driver architecture and hope I can merge it in the weekend. 

I'll comment about it in https://github.com/lvgl/lvgl/issues/2720

Until that I close this issue. However, if you have any other ideas or questions feel free to comment here and I'll reopen it.

closed 2022-10-19 by kisvegabor