lvgl/lvgl #3708: Annoying warning stemming from the header file

tripathiu, 2022-09-28:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

When using LVGL headers in my code, I get this warning...

```
In file included from ../lvgl/src/core/lv_obj.h:140,
                 from ../lvgl/lvgl.h:34,
                 from ../MyProject/File.hpp:3,
                 from ../MyProject/File.cpp:2:
../lvgl/src/core/lv_obj_style.h: In function 'void lv_obj_remove_style_all(_lv_obj_t*)':
../lvgl/src/core/lv_obj_style.h:94:48: warning: bitwise operation between different enumeration types '<unnamed enum>' and '<unn
amed enum>' is deprecated [-Wdeprecated-enum-enum-conversion]
   94 |     lv_obj_remove_style(obj, NULL, LV_PART_ANY | LV_STATE_ANY);
      |                                    ~~~~~~~~~~~~^~~~~~~~~~~~~~
```

It is a CMake/C++20 code base, which, I guess, is the reason for this warning. While I have suppressed warnings for the LVGL target using CMake, I still get it from all the files that include `lvgl.h`. Therefore, for every file that include `lvgl.h`, a new warning comes up. 

The source of the warning is from the following function:

```c
// lvgl/src/core/lv_obj_style.h:94

static inline void lv_obj_remove_style_all(struct _lv_obj_t * obj)
{
    lv_obj_remove_style(obj, NULL, LV_PART_ANY | LV_STATE_ANY);
}
```

Moving the implementation to a `.c` file would probably fix the issue but I am not sure of the implications. 

### To Reproduce
```c
#include "lvgl.h"
```

kisvegabor, 2022-09-28:
Only `lv_obj_remove_style_all` causes this warning? If so we can move it into C file as it's not performance critical.

tripathiu, 2022-09-29:
Hi @kisvegabor, yes, only this function causes warnings.

PR #3709 (2022-09-29): fix(lv_obj_style): move lv_obj_remove_style_all implementation to c file
### Description of the feature or fix

The PR fixes #3708.

### Checkpoints
- [x] Follow the [styling guide](https://github.com/lvgl/lvgl/blob/master/docs/CODING_STYLE.md)
- [x] Run `code-format.py` from the `scripts` folder. [astyle](http://astyle.sourceforge.net/install.html) needs to be installed.
- [] Update the documentation
closed 2022-09-29 by kisvegabor