lvgl/lvgl #3711: lvgl.h not updated to 8.3.2

elgerg, 2022-09-29:
Hi,

I just updated to 8.3.2 but noticed that the version number hasnt been updated.

Made me think I was referencing an older version..

Thanks
Alex

tripathiu, 2022-10-01:
Hello, where specifically can we see the version number?

kisvegabor, 2022-10-02:
I really missed to update version number on one place and already fixed it with https://github.com/lvgl/lvgl/commit/c8bee40410da4945d0267a4f4af7fa189bc33ed1 ... :sob: 

But to be honest, I have no good idea how to fix it after the release.
 
cc @embeddedt

embeddedt, 2022-10-03:
I think the cleanest solution is to release 8.3.3 with no other changes.

kisvegabor, 2022-10-05:
I have no better idea either. I'll do it tomorrow.

kisvegabor, 2022-10-07:
I released v8.3.3 so I close this issue.

closed 2022-10-07 by kisvegabor