lvgl/lvgl #3714: Troubles with STM32 DMA2D acceleration

tdjastrzebski, 2022-09-30:
Yesterday I have tried enabling STM32 DMA2D acceleration with D-cache enabled and MPU disabled.  
The results were not great, lots of 'garbage' on the screen, typically resulting from D-cache problems. Problems were clearly visible only with `lv_demo_music`. I reviewed [lv_gpu_stm32_dma2d.c](https://github.com/lvgl/lvgl/blob/965552e4465772dd3c12892216412fc01e651f6e/src/draw/stm32_dma2d/lv_gpu_stm32_dma2d.c) file and some question marks popped up. But first some background. I think the problem to be solved is that DMA (including DMA2D) does not clean (flush) the source buffer before starting the transfer, so dirty reads may happen. It, however, properly flushes whatever data it reads to the destination buffer.
1. Why do we clean AND **invalidate** cache before transfer? I think just cleaning should be enough.
2. Why do we clen the entire memory cache? Cleaning only source buffer area should be sufficient.
3. Why do we clean/invalidate cache before filling the buffer with color in `lv_draw_stm32_dma2d_blend_fill()` method? There is no source buffer to be cleaned/flushed.
4. I can be wrong, but I think source buffer address must be 32b-aligned and data length must be multiple of cache line size (32b). Do we ensure that?  

What I suggest:
1. Using `CleanDCache_by_Addr_Aligned()` method (below) or similar.
2. Modifying `clean_dcache_cb` so it takes address and length args.  

After making the changes STM32 DMA2D acceleration on my STM32F769I-DISCO board works without glitches but I am still not certain about source buffer alignment.

```C
void CleanDCache_by_Addr_Aligned(uint32_t sourceAddress, uint32_t dsize) {
	uint32_t sourceAddressAligned = sourceAddress & ~(uint32_t)(32U - 1);
	uint32_t lengthAligned = dsize + (sourceAddress & (uint32_t)(32U - 1));
	lengthAligned += 32U - (lengthAligned % 32U);
	__DSB();
	while (lengthAligned > 0) {
		SCB->DCCMVAC = sourceAddressAligned;  // register accepts only 32byte aligned values, only bits 31..5 are valid
		sourceAddressAligned += 32U;
		lengthAligned -= 32U;
	};
	__DSB();
	__ISB();
}
```
[image]

kisvegabor, 2022-10-02:
Wow,  thin the invalidations remained in the GPU code remained there by accident from earlier tested and debugging. 

We usually invalidate the cache at [the beginning `flush_cb`](https://github.com/lvgl/lv_port_stm32f746_disco/blob/master/hal_stm_lvgl/tft/tft.c#L197). To be honest I didn't know about `CleanDCache_by_Addr_Aligned` but I will try it out.

tdjastrzebski, 2022-10-02:
I think it is different story, different stage and different cache.  
My understanding is that within `lv_draw_stm32_dma2d_blend_map()` function `src_buf` is blended with `dest_buf` and THEN, eventually, `dest_buf` is passed to `flush_cb`, but if `src_buf` is dirty, cleaning cache, even in the first line of `flush_cb`, will not help - it is too late because it is `src_buf`, not `dest_buf` which needs to be cleaned first to get clean results.

kisvegabor, 2022-10-06:
You are right, `src_buf` can be also in a cached region. However, in `invalidate_cache` it really seems useless. 

Using `CleanDCache_by_Addr_Aligned` in `lv_draw_stm32_dma2d_blend_map` on `src_buf` can make a difference in performance.

tdjastrzebski, 2022-10-06:
I guess, my point was that this part of code, if intended to be used, should be carefully revised.
I am quite convinced d-cache invalidation is not needed at all - only d-cache cleaning is since DMA, best to my knowledge, does not use d-cache. I think cleaning entire cache is unnecessary and can have adverse impact on the overall performance. The same is true for cache invalidation. Invalidating perfectly valid cache (entire cache, without a reason) will hit performance.

tdjastrzebski, 2022-10-07:
I have been thinking about it a bit more. Source buffer needs to be cleaned right before DMA transfer and destination buffer needs to be invalidated after the transfer.

kisvegabor, 2022-10-10:
Can you send a PR with the suggested changes to one of our STM projects? So that I can also test the updates.

tdjastrzebski, 2022-10-13:
I will look into this in two weeks

GLASS-FR, 2022-10-21:
Hi, please consider that STM32F7/Cortex-M7 D-Cache SW management need buffer to be 32 **bytes** aligned (as like cache line boundary) and not only to 32bits alignement DMA requirement.

Using work buffer in DTCRAM (full speed without cache) can be a good tradeof to simplify SW management of D-cache coherency. But space is very limited/not suitable for big picture/screen.
So maybe be you willl want use SRAM1/SRAM2 or external SDRAM...

So :
1. **After CPU write access** to buffer (in internal SRAM1 or SRAM2 or external SDRAM write back cached),
**and before start DMA READ** from this buffer it's necessary, sufficient and more efficient  to use SCB_CleanDCache_by_Addr(addr, size).
2. BEFORE DMA write access to a buffer, the targeted address MUST be invalidated (to be sure that dirty bit is not already set, after that, during DMA the SW (CPU) **must avoid access** to the same buffer (to avoid partially DMA filled target to be placed in cache). **To be sure (more safe) a 2nd invalide can be used just at the end of DMA**.
3. SCB_CLeanDCache_by_addr() or for SCB_InvalideDCache_by_addr() MUST be called with "address" parameter value aligned to 32-byte boundary of D-cache line.
4. be carefull with STM32/M7 D-cache errata if trying write-throuht instead of write-bak !
5.  also CMSIS ARM CORTEX-M7 github work in progress on SCB_xxx() functions is not already ported to STM32 github (as I can see at this time ;>)) )

@kisvegabor  and @tdjastrzebski : nice job with this LVGL!

and Many thanks to PIRANHA user from STM32 FORUM that explained clearly cache coherency vs DMA here:
https://community.st.com/s/question/0D53W00001Z9K9TSAV/maintaining-cpu-data-cache-coherence-for-dma-buffers
I found also another valuable doc : "Managing Cache Coherency on Cortex-M7 Based MCUs" : Microchip TB3195..

tdjastrzebski, 2022-10-23:
@GLASS-FR Cache clean/invalidation is always 32-bit aligned since cache line size is 32 bytes. One of the best publications on this topic I am aware of is: [DMA is not working on STM32H7 devices](https://community.st.com/s/article/FAQ-DMA-is-not-working-on-STM32H7-devices)
According to [RM0410](https://www.st.com/resource/en/reference_manual/rm0410-stm32f76xxx-and-stm32f77xxx-advanced-armbased-32bit-mcus-stmicroelectronics.pdf), DMA2D memory addresses have to be 32-bit aligned IF pixel format is 32-bit. Otherwise, 16-bit alignment is required. However, the former requirement prevails.  

RM0410:
* 9.5.4 DMA2D foreground memory address register (DMA2D_FGMAR)
* 9.5.6 DMA2D background memory address register (DMA2D_BGMAR)
* 9.5.12 DMA2D foreground CLUT memory address register (DMA2D_FGCMAR)
* 9.5.16 DMA2D output memory address register (DMA2D_OMAR)

tdjastrzebski, 2022-10-25:
@kisvegabor I promise, very soon

tdjastrzebski, 2022-10-31:
I have started digging but there is more than cache maintenance issue. Anyway, I already am on it.

tdjastrzebski, 2022-11-03:
@kisvegabor I made an attempt to get it working but I am no longer convinced it makes sense. Here is why. For a test, in [lv_draw_stm32_dma2d_blend()](https://github.com/lvgl/lvgl/blob/965552e4465772dd3c12892216412fc01e651f6e/src/draw/stm32_dma2d/lv_gpu_stm32_dma2d.c#L112) method I commented out call to `lv_draw_stm32_dma2d_blend_map()`. To my surprise, music demo and demo widgets still rendered almost fine. It seems that LVGL heavily relies on alpha mask which disables DMA2D acceleration.  
Certainly, cache maintenance can be greatly improved, some calls can probably be eliminated (e.g. call to [lv_draw_sw_blend_basic()](https://github.com/lvgl/lvgl/blob/965552e4465772dd3c12892216412fc01e651f6e/src/draw/stm32_dma2d/lv_gpu_stm32_dma2d.c#L127) in line 127 or using `lv_area_move()`) but is it worth the effort? I am no longer so convinced.  
It seems like LVGL already went the direction where ST DMA2D acceleration cannot be used effectively.
What do you think?

Below cache maintenance functions I put together.

```C++
#define CACHE_ROW_SIZE 32U // cache row size in Bytes
#define PIXELS_PER_CACHE_ROW (CACHE_ROW_SIZE / LV_IMG_PX_SIZE_ALPHA_BYTE)

static void invalidate_cache(uint32_t address, lv_coord_t offset, lv_coord_t width, lv_coord_t height)
{
#if defined (__DCACHE_PRESENT) && (__DCACHE_PRESENT == 1U)
    uint32_t stride = LV_IMG_PX_SIZE_ALPHA_BYTE * (width + offset);
    uint16_t lcrc = (width / PIXELS_PER_CACHE_ROW) + 1; // line cache row count
    __DSB();

    while (height > 0) {
        height--;
        uint32_t a = address + height * stride;
        SCB->DCCMVAC = a; // clean the first line cache row first

        for (uint16_t r = 0; r < lcrc; r++) {
            SCB->DCIMVAC = a;
            a += CACHE_ROW_SIZE;
        }
        
        SCB->DCCMVAC = a; // clean the last line cache row first
        SCB->DCIMVAC = a;
    };

    __DSB();
    __ISB();
#endif
}

static void clean_cache(uint32_t address, lv_coord_t offset, lv_coord_t width, lv_coord_t height)
{
#if defined (__DCACHE_PRESENT) && (__DCACHE_PRESENT == 1U)
    uint32_t stride = LV_IMG_PX_SIZE_ALPHA_BYTE * (width + offset);
    uint16_t lcrc = (width / PIXELS_PER_CACHE_ROW) + 2; // line cache row count

    __DSB();
    
    while (height > 0) {
        height--;
        uint32_t a = address + height * stride;
        
        for (uint16_t r = 0; r < lcrc; r++) {
            SCB->DCCMVAC = a;
            a += CACHE_ROW_SIZE;
        }
    };

    __DSB();
    __ISB();
#endif
}
```

tdjastrzebski, 2022-11-04:
@GLASS-FR Do you think the above code is sufficient? I do not consider myself an expert

tdjastrzebski, 2022-11-04:
@kisvegabor ..unless **alpha mask** can be applied using hardware acceleration too.  
Maybe I need more research and code comments are rather scant.

kisvegabor, 2022-11-04:
With 16 bit color depth LVGL uses a non standard color format for ARGB images. Basically it's ARGB8565. I've never seen any GPU (or system in general) supporting this format but it allows using ARGB images with 16 bit color depth too. 

From the ARGB formats DMA2D supports only ARGB8888. So with with 32 bit color depth ARGB images could be supported by LVGL and DMA2D. Now it's still a `TODO`  in `lv_draw_stm32_dma2d_img_decoded`. It bypasses the image decoding of LVGL and allows processing the images directly. 

DMA2D has a pixel format conversation feature too, but I'm not sure if it can blend an ARGB8888 image to an RGB565 target. If it can we can use ARGB8888 source data for 16 bit color depth too.

tdjastrzebski, 2022-11-06:
@kisvegabor I have done a bit more research today. It appears that when using ARGB 8888 32b color in `lv_draw_stm32_dma2d_blend()` we have:  
* draw context (canvas?) containing ARGB destination buffer
* sw_blend_dsc containing ARGB source buffer (what does `sw` stand for?)
* 8-bit mask source buffer
* source `opa` constant 8-bit value

The above makes four alpha values to deal with, unless the first two should always be ignored. That would mean that internal buffers are not ARGB888. They actually are RGB888 but use 32 bits since 32-bit values are easier to deal with than 24-bit values. 
The same is true for the output buffer rendered to the actual hardware. After all, at this final stage alpha channel is probably never needed.  
Please comment/confirm.

kisvegabor, 2022-11-07:
> However, ARGB8565 is not supported by STM32F769. I think only new ST SPC58 automotive chips support it.

Nice to see that my genius color format is being adopted in the industry :laughing: 

> The difficulty I am having is getting full understanding of how low-level drawing is expected to work.

I can explain it but it doesn't really matter. If you add and "STM image draw callback" to draw the images you can do it as you wish. It doesn't matter how SW render handles the images with alpha channel and how it applies other masks. 

Just a slight modification: you should use `draw_ctx->draw_img` instead of `draw_img_decoded` to get the raw images as they are. 

> I am showing that A8 format is supported

I've tried to handle A8 blending (mask with a color) with DMA2D for letter drawing but was terribly slow. :slightly_frowning_face: 

> what does sw stand for?

SoftWare render

>  They actually are RGB888 but use 32 bits since 32-bit values are easier to deal with than 24-bit values.

Yes, actually it's XRGB8888 + a separate mask

> After all, at this final stage alpha channel is probably never needed.

E.g. for OSD menus (when the UI overlays and other layer) the final frame buffer has alpha channel too. 

---

I think the gist is to blend ARGB888 images to and RGB565 buffer a callback in `draw_ctx->draw_img` can be used.

kisvegabor, 2022-11-08:
> I see! Then in order use ST DMA2D hardware acceleration and alpha-blend foreground/background, XRGB8888+A8 mask has to be converted to true ARGB8888 first!
Probably two-step process is required, which, of course, performance-wise, is not going to perform as well as one-step process would.

Not really, you can create ARGB8888 "C array images" with LVGL's image converter: https://lvgl.io/tools/imageconverter
Just select the `TRUE_COLOR_ALPHA` format and the ARGB8888 image will on the bottom of the file guarded with `#if LV_COLOR_DEPTH == 32`. So it's not ready to use, but you can remove the other color formats and that's it.

> If source opa constant 8-bit value has to be applied as well, simple alpha-blending would require a three-step process.

I believe there a setting in DMA2D to set the overall opacity of the source image.

tdjastrzebski, 2022-11-08:
I am talking [lv_draw_stm32_dma2d_blend()](https://github.com/lvgl/lvgl/blob/965552e4465772dd3c12892216412fc01e651f6e/src/draw/stm32_dma2d/lv_gpu_stm32_dma2d.c#L112) and [lv_draw_stm32_dma2d_blend_map()](https://github.com/lvgl/lvgl/blob/965552e4465772dd3c12892216412fc01e651f6e/src/draw/stm32_dma2d/lv_gpu_stm32_dma2d.c#L112) functions in [lv_gpu_stm32_dma2d.c](https://github.com/lvgl/lvgl/blob/965552e4465772dd3c12892216412fc01e651f6e/src/draw/stm32_dma2d/lv_gpu_stm32_dma2d.c) file.  
The current code seems to be missing the necessary XRGB -> ARGB conversion before DMA2D transfer and likely is defective.  
It does not seem that constant alpha value can be applied - refer to [RM0410](https://www.st.com/resource/en/reference_manual/rm0410-stm32f76xxx-and-stm32f77xxx-advanced-armbased-32bit-mcus-stmicroelectronics.pdf), page 290, Memory-to-memory with PFC and blending.
Yes, this or the other way I can create ARGB8888 programmatically, but we are talking about hardware acceleration here.

kisvegabor, 2022-11-09:
> I am talking [lv_draw_stm32_dma2d_blend()](https://github.com/lvgl/lvgl/blob/965552e4465772dd3c12892216412fc01e651f6e/src/draw/stm32_dma2d/lv_gpu_stm32_dma2d.c#L112) and [...]

The point is to NOT use `blend()` if DMA2D knows a shorter path. `blend` is the very last operation when the SW render already calculated and prepared everything and it only needs to be blended to the frame buffer. If a GPU can handle the drawing in an earlier stage (i.e. as soon as it sees an image to be drawn) we can bypass to whole SW render path. And this we can use `draw_ctx->draw_img` for.

> It does not seem that constant alpha value can be applied by DMA2D - refer to [RM0410](https://www.st.com/resource/en/reference_manual/rm0410-stm32f76xxx-and-stm32f77xxx-advanced-armbased-32bit-mcus-stmicroelectronics.pdf), page 290, Memory-to-memory with PFC and blending.

In `Table 39. Alpha mode configuration` isn't  `10     Replaced by original value multiplied by the value in DMA2D_xxPFCCR / 255` what we need here? 

> Yes, this or the other way I can create ARGB8888 image map programmatically, but we are talking about hardware acceleration in the context of this, aforementioned LVGL part, not some image file conversion.

Usually images are stored in the flash anyway in a "pre-built" format (i.e. PNG converted to raw C pixel array). Why we can't covert the image to ARGB8888 then to let  DMA2D work with that directly?

tdjastrzebski, 2022-11-09:
All right, I got you. Apparently, I have missed this FPC.AM registry setting.
I think the ground is set. It seems promising and doable.  
Here is what suggest: create empty functions and I will try to fill them in.
I made a new fork at https://github.com/tdjastrzebski/lvgl and gave you access.

kisvegabor, 2022-11-10:
Thanks, I accepted the invitation. I can work on it on the weekend.

kisvegabor, 2022-11-16:
Sorry for the late reply. I've just added pushed a commit to your repo (release/v8.3 branch): https://github.com/tdjastrzebski/lvgl/commit/c658c5b931fff34d1958edbd17f0486a00e38490

It was required to add a few minor changed to LVGL too. 

As an example I added  a simple fill with random color in place of ARGB888 blend. 

Is it something you can use?

tdjastrzebski, 2022-11-17:
Also, the build fails when linking:  
`undefined reference to 'img_demo_widgets_avatar_argb8888'`

kisvegabor, 2022-11-17:
> Thanks, I will look more into it. One thing which immediately caught my attention: LV_IMG_CF_RGBA8888.

`LV_IMG_CF_RGBA8888` is really incorrect. The actual order is 
- byte1: Blue
- byte2: Green
- byte3: Red
- byte4: Alpha

> Also, the build fails when linking:

Sorry I forgot to add it.

tdjastrzebski, 2022-11-17:
RGBA bit order is not a good news. Bits need to be reordered before hardware processing, but it is not a show stopper.

kisvegabor, 2022-11-19:
Actually, I think DMA2D is compatible with LVGL color format. I'm pretty sure I've already tried it out earlier. 

I think the root of the confusion is that B+G+R+A byte-by-byte, will be ARGB if you write it as a 32 bit integer as the MCU is little endian. See [here](https://en.wikipedia.org/wiki/RGBA_color_model#RGBA8888).

tdjastrzebski, 2022-11-19:
DMA2D expects bytes in the order depictured below (RM0410, p 284). I do not think the order of R, G & B actually matter since all three colors are processed the same way, however, **A** must come last in the highest byte - MSB by STM32 little-endian convention.  

Do I correctly understand that the rest of LVGL names like `BGRA8888` are based on big-endian convention? If so, `LV_IMG_CF_BGRA8888` would correctly translate to STM32 ARGB8888 byte order.

[image]

tdjastrzebski, 2022-11-23:
I could not figure out for some time what was happening. DMA2D was used to transfer data in `flush_cb()`. This way DMA2D became cross-thread shared resource.  
It seems that although `flush_cb()` takes `lv_area_t* area` argument, the memory to be transferred is always (?) continuous. In another words, area width=screen width. If that is the case, maybe `flush_cb()` should accept just lines count and y offset? That would encourage regular DMA use in `flush_cb()` vs DMA2D.  
That is just a thought, DMA2D use must be thread-safe in any case.
What linter settings are used for LVGL? Is there any official VS Code `.editorconfig` available?

kisvegabor, 2022-11-25:
> I could not figure out for some time what was the source of the problem in my test code. DMA2D was used to transfer data in flush_cb(). This way DMA2D became cross-thread shared resource but access to this resource was not properly protected.

LVGL is single threaded, so all DMA2D calls are "sequential". 

> It seems that although flush_cb() takes lv_area_t* area argument, the memory to be transferred is always (?) continuous. In another words, area width=screen width, no memory gaps between image lines. If this is the case, maybe flush_cb() should accept just lines count and y offset? 

It `color_p` is always continuous and `area` tells its dimensions.  However it's not sure that `area width == screen width`. E.g. is a button is pressed on the middle of the screen only that buttons area will be redrawn. (assuming `disp_drv.full_refresh=0`).

> That would encourage regular DMA use in flush_cb() vs DMA2D. This could also help to eliminate DMA2D cross-thread sharing. 

For larger areas I usually use DMA to updat ethe frame buffer line-by-line. See [here](https://github.com/lvgl/lv_port_stm32f429_disco/blob/master/hal_stm_lvgl/tft/tft.c#L183) and [here](https://github.com/lvgl/lv_port_stm32f429_disco/blob/master/hal_stm_lvgl/tft/tft.c#L720).

> That is just a thought, DMA2D use must be thread-safe. That is, it must be taken into account that it may run, also in interrupt mode, started somewhere else, by some other thread.

Can't you just simply `while(DMA2D->CR & DMA2D_CR_START_Msk)` before any operation? It doesn't matter where the operation started, we just need to be sure we won't start a new operation while the previous is running.  

> 
I could not figure out for some time what was the source of the problem in my test code. DMA2D was used to transfer data in flush_cb(). This way DMA2D became cross-thread shared resource but access to this resource was not properly protected.
It seems that although flush_cb() takes lv_area_t* area argument, the memory to be transferred is always (?) continuous. In another words, area width=screen width, no memory gaps between image lines. If this is the case, maybe flush_cb() should accept just lines count and y offset? That would encourage regular DMA use in flush_cb() vs DMA2D. This could also help to eliminate DMA2D cross-thread sharing.
That is just a thought, DMA2D use must be thread-safe. That is, it must be taken into account that it may run, also in interrupt mode, started somewhere else, by some other thread. By thread I do not necessarily mean some RTOS thread. Just using tick timer with interrupt handler effectively creates a new thread/task.

> What linter settings are used for LVGL? Is there any official VS Code .editorconfig file available?

What do you mean by linter in case of LVGL? Usually I use Eclipse which parses the source code and underlines the mistyped functions and variable names automatically and handles autocomplete too.

tdjastrzebski, 2022-11-25:
I did not mean to suggest that LVGL was multi-thread, but that it may be, and usually is, used in multi-thread/task environment just because it relies on tick timer interrupt handler.
I need to do more research to come up with correct solution but, as strange as it sounds, it seems that just testing for CR reg START flag is not enough.  

I do not use Eclipse at all, I usually code in VS Code. What I meant to ask was what coding style/formatting has been adopted by LVGL project? E.g. "Visual Studio", "LLVM", "Google", "WebKit".

tdjastrzebski, 2022-11-25:
If flush_cb() may want to draw just a button in the middle of the screen (which I was not sure about) then, of course, DMA2D being aware of drawn image size and position is far better solution than DMA.

kisvegabor, 2022-11-28:
> I did not mean to suggest that LVGL was multi-thread, but that it may be, and usually is, used in multi-thread/task environment just because it relies on tick timer interrupt handler.

However GPU calls are always soming from LVGL's single threaded environment. So it shouldn't matter is you have tasks and interrupts next to it. All GPU calls will come from  the same source sequentially.  Or am I missing something?

> I do not use Eclipse at all, I usually code in VS Code. What I meant to ask was what coding style/formatting has been adopted by LVGL project? E.g. "Visual Studio", "LLVM", "Google", "WebKit", something else?

To be IDE independent we have `code_format.py` in the script folder which uses  `astyle` for code formatting.

> If flush_cb() may want to draw just a button in the middle of the screen (which I was not sure about) then, of course, DMA2D being aware of drawn image size and position is far better solution than DMA.

For large areas it shouldn't be so much better. E.g. for a 200x200 area it means 200 extra interrupt and a few lines of code. If it takes 1 clock cycle to copy 1 pixel, then it needs 40,000 instructions to copy the 200x200 area. If each interrupt takes 50 instructions then it's an other 10,000 instructions. So it's 25% overhead. However we kept DMA2D free for the next rendering cycle. The first item to draw is usually the background rectangle which could be delegated to DMA2D. But if it's busy with flushing we need to either wait for it or render the background from software. 

> Am I correct saying that use of all-purpose DMA practically demands disabling partial redraw?

Nope, if you restart DMA with the next line in its complete interrupt.

tdjastrzebski, 2022-11-29:
What I was afraid of was that timer tasks may not finish before a new one starts. This way it could have happened that two tasks were active the same time. But anyway, probably I was just looking at this code for too long to spot much different obvious bug - not all the important DMA2D settings were being set, some values were carried over from the previous transfer(s) and this was causing the problem.  
I have first preliminary yet promising version successfully tested [here](https://github.com/tdjastrzebski/lvgl/blob/test1/src/draw/stm32_dma2d/lv_gpu_stm32_dma2d.c). Note, it is `test1` branch.  
* See my comments in lines 153 and 154. It seems that `dest_stride` and `dest_area` arguments of `lv_draw_stm32_dma2d_buffer_copy()` are redundant.  
* Line 175 - probably at least zoom and angle needs to be checked before hardware blending is used
* Source and designation DMA buffers must be 32-byte (not bit) aligned - **both the beginning and the ending**. Otherwise, DMA, which operated on 32-byte blocks (called rows), may clean/invalidate memory bytes which shouldn't or do not need to be touched.

tdjastrzebski, 2022-11-29:
Btw, astyle does not appear to be an excellent tool. This code fragment has been left like unfixed
```
    DMA2D->OPFCCR = LV_DMA2D_COLOR_FORMAT;
    DMA2D->OMAR    = (uint32_t)(dst_buf + (dst_stride * dst_area->y1) + dst_area->x1);
    DMA2D->OOR     = dst_stride - draw_width;
    DMA2D->OCOLR = 0;
```

kisvegabor, 2022-11-30:
> See my comments in lines 153 and 154. It seems that dest_stride and dest_area arguments of lv_draw_stm32_dma2d_buffer_copy() are redundant.

Let's say there is a 320x240 screen and in `lv_draw_stm32_dma2d_buffer_copy` we want to copy to its (20;20);(100;50) `dest_area`. However `dest_stride` will be 320 to tell how many pixels to step to get to the next line.  

> Line 175 - probably at least zoom and angle needs to be checked before hardware blending is used

True.

> Source and designation DMA buffers must be 32-byte (not bit) aligned - both the beginning and the ending. Otherwise, clean/invalidate which operated on 32-byte blocks (called rows), may touch memory bytes which shouldn't or do not need to be touched.

So the worst case is losing some performance by invalidating the cache of extra 31 byte at the beginning and the end?

>  This code fragment has been left like unfixed

Yes, `astyle` doesn't align assignment operators.

tdjastrzebski, 2022-11-30:
As far as cache clean/invalidation; it is not just performance. Although cleaning cache is relatively safe, it can lead to errors when cleaned bytes belong to some device memory area. In such case, correct, current values are overwritten with old, already cached values.  
Cache invalidation is even more dangerous as it replaces core L1 cache values with whatever is/was stored in RAM. It may be some variable previous value, stored next to unaligned buffer, which has not been written to RAM yet waiting in cache. This may easily cause, and often does, program to crash.

tdjastrzebski, 2022-11-30:
My current version seems to be handling DMA2D acceleration quite well, the "music" demo shows no visible glitches and runs smoothly. That does not mean it is bug free and everything is 'as good as it gets'. Now I would like to improve opacity handling. I am showing that [`fill_normal()`](https://github.com/tdjastrzebski/lvgl/blob/master/src/draw/sw/lv_draw_sw_blend.c#L170) function uses both `opa` and `mask` arguments. The same is true for [`map_normal()`](https://github.com/tdjastrzebski/lvgl/blob/master/src/draw/sw/lv_draw_sw_blend.c#L528) function.  
Is it desired to make a similar implementation using DMA2D? - if, of course, possible.

tdjastrzebski, 2022-12-01:
> all we need to do is placing it on 32 byte boundary and be sure its size is N x 32 byte
Correct, since cache uses 32-byte blocks (rows), otherwise we either do not clean/invalidate all the bytes which need to be cleaned/invalidated or clean/invalidate bytes outside of the given buffer range, which may lead to a disaster.

embeddedt, 2022-12-01:
> @embeddedt maybe it was the same issue with https://github.com/lvgl/lv_port_stm32f746_disco/commit/4f7dda9792996caf965625de02b3f72155f3d461. I traced it down to a code line in LVGL where it incorrectly evaluated an a & b or so. I didn't know how is it possible but @embeddedt found a fix.

Hmm, the framebuffer was placed at the very beginning of SDRAM in both cases (`0xC0000000` and `0x60000000`) so the alignment should have been the same, unless I overlooked something.

kisvegabor, 2022-12-02:
I'm also not sure if it's related and don't remember the details. Anyway, it's good to keep in mind that cache invalidation/cleaning can mess up things.

tdjastrzebski, 2022-12-10:
I made some good progress lately but there is one glitch I am having trouble getting rid of. Mask bytes passed to `lv_draw_stm32_dma2d_blend_paint()` method sometimes are incorrect, it seems is like it is a completely wrong memory area.
Maybe it is actually caused by cache invalidation? What would it take to make sure used buffers, especially those modified, used as dma2d output, are 32-bit aligned? - **both start and length**.

Could you please take a look at [lv_gpu_stm32_dma2d.c](https://github.com/tdjastrzebski/lvgl/blob/test1/src/draw/stm32_dma2d/lv_gpu_stm32_dma2d.c), lines 163-202?
Does this if-else leader make a sense?

kisvegabor, 2022-12-12:
> Maybe it is actually caused by cache invalidation? What would it take to make sure used buffers, especially those modified, used as dma2d output, are 32-bit aligned? - both start and length.

Mask is always in internal RAM so it shouldn't cached.

> I will try to handle the case when mask != NULL && dsc->src_buf != NULL as well, but it as a specific case where two input buffers are in play. Handling it by DMA2D would require copying alpha bits from mask to dsc->src_buf first, thus modifying source buffer. Do you think this is allowable?

What could happen with RGRB565 + alpha format?

> Also, lv_draw_stm32_dma2d_buffer_copy() method does not seem to be ever called.

It's set in `dma2d_draw_ctx->base_draw.buffer_copy = lv_draw_stm32_dma2d_buffer_copy;`.  So it can be called by the users if needed from the `draw_ctx`.

kisvegabor, 2022-12-13:
> So, the question what would it take to make sure used buffers, especially those modified, used as dma2d output, are 32-bit aligned still stands. It may the biggest single obstacle I am facing right now.

Can you test if really the misalignment is the issue? E.g. please manually allocate a static, correctly aligned buffer for mask in DMA2D and copy the original mask there. I mean it only for testing purposes.

> I took some stats, mask != NULL && dsc->src_buf != NULL case currently performs somewhat slower than pure SW implementation (lv_draw_sw_blend_basic()) since alpha bytes need to be merged with RGB bytes first, but I may be able to optimize it better. In any case, since most work is done by hardware DMA2D it is still worth it.

For ARGB8888 we can use by pass the SW render at all as we have discussed it at the beginning.

tdjastrzebski, 2022-12-13:
As far as ARGB8888 - correct, but it seems that dma2d may still be a better option, even if current non-optimized code is slightly slower. I mean, it is a god news.  
I will do more tests to detect misalignment, but even if it does not happen in my setup, proper measures must be taken.

kisvegabor, 2022-12-14:
> As far as ARGB8888 - correct, but it seems that dma2d may still be a better option, even if current non-optimized code is slightly slower. I mean, it is a good news

I mean ideally you don't need to merge the mask and the XRGB8888 image but use the ARGB8888 image before LVGL decomposes it.

> I will do more tests to detect misalignment, but even if it does not happen in my setup, proper measures must be taken.

I agree, and we can solve it by having an `lv_malloc_aligned()` function. We had this in v7 but somehow it disappeared in v8. :confused:

tdjastrzebski, 2022-12-14:
I did more tests. Unfortunately, it  happens that dma2d output buffer is misaligned; not the start address but length, assert in line 117 occasionally fails.  
What is worst, it appears that `wait_for_finish`/`lv_gpu_stm32_dma2d_wait_cb()` is not always called. As a result, output buffer memory cache is not invalidated and ugly artefacts start to show up.  
Enabling line 212 and waiting for dma2d to finish right after it starts solves the problem, but it is not a preferred solution.  
Please advise.  
Other than that, the version I just pushed ([lv_gpu_stm32_dma2d.c](https://github.com/tdjastrzebski/lvgl/blob/test1/src/draw/stm32_dma2d/lv_gpu_stm32_dma2d.c)) works smoothly and noticeably much faster than software rendering.

kisvegabor, 2022-12-18:
>  If buffer length is a not multiple of 32-bytes we risk invalidating too few bytes (->artifacts) or too many, from outside of the buffer area (->crashes).

So let's say there is a 800x480 16 bit screen, and we redraw e.g. a (13;27) (24; 34) (some random) area which might be not aligned. In this case can't we take the larger area as it will be still inside the original buffer.

>  As a result, output buffer memory cache is not invalidated (before flush_cb() kicks in?) and ugly artefacts start to show up.

`wait_for_finish` should be called here:
https://github.com/lvgl/lvgl/blob/47c8f8f9822f4c0c0ffbe2f12b380bddefcec475/src/core/lv_refr.c#L1189

> It works smoothly and noticeably much faster than software rendering. 

It's amazing! Especially as the current implementation doesn't added a noticeable performance boost.

tdjastrzebski, 2022-12-19:
As long as drawn area bytes stay within the original buffer and the buffer itself is 32-byte aligned no crashes should happen. However, if area adjacent to the area being drawn holds meaningful data and this area was drawn sw-fashion, some L1-cached bytes which were not flushed yet may get invalidated and bitmap may get damaged. The remedy for that is to call cache-clean on the output buffer as well, before dma2d transfer starts, to sync memory content with L1 cache.

kisvegabor, 2022-12-20:
It seems to me the only solution would be prevent SW rendering run closer than 32 byte to a DMA rendered area, right?

> The remedy for that is to call cache-clean on the output buffer, after partial sw render to sync L1 cache with memory before dma2d transfer starts again.

Could you measure its performance impact?

tdjastrzebski, 2022-12-26:
I do not see how preventing SW rendering close to hardware rendering would be possible since both methods use the same output buffer and often adjacent areas. Cleaning output memory cache helps but it adds extra delay (2-3µs).    

I implemented and initially tested 16-bit color support using all four available demos. It looks good.    
I tried to implement LV_COLOR_16_SWAP as well since red/blue bits swap is supported by DMA2D. I got some, yet limited success. It is not quite clear where bits have to be swapped and where they do not.  

Why `lv_draw_stm32_dma2d_buffer_copy()` has `dest_stride` argument, which makes perfect sense, while `lv_draw_stm32_dma2d_blend()` does not need it? I am not certain that `dest_stride` within `lv_draw_stm32_dma2d_blend()` is obtained correctly.

I have added three `dwt_*` functions to measure elapsed time with µs resolution. They do not belong to dmad2 rendering but require CortexM7 core.  

I think the [latest vision](https://github.com/tdjastrzebski/lvgl/blob/test1/src/draw/stm32_dma2d/lv_gpu_stm32_dma2d.c) is good enough to be merged with 8.3 main branch.

Two biggest unsolved chellenges:  
* output buffer misalignments - temp partially solved by extra clean cache called before dma2d start  
* input buffer (especially mask) probably being modified while waiting for dma2d to complete - temp solved by waiting for dma2d to finish right after it starts

kisvegabor, 2023-01-02:
> I implemented and tested 16-bit color support using all four available demos. It looks good.

Approximately how mush faster was it with DMA2D?

> I think SW rendering does not do a perfect job either. It is not quite clear to me where bits have to be swapped and where they do not.

In which cases does SW rendering fail? Note that it's not R-G swapping but  low and high byte swapping:
```
RRRRRGGG GGGBBBBB
GGGBBBBB RRRRRGGG 
```

> I think that the [latest version](https://github.com/tdjastrzebski/lvgl/blob/test1/src/draw/stm32_dma2d/lv_gpu_stm32_dma2d.c) is good enough to be merged with 8.3 main branch, although it can and should be further improved.

If it's stable enough we can merge the updates to `release/v8.3`.

> output area misaligned, which is a problem only when both sw and dma2d writing to the same output buffer is used and probably does not have a good solution, other than extra cache clean, at least on line edges, unless implementing rounder_cb() is sufficient - I do not know

I was also thinking about invalidating/cleaning only the line edges. `rounder_cb` can not solve the issue as it only rounds the coordinates of the area to redraw. Regardless to the size of this area objects can be inside it with any size.

> Input buffer (especially mask) probably being modified while waiting for dma2d to complete 

Oh, yes, it really can be a problem. In v9 there will be a much better rendering architecture. Render engines (e.g. DMA2D, or SW render) will get full `draw_task`s (e.g. draw this rectangle with this radius, gradient, etc). The render engines can decide if they can take a certain task or not (i.e. whether they are capable of drawing it). So it will be up to the render engine to handle masks as it wants. (The current mask engine can be used by any renderer)

> It does not seem current LVGL architecture allows for using DMA2D interrupts, which probably would be much more efficient than while((DMA2D->CR & DMA2D_CR_START) != 0U) waiting loops burning all the CPU cycles gained by hardware acceleration.

It will be also addressed in v9. We are adding more OS support to LVGL itself, so DMA2D will be able to yield while waiting.

> To do that, it needs to be known when flush_cb() is about to be called (pre_flush_cb()?) and there must be some way to suppress cache invalidation if DMA(2D) is used in flush_cb() - e.g. by calling dma2d_buffer_copy().

Can't we simply do it at the beginning of the `flush_cb`?

tdjastrzebski, 2023-01-03:
My mistake, I meant to say red/blue bits swap. In code it is described correctly.  

It is hard to say how much DMA2D is faster because it depends how speed is measured. Running all the four demos for approximately equal time I found out that 93.5% of the time (mask != null) and (dsc->src_buf == null). Consequently, `lv_draw_stm32_dma2d_blend_paint()` method used (e.g.) to paint anti-aliased fonts is the most critical for the overall performance. For larger areas it is 2-3x faster than sw, for smaller areas it can be slower. It would be good to implement set of representative tests.  
> Can't we simply do it at the beginning of the flush_cb?  

I think so.  
Yesterday I added one more dma2d function handling simple buffer copy.   
I suggest merging the current version with 8.3 as an experimental feature and let others contribute.  
It is good to hear that v9 will be more 'dma2d friendly' - it should allow for showing its full potential.

tdjastrzebski, 2023-01-04:
I could not get this PR right at the first attempt but it should be OK now.
For some reason `Micropython CI / Build unix port` check is failing with no detailed info, but 16 other succeed.

closed 2023-01-23 by tdjastrzebski

vahidajalluian, 2023-06-10:
@kisvegabor  @tdjastrzebski  Thank you for the efforts. First, I want to ask how this rework can be used? Documentation missing examples. Second, is it also compatible with f4 series more specifically f429 series?

kisvegabor, 2023-06-12:
To use it you just need to enable `LV_USE_GPU_STM32_DMA2D`.

It should work with F429 too.

tdjastrzebski, 2023-06-12:
@vahidajalluian I have one example available at https://github.com/tdjastrzebski/STM32F769I-DISCO-LVGL
F429 has no DCache so it should just work without any extra steps.

vahidajalluian, 2023-06-13:
@kisvegabor So how does it manage to flush the screen? I could not find any routines to be assigned while enabling LV_USE_GPU_STM32_DMA2D?

vahidajalluian, 2024-04-10:
> @vahidajalluian I have one example available at https://github.com/tdjastrzebski/STM32F769I-DISCO-LVGL F429 has no DCache so it should just work without any extra steps.

Thank you for your repo, It did help, If you let me I will distribute one cleaned project based on your work and idea with LVGL 8.4. I permit the library to start updating in dma2d finish callback. By which I mean
```
void DMA2D_IRQHandler(void)
{
	lv_disp_flush_ready(&disp_drv);
	HAL_DMA2D_IRQHandler(&hdma2d);
}
```

tdjastrzebski, 2024-04-10:
@vahidajalluian, No problem, I will be honored :)
Btw, there is room for improvements, e.g. LTDC buffer could use only 3 bytes per pixel (RGB instead of ARGB).

[5 more replies]

PR #3904 (2023-01-23): stm32 dma2d rework (8.3)
This PR is a major rework of the current LVGL STM32 DMA2D support. It adds more hardware acceleration, especially for bitmap-blended solid colors used (e.g.) to draw anti-aliased fonts. In 32-bit color mode all four types of rectangles are painted with dma2d. In 16-bit color three types are dma2d painted, including the type which is the most performance critical.
The feature has been fairly tested using LVGL demos with both 32 and 16 bit colors, but it should still be considered experimental.
Resolves #3714, see [this](https://github.com/lvgl/lvgl/issues/3714#issuecomm [... cut]