lvgl/lvgl #3720: Crash in 'test_arc.exe'

gvanem, 2022-10-04:
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

I just discovered LVGL and trying to build with a static version of SDL2 (built by `vcpkg install sdl2:x64-windows-static`).
I'm using MSVC v19 (and clang-cl) on Win-10.

I've enabled the `tests/` targets. But running the simple `test_arc.exe`, crashes on this line:
```c
void lv_draw_sdl_init_ctx(lv_disp_drv_t * disp_drv, lv_draw_ctx_t * draw_ctx)
{
...
draw_ctx_sdl->renderer = ((lv_draw_sdl_drv_param_t *) disp_drv->user_data)->renderer;
```

since `draw_ctx_sdl->renderer` is NULL. I fail to see where `draw_ctx_sdl->renderer` should be set.

But with this patch:
```diff
--- a/draw/sdl/lv_draw_sdl.c 2022-10-03 11:10:15
+++ b/draw/sdl/lv_draw_sdl.c 2022-10-04 12:21:07
@@ -75,6 +75,7 @@
     draw_ctx->layer_destroy = lv_draw_sdl_layer_destroy;
     draw_ctx->layer_instance_size = sizeof(lv_draw_sdl_layer_ctx_t);
     lv_draw_sdl_ctx_t * draw_ctx_sdl = (lv_draw_sdl_ctx_t *) draw_ctx;
-    draw_ctx_sdl->renderer = ((lv_draw_sdl_drv_param_t *) disp_drv->user_data)->renderer;
+    if (draw_ctx_sdl->renderer)
+       draw_ctx_sdl->renderer = ((lv_draw_sdl_drv_param_t *) disp_drv->user_data)->renderer;
     draw_ctx_sdl->internals = lv_malloc(sizeof(lv_draw_sdl_context_internals_t));
     lv_memzero(draw_ctx_sdl->internals, sizeof(lv_draw_sdl_context_internals_t));
```

I'm able to successfully run this test:
```
src/test_cases/test_arc.c:10:test_arc_creation_successfull:PASS
src/test_cases/test_arc.c:11:test_arc_should_truncate_to_max_range_when_new_value_exceeds_it:PASS
src/test_cases/test_arc.c:12:test_arc_should_truncate_to_min_range_when_new_value_is_inferior:PASS
src/test_cases/test_arc.c:13:test_arc_should_update_value_after_updating_range:PASS
src/test_cases/test_arc.c:14:test_arc_should_update_angles_when_changing_to_symmetrical_mode:PASS
src/test_cases/test_arc.c:15:test_arc_should_update_angles_when_changing_to_symmetrical_mode_value_more_than_middle_range:PASS
src/test_cases/test_arc.c:16:test_arc_angles_when_reversed:PASS
src/test_cases/test_arc.c:126:test_arc_click_area_with_adv_hittest:PASS

-----------------------
8 Tests 0 Failures 0 Ignored
OK
```

Someone please explain what's going on. Has it to do with the use of a static SDL2?

gvanem, 2022-10-04:
Well, both are NULL. Adding:
```c
printf ("draw_ctx_sdl->renderer: 0x%p, disp_drv->user_data: 0x%p\n", draw_ctx_sdl->renderer, disp_drv->user_data);
```
shows it: `draw_ctx_sdl->renderer: 0x0000000000000000, disp_drv->user_data: 0x0000000000000000`
How come?

kisvegabor, 2022-10-07:
cc @mariotaku

mariotaku, 2022-10-09:
I think the SDL backend was not properly initialized in this test case. Should I add this support, and will you introduce this to action workflows?

kisvegabor, 2022-10-11:
>  Should I add this support, and will you introduce this to action workflows?

I will merge the new driver architecture in a few weeks. I think it'd be better to port SDL backend to it and add workflow after that.  You can see the  current SDL driver is [here](https://github.com/lvgl/lvgl/blob/arch/drivers/src/dev/sdl/lv_sdl_window.c) to check what the idea is.

mariotaku, 2022-10-26:
@kisvegabor Sorry for the late reply. I kinda like the concept of creating SDL window via LVGL directly. Also the part it handles event.

closed 2022-11-19 by lvgl-bot