lvgl/lvgl #3724: Repeated lv_obj_add_style crashes on ESP32-S3 in _DoubleExceptionVector with corrupted stack

martaaay, 2022-10-05:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version - **Using latest LVGL v8.3 and latest ESP IDF v4.4**
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html) - **Cant reproduce**

### Describe the bug
On an ESP32-S3, I have a dedicated thread for the UI. I'm running LVGL v8.3 with vgl_esp32_drivers. I found if I called lv_obj_add_style rather rapidly (with a delay in between), my app will crash within seconds in _DoubleExceptionVector with, unfortunately, a corrupt stack. I've been trying to get a full stack trace, but haven't quite achieved that yet. Here's some pseudo code to illustrate:
```c
    while (1) {
        vTaskDelay(pdMS_TO_TICKS(screen_is_enabled ? 10 : 100));

        if (screen_is_enabled) {
            if (pdTRUE == xSemaphoreTake(xGuiSemaphore, portMAX_DELAY)) {
                lv_task_handler();
                ui_loop(pdTICKS_TO_MS(xTaskGetTickCount()));
                xSemaphoreGive(xGuiSemaphore);
            }
        }
```
```c
void ui_loop(uint64_t current_time_ms) {
    if (every_1_second_ms == UINT32_MAX || current_time_ms - every_1_second_ms > 1*1000) {
        lv_obj_add_style(_obj, &style_text_container_with_border, 0);
        
        every_1_second_ms = current_time_ms;
    }
```

I can speed up the crash from roughly 60 seconds to a few seconds by dropping the `1*1000`. I'm curious if this is a known issue that I haven't been able to find with google/github search. If not, I can try to really drill down to a simple test case that crashes reliably and quickly, as well as fight to get a proper stack trace.

### To Reproduce
(See above)

### Expected behavior
Not crashing :)

### Screenshots or video
n/a

embeddedt, 2022-10-05:
Hmm. Are you 100% sure that this is the only thread that calls LVGL APIs?

martaaay, 2022-10-05:
Ooo interesting. I'll do a deep crawl.

martaaay, 2022-10-06:
Aha! Made progress. This crasher occurs rapidly if you call lv_obj_add_style repeatedly on the same object. Once it gets to obj->style_cnt == 63 inside lv_obj_add_style, incrementing obj->style_cnt flips it to 0 since style_cnt is allotted 6 bits in the _lv_obj_t struct. When that happens, lv_mem_realloc is handed new_size == obj->style_cnt * sizeof(_lv_obj_style_t) == 0. 

The lv_mem_realloc code is the following:
```c
void * lv_mem_realloc(void * data_p, size_t new_size)
{
    MEM_TRACE("reallocating %p with %lu size", data_p, (unsigned long)new_size);
    if(new_size == 0) {
        MEM_TRACE("using zero_mem");
        lv_mem_free(data_p);
        return &zero_mem;
    }

    if(data_p == &zero_mem) return lv_mem_alloc(new_size);
```

On new_size == 0, lv_mem_realloc returns the address of zero_mem (a uint32_t), which is then assigned to obj->styles in lv_obj_add_style, which then writes all over the code / variables near zero_mem which leads to a panic. I think a good assert would be something like the following:

```c
void lv_obj_add_style(lv_obj_t * obj, lv_style_t * style, lv_style_selector_t selector)
{
    LV_ASSERT(obj->style_cnt < 63);
 
    trans_del(obj, selector, LV_STYLE_PROP_ANY, NULL);
```

So really, this is an error in my code by calling too many add styles to the same object. It was just super hard to debug by the way it crashed. Would you like me to submit a pull request with this?

embeddedt, 2022-10-06:
True, there really should be an assert there. Please send a PR. :slightly_smiling_face: 

(... what were you doing to need 64 styles?)

martaaay, 2022-10-06:
> (... what were you doing to need 64 styles?)

Answer: Writing incorrect code haha :). I was using it to “replace” the current style on an object. It worked in the immediate sense. 

Incidentally, is there guidance on how I should be replacing a style? Or should I not be replacing styles? My use case is changing an input field to look like it no longer looks interactable.

PR incoming. A bigger change I thiiink should be considered is to make realloc not return an address to the code area but rather NULL (or just crashes). Something so it fails fast.

martaaay, 2022-10-06:
Here's the PR: https://github.com/lvgl/lvgl/pull/3727

kisvegabor, 2022-10-07:
> My use case is changing an input field to look like it no longer looks interactable.

For this we have "states": Just add a  style to `LV_STATE_DISABLED` and put the object into disabled state with `lv_obj_add_state(obj, LV_STATE_DISABLED)`.

You can also remove a style with `lv_obj_remove_style(screen, &style, selector);`.

Or you can use local styles, e.g. `lv_obj_set_style_border_color(obj, color, selector)`

---

Anyway, the PR is merged so I close this issue. Thank you!

closed 2022-10-07 by kisvegabor