lvgl/lvgl #3729: Help: Documentation on LV_USE_ASSERT_...

wrgallo, 2022-10-07:
###  LV_USE_ASSERT_...
If your application randomly crashes, [it is recommend to enable asserts](https://docs.lvgl.io/master/intro/index.html#lvgl-doesn-t-start-randomly-crashes-or-nothing-is-drawn-on-the-display-what-can-be-the-problem).
There are a few assert options to detect something wrong in your application.
- `LV_USE_ASSERT_NULL`
- `LV_USE_ASSERT_MALLOC`
- `LV_USE_ASSERT_STYLE`
- `LV_USE_ASSERT_MEM_INTEGRITY`
- `LV_USE_ASSERT_OBJ`

Now when your application crashes, it's possible to catch a few messages such as
```
lv_style_reset: Asserted at expression: style->sentinel == LV_STYLE_SENTINEL_VALUE (Style is not initialized or corrupted)
```

### What does it mean?
But the thing is, what does this mean, and how to fix it?
**_There is no documentation or guide on how to prevent each assertion, such as examples of good and bad practices! 
The lack of information makes it feel like the crashes are just LVGL’s fault, since there is nothing one could do to fix it._**

For me `LV_ASSERT_HANDLER` is called randomly when deleting an object with `lv_obj_clean` or `lv_obj_del` on a object or object's parent which was working fine. 
My application is huge and unfortunately I couldn't find a simple way to reproduce the problem...

kisvegabor, 2022-10-09:
You are right we could be more explicit about the possible issues. At least we can add comments in `lv_conf.h` and make the assert's text more verbose.

As I mentioned in [Forum](https://forum.lvgl.io/t/how-to-prevent-lv-use-assert/9924/2)  
> It means that you want to lv_style_reset() a style which wasn’t lv_style_init()-ed yet.

wrgallo, 2022-10-10:
> You are right we could be more explicit about the possible issues. At least we can add comments in `lv_conf.h` and make the assert's text more verbose.
>
> As I mentioned in [...]

I am not doing this, the `lv_obj_destructor` is, not being trivial to understand what have I done wrong.

What am I doing:
1. The very first thing I do: initialize and define all `static lv_style_t` styles used by my application (no theme is being used);
1. Create a global variable `lv_obj_t* main = lv_obj_create( lv_scr_act() )`, (this `main` will be used as a screen container);
1. Inside the `main` object, create many LVGL widgets using the **static styles** and lots of local styles (`lv_obj_set_style_...`);
1. After some time, clean the `main` variable with `lv_obj_clean(main)`, and re-use the `main` to create another screen.
The `LV_ASSERT_HANDLER` will be trigger here **eventually**

[image]

kisvegabor, 2022-10-11:
Your scenario looks good. 

It seems you can use a debugger so please check if `some_style.sentinel` is really set to `0xAABBCCDD` after `lv_style_init()` and it remains `0xAABBCCDD` at later points too. If you can,  set a breakpoint on the writes of the `sentinel` field.

kisvegabor, 2022-10-20:
`lv_timer_is_valid` makes sense for sure. 

However I'm not sure about the other two. The user can easily implement them using `lv_timer_get_next` exactly as he/she needs. E.g. what if there are more timers with the same `cb` and I'd like to get all of them to pause them all? 

There is no explicit description about  `lv_timer_get_next` in the [docs](https://docs.lvgl.io/master/overview/timer.html) so I'd rather mention it there with an example.

closed 2022-11-11 by lvgl-bot