lvgl/lvgl #3730: about label alpha blend on transparent backgroud

docterling, 2022-10-08:
### Perform all steps below and tick them with [x]
- [ x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x] Update lvgl to the latest version
- [ x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
lv_disp_set_bg_opa set to LV_OPA_0,label alpha blend with jpg result looks like,the actual label text is "pin code:1234".but if LV_OPA above LV_OPA_0,the text looks ok.
[image]

### To Reproduce
cause my jpg is 4k res,so i can't use lv_img_create to decode it,i decode the jpg to BGRA buf1,and init lvgl disp to 400x300,disp buf size 400x300x4,set text like:
lv_obj_t * label2 = lv_label_create(lv_scr_act());
lv_label_set_long_mode(label2, LV_LABEL_LONG_SCROLL_CIRCULAR);
lv_label_set_recolor(label2, true);
lv_label_set_text(label2, "pin code:1234");
lv_obj_set_width(label2, lvgl_disp_w);
lv_obj_set_style_text_align(label2, LV_TEXT_ALIGN_LEFT, 0);
lv_obj_align(label2, LV_ALIGN_DEFAULT, x, y);
finally alpha blend two buf,the text looks so bad.

### Expected behavior
i just want to display a label on top of the jpg,totally transparent backgroud,it looks like:
[image]

docterling, 2022-10-11:
[image]
i've tried and found that LV_COLOR_SCREEN_TRANSP and lv_disp_drv_t.screen_transp has the same effect
the font was used https://lvgl.io/tools/fontconverter to convert arail.ttf:size 26,bpp 4
the disp setting:
lvgl_disp_drv.antialiasing = 1;
lvgl_disp_drv.full_refresh = 1;
lvgl_disp_drv.screen_transp = 1;
bg setting:
lv_disp_set_bg_opa(lv_disp_get_default(), LV_OPA_0);
first line:
lv_obj_t * label1 = lv_label_create(lv_scr_act());
lv_label_set_long_mode(label1, LV_LABEL_LONG_SCROLL_CIRCULAR);
lv_label_set_recolor(label1, true);
lv_label_set_text(label1, #ffffff PC     IP: 192.168.1.34#");
lv_obj_set_width(label1, lvgl_disp_w);
lv_obj_set_style_text_align(label1, LV_TEXT_ALIGN_LEFT, 0);
lv_obj_align(label1, LV_ALIGN_DEFAULT, x, y);
sec line:
lv_obj_t * label2 = lv_label_create(lv_scr_act());
lv_label_set_long_mode(label2, LV_LABEL_LONG_SCROLL_CIRCULAR);
lv_label_set_recolor(label2, true);
lv_label_set_text(label2, "#000000 TEST   IP: 192.168.11.1#");
lv_obj_set_width(label2, lvgl_disp_w);
lv_obj_set_style_text_align(label2, LV_TEXT_ALIGN_LEFT, 0);
lv_obj_align(label2, LV_ALIGN_DEFAULT, x, y);
the rest of lines:
static lv_color_t cbuf[LV_CANVAS_BUF_SIZE_TRUE_COLOR_ALPHA(400, 300)];
static lv_obj_t *canvas;
canvas = lv_canvas_create(lv_scr_act());
lv_canvas_set_buffer(canvas, cbuf, 400, 300, LV_IMG_CF_TRUE_COLOR_ALPHA);
lv_obj_set_align(canvas, LV_ALIGN_TOP_LEFT);
lv_draw_label_dsc_t label_dsc;
lv_draw_label_dsc_init(&label_dsc);
label_dsc.color = lv_color_black();
lv_canvas_draw_text(canvas, x, y+text_h*num, lvgl_disp_w, &label_dsc, "#000000 TEST ID: label1#");
......
as the pic show
1.first line antialiasing looks good,but there's some black artifact behind it
2.the sec line there's no antialiasing effect and looks blurred
3.the rest of lines look good,but there are no antialiasing effect.why the labels paint on the canvas was not like the first two lines.

kisvegabor, 2022-10-20:
Can you send a full project that I can try out in a simulator?

docterling, 2022-10-21:
sorry,This project is strongly related to hardware,so i can't just give u the lvgl code,cause that will not reproduce the problem.
the backgroud jpg is totally  independent of lvgl,the pic shows above is lvgl and jpg alpha blend by hardware.
the problem maybe fix in two ways:
1.later lvgl version support 4k res jpg(lv_img_header_t only use 11 bits to store w and h,or maybe i misunstand this),so i can use lvgl img to show the pic and then draw label on it.
2.find out what's the difference label direct draw on LV_OPA_0 backgroud and draw on canvas first,also the canvas is LV_OPA_0.maybe you can tell me how to do some debugging that u want to.

kisvegabor, 2022-10-23:
> 1.later lvgl version support 4k res jpg(lv_img_header_t only use 11 bits to store w and h,or maybe i misunstand this),so i can use lvgl img to show the pic and then draw label on it.

It'd the simplest. If it's a static image, can you split it into 4 parts and put them into an `lv_obj` container next to/ below each other?

docterling, 2022-10-25:
yeah,that would be fine just some complicate,my jpg may change by other peoples,so i can't tell them to split,they would be feel weird.
Won't the future version be support 4k res img?

kisvegabor, 2022-10-26:
> Won't the future version be support 4k res img?

Yes, definitely.  It will be updated in v9.

closed 2022-10-26 by docterling

LaterBird, 2023-01-31:
hi, I'm having the same issue; I tried switching versions, Since 8.3.0, the font display is abnormal;
[image]
```
void lv_example_freetype_830(void)
{
    /*Create a font*/
    static lv_ft_info_t info;
    /*FreeType uses C standard file system, so no driver letter is required.*/
    info.name = "/sdcard/font/xxx.ttf";
    info.weight = 64;
    info.style = FT_FONT_STYLE_NORMAL;
    // info.mem = NULL;
    if(!lv_ft_font_init(&info)) {
        LV_LOG_ERROR("create failed.");
    }

    /*Create style with the new font*/
    static lv_style_t style;
    lv_style_init(&style);
    lv_style_set_text_align(&style, LV_TEXT_ALIGN_CENTER);
    lv_style_set_text_color(&style, LV_COLOR_MAKE(0xFF, 0, 0));
    lv_style_set_text_font(&style, info.font);

    /*Create a label with the new style*/
    lv_obj_t * label = lv_label_create(lv_scr_act());
    lv_obj_add_style(label, &style, 0);
    lv_label_set_text(label, "1 M 0 O");
    lv_obj_center(label);
    lv_obj_set_style_bg_opa(lv_scr_act(), LV_OPA_TRANSP, 0);
}
```
The data is saved to memory in the flush callback and eventually written to a file.

kisvegabor, 2023-02-01:
Is `LV_COLOR_SCREEN_TRANSP` enabled in `lv_conf.h`?

LaterBird, 2023-02-02:
> Is `LV_COLOR_SCREEN_TRANSP` enabled in `lv_conf.h`?

yes, LV_COLOR_SCREEN_TRANSP is enable; I seem to have dealt with the situation, but I can't understand it. in this issue #3959