lvgl/lvgl #3734: Canvas example/documentation using bad calculation for the size of the color array

0x0203, 2022-10-09:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
This seems to apply to both the documentation, and lv_example_canvas_1.c, and possibly elsewhere.  LV_CANVAS_BUF_SIZE_TRUE_COLOR is being used to calculate the size of the drawing buffer 'cbuf', used to hold the canvas pixels. The LV_CANVAS_BUF_SIZE_TRUE_COLOR macro expands to (LV_COLOR_SIZE / 8 ) * w * h. But since cbuf is declared as an array of lv_color_t's, the size in bytes will already be multiplied by the size of a pixel/color. The cbuf then ends up being multiple times larger than is necessary. 

In lvgl/examples/widgets/canvas/lv_example_canvas_1.c:28
```c
static lv_color_t cbuf[LV_CANVAS_BUF_SIZE_TRUE_COLOR(CANVAS_WIDTH, CANVAS_HEIGHT)];
```
should probably be:

```c
static lv_color_t cbuf[CANVAS_WIDTH * CANVAS_HEIGHT];
```
see also lvgl/docs/widgets/canvas.md:21

kisvegabor, 2022-10-11:
Thank you. I've fixed the docs and the examples too.

closed 2022-10-11 by kisvegabor

el-samiyel, 2022-10-25:
Hi Both, I still feel like there may be something not quite right here. I see the changes were made as per https://github.com/lvgl/lvgl/commit/853163f61534c144bf58895100012dad0e323ffe but I don't see how this refers to `CANVAS_WIDTH * CANVAS_HEIGHT`. All that has been changed is the removal of 'lv_color_t `in exchange for `uint8_t`. 

I'm no low level programmer, so its probably just me, however, I do find it strange that I am able to set my own buffer with `static uint8_t cbuf[CANVAS_WIDTH *CANVAS_HEIGHT]; //(320*240)` and my program compiles fine, but as soon as I call `lv_canvas_set_buffer(ui_mCanvas, cbuf, CANVAS_WIDTH , CANVAS_HEIGHT, LV_IMG_CF_TRUE_COLOR);` my teensy 4 continues to reset as if its out of memory.

I can confirm that when setting like this `static uint8_t cbuf[LV_CANVAS_BUF_SIZE_TRUE_COLOR(CANVAS_WIDTH , CANVAS_HEIGHT)];` the program fails to compile and says its out of memory.

Is it me or does something need doing with the `lv_canvas_set_buffer` function? I just find it hard to believe that I can manually set a buffer, thus assigning space, but when running the MCU is out of memory...

E

0x0203, 2022-10-26:
His change is indeed correct, although I would have preferred my suggestion as it eliminates some confusion and allows you to index into the cbuf array by pixel index instead of pixel index * `sizeof( lv_color_t )`. But the change does indeed allocate the correct number of bytes. I suspect that he went with the uint8_t change so that the example could still document the use of the `LV_CANVAS_BUF_SIZE_TRUE_COLOR` macro, which I don't think are really documented anywhere else. 

You can either allocate `cbuf` as an array of `lv_color_t`'s sized to be `CANVAS_WIDTH * CANVAS_HEIGHT`, or do as the fix does and use `uint8_t`'s and use the `LV_CANVAS_BUF_SIZE_TRUE_COLOR(CANVAS_WIDTH, CANVAS_HEIGHT)` macro to calculate the length, which accounts for both the length/width and the size of each pixel when using `LV_IMG_CF_TRUE_COLOR`. Either one of those will calculate the correct number of bytes needed for the cbuf array. 

 `lv_canvas_set_buffer` operates on the number of pixels, not bytes, so it does need `CANVAS_WIDTH, CANVAS_HEIGHT`.

el-samiyel, 2022-10-26:
Yeesh, understood. More for me to learn. Interesting stuff though. The usage of LVGL is really pushing me to learn to code correctly and understand what each segment of code actually does rather than copy and paste... Good to know the implementation is correct, I guess my issue is down to the ram spec of my teensy 4.0. Looks like I will need to upgrade to the teensy 4.1 and attempt to utilise the psram in order to fully utilise LVGLs image/canvas drawing functionality. 

Many thanks for the pointers.

E

commit 853163f6 (2022-10-11): docs(calvas): fix that LV_CANVAS_BUF_SIZE_ return byte count and not pixel count

fixes #3734