lvgl/lvgl #3742: Use PSRAM for custom malloc/free in ESP32 platform but crash LoadProhibited

maxpromer, 2022-10-14:
### Describe the bug
I use LVGL V8.3.3 and configs `lv_conf.h` here

```c++
/*=========================
   MEMORY SETTINGS
 *=========================*/

/*1: use custom malloc/free, 0: use the built-in `lv_mem_alloc()` and `lv_mem_free()`*/
#define LV_MEM_CUSTOM 1
#if LV_MEM_CUSTOM == 0
    /*Size of the memory available for `lv_mem_alloc()` in bytes (>= 2kB)*/
    #define LV_MEM_SIZE (48U * 1024U)          /*[bytes]*/

    /*Set an address for the memory pool instead of allocating it as a normal array. Can be in external SRAM too.*/
    #define LV_MEM_ADR 0     /*0: unused*/
    /*Instead of an address give a memory allocator that will be called to get a memory pool for LVGL. E.g. my_malloc*/
    #if LV_MEM_ADR == 0
        #undef LV_MEM_POOL_INCLUDE
        #undef LV_MEM_POOL_ALLOC
    #endif

#else       /*LV_MEM_CUSTOM*/
    // #define LV_MEM_CUSTOM_INCLUDE <stdlib.h>   /*Header for the dynamic memory function*/
    #define LV_MEM_CUSTOM_INCLUDE <esp32-hal-psram.h>
    #define LV_MEM_CUSTOM_ALLOC   ps_malloc
    #define LV_MEM_CUSTOM_FREE    free
    #define LV_MEM_CUSTOM_REALLOC ps_realloc
#endif     /*LV_MEM_CUSTOM*/
```

### To Reproduce

I don't test on small code but i test my full code in SRAM (original malloc and free function) it work well no crash.

### Expected behavior
here is log

```
[Trace] (49.423, +11)    event_send_core: Sending event 44 to 0x3d8a5208 with 0x3fce5a3c param  (in lv_event.c line #428)
[Trace] (49.433, +10)    event_send_core: Sending event 44 to 0x3d8a5318 with 0x3fce5a3c param  (in lv_event.c line #428)
[Trace] (49.444, +11)    event_send_core: Sending event 44 to 0x3d8a5318 with 0x3fce5a3c param  (in lv_event.c line #428)
[Trace] (49.464, +20)    event_send_core: Sending event 44 to 0x3d8a229c with 0x3fce5a1c param  (in lv_event.c line #428)
[Trace] (49.475, +11)    event_send_core: Sending event 44 to 0x3d8a229c with 0x3fce5a1c param  (in lv_event.c line #428)
[Trace] (49.486, +11)    event_send_core: Sending event 34 to 0x3d8a34ac with 0x3d8a229c param  (in lv_event.c line #428)
[Trace] (49.496, +10)    event_send_core: Sending event 44 to 0x3d8a2ab4 with 0x3fce5a1c param  (in lv_event.c line #428)
[Trace] (49.507, +11)    event_send_core: Sending event 44 to 0x3d8a2ab4 with 0x3fce5a1c param  (in lv_event.c line #428)
[Trace] (49.518, +11)    event_send_core: Sending event 34 to 0x3d8a27bc with 0x3d8a2ab4 param  (in lv_event.c line #428)
[Trace] (49.528, +10)    event_send_core: Sending event 44 to 0x3d8a2ea8 with 0x3fce5a1c param  (in lv_event.c line #428)
[Trace] (49.549, +21)    event_send_core: Sending event 44 to 0x3d8a2ea8 with 0x3fce5a1c param  (in lv_event.c line #428)
[Trace] (49.560, +11)    event_send_core: Sending event 34 to 0x3d8a2be8 with 0x3d8a2ea8 param  (in lv_event.c line #428)
[Trace] (49.570, +10)    event_send_core: Sending event 44 to 0x3d8a3294 with 0x3fce5a1c param  (in lv_event.c line #428)
[Trace] (49.581, +11)    event_send_core: Sending event 44 to 0x3d8a3294 with 0x3fce5a1c param  (in lv_event.c line #428)
[Trace] (49.591, +10)    event_send_core: Sending event 34 to 0x3d8a2fd0 with 0x3d8a3294 param  (in lv_event.c line #428)
[Trace] (49.602, +11)    lv_obj_update_layout: Layout update end        (in lv_obj_pos.c line #317)
[Info]  (49.612, +10)    lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Trace] (49.623, +11)    lv_obj_update_layout: Layout update end        (in lv_obj_pos.c line #317)
[Info]  (49.633, +10)    lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Trace] (49.643, +10)    lv_obj_update_layout: Layout update end        (in lv_obj_pos.c line #317)
[Trace] (49.654, +11)    event_send_core: Sending event 18 to 0x3d8a1fe8 with 0x3fce5a58 param  (in lv_event.c line #428)
[Trace] (49.664, +10)    event_send_core: Sending event 20 to 0x3d8a1fe8 with 0x3d8a10e8 param  (in lv_event.c line #428)
[Trace] (49.675, +11)    event_send_core: Sending event 21 to 0x3d8a1fe8 with 0x3d8a10e8 param  (in lv_event.c line #428)
[Trace] (49.685, +10)    event_send_core: Sending event 26 to 0x3d8a1fe8 with 0x3fce58a8 param  (in lv_event.c line #428)
[Trace] (49.717, +32)    event_send_core: Sending event 27 to 0x3d8a1fe8 with 0x3fce58a8 param  (in lv_event.c line #428)
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x420ee85b  PS      : 0x00060630  A0      : 0x82015f38  A1      : 0x3fce57d0
A2      : 0xdff7dff7  A3      : 0x00000000  A4      : 0x0000c235  A5      : 0x3c122f78
A6      : 0x0000c235  A7      : 0x00000012  A8      : 0x3fc9a160  A9      : 0x3fce5510
A10     : 0xbf15b9a8  A11     : 0x00000200  A12     : 0x3c12eba4  A13     : 0x3c12ebd4
A14     : 0x00000031  A15     : 0x000002cd  SAR     : 0x00000003  EXCCAUSE: 0x0000001c
EXCVADDR: 0xdff7e00f  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff

Backtrace:0x420ee858:0x3fce57d00x42015f35:0x3fce57f0 0x42016072:0x3fce5820 0x420185d3:0x3fce5860 0x42018d18:0x3fce5910 0x420ee8a5:0x3fce5940 0x42015f91:0x3fce5960 0x42016072:0x3fce5990 0x4201eeb9:0x3fce59d0 0x4201ef29:0x3fce5a10 0x4201f11f:0x3fce5a60 0x4201f34a:0x3fce5a80 0x4201f491:0x3fce5af0 0x4201f5d9:0x3fce5b30 0x4201f7cd:0x3fce5b50 0x42027cab:0x3fce5b80 0x42027d91:0x3fce5ba0 0x4200dc35:0x3fce5bc0

  #0  0x420ee858:0x3fce57d0 in lv_obj_get_event_dsc at lib/lvgl/src/core/lv_event.c:420
  #1  0x42015f35:0x3fce57f0 in event_send_core at lib/lvgl/src/core/lv_event.c:439
  #2  0x42016072:0x3fce5820 in lv_event_send at lib/lvgl/src/core/lv_event.c:75
  #3  0x420185d3:0x3fce5860 in lv_obj_draw at lib/lvgl/src/core/lv_obj.c:558
  #4  0x42018d18:0x3fce5910 in lv_obj_event at lib/lvgl/src/core/lv_obj.c:846
  #5  0x420ee8a5:0x3fce5940 in lv_obj_event_base at lib/lvgl/src/core/lv_event.c:98
  #6  0x42015f91:0x3fce5960 in event_send_core at lib/lvgl/src/core/lv_event.c:458
  #7  0x42016072:0x3fce5990 in lv_event_send at lib/lvgl/src/core/lv_event.c:75
  #8  0x4201eeb9:0x3fce59d0 in lv_obj_redraw at lib/lvgl/src/core/lv_refr.c:147
  #9  0x4201ef29:0x3fce5a10 in refr_obj at lib/lvgl/src/core/lv_refr.c:885
  #10 0x4201f11f:0x3fce5a60 in refr_obj_and_children at lib/lvgl/src/core/lv_refr.c:767
  #11 0x4201f34a:0x3fce5a80 in refr_area_part at lib/lvgl/src/core/lv_refr.c:702
  #12 0x4201f491:0x3fce5af0 in refr_area at lib/lvgl/src/core/lv_refr.c:594 (discriminator 2)
  #13 0x4201f5d9:0x3fce5b30 in refr_invalid_areas at lib/lvgl/src/core/lv_refr.c:532
  #14 0x4201f7cd:0x3fce5b50 in _lv_disp_refr_timer at lib/lvgl/src/core/lv_refr.c:324
  #15 0x42027cab:0x3fce5b80 in lv_timer_exec at lib/lvgl/src/misc/lv_timer.c:313 (discriminator 2)
  #16 0x42027d91:0x3fce5ba0 in lv_timer_handler at lib/lvgl/src/misc/lv_timer.c:109
  #17 0x4200dc35:0x3fce5bc0 in Display_loopTask(void*) at src/LCD/Display.cpp:184

ELF file SHA256: 0000000000000000

Rebooting...
�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377938
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
SHA-256 comparison failed:
Calculated: dcde8d8a4817d9bf5d5d69a7247667264e4e10ac7493514868b61f5aa6146539
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
```

### Screenshots or video
---

kisvegabor, 2022-10-15:
It's hard to tell what could be the problem. 

> I don't test on small code but i test my full code in SRAM (original malloc and free function) it work well no crash.

Just to confirm: it was working well with `LV_MEM_CUSTOM 0`, right?

closed 2022-11-07 by lvgl-bot

sukesh-ak, 2022-11-21:
You can allocate the buffer like this with `EXT_RAM_BSS_ATTR`
```c
EXT_RAM_BSS_ATTR lv_color_t * buf1 = (lv_color_t *)malloc(screenWidth * BUFF_SIZE * sizeof(lv_color_t));
```