lvgl/lvgl #3745: unknown type name 'LV_ATTRIBUTE_LARGE_CONST' on Vex v5 displays

Turtgun, 2022-10-15:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
When using the image to c array and pasting it into a project and refactoring it I got the error
```
unknown type name 'LV_ATTRIBUTE_LARGE_CONST'
```
I am using Vex's V5 robot lcd display and don't think I can edit lv_conf.h without it spitting errors at me

### To Reproduce

Turtgun, 2022-10-15:
lv_conf.h
```c
#define LV_ATTRIBUTE_LARGE_CONST
```

is what took away the errors, however I do not have the robot with me to test if the brain supports LV_ATTRIBUTE_LARGE_CONST,
If it doesn't is there an alternative to LV_ATTRIBUTE_LARGE_CONST?

embeddedt, 2022-10-15:
I'm guessing your LVGL version didn't have the LARGE_CONST attribute defined; the solution you used should work to get around the error.

Turtgun, 2022-10-18:
I have tested it on the robot, the Vex v5 displays No Image data even though the C array is being called

src/main.cpp
```cpp
        #include "Display.h"
        using namespace Display;
        ...
	LV_IMG_DECLARE(alpha);
	lv_img_disp(alpha);
```
src/Display.h
```cpp
namespace Display {
    static void lv_img_disp(lv_img_dsc_t cArr)
    {
        lv_obj_t* img = lv_img_create(lv_scr_act(), NULL);
        lv_img_set_src(img, &cArr);
        lv_obj_align(img, NULL, LV_ALIGN_IN_TOP_LEFT, 0, 0);
    }
    ...
};
```
alpha.c is in the same directory
src/alpha.c
```c
const LV_ATTRIBUTE_MEM_ALIGN LV_ATTRIBUTE_LARGE_CONST LV_ATTRIBUTE_IMG_ALPHA uint8_t alpha_map[] = {...}

const lv_img_dsc_t alpha = {
  .header.cf = LV_IMG_CF_TRUE_COLOR_ALPHA,
  .header.always_zero = 0,
  .header.reserved = 0,
  .header.w = 480,
  .header.h = 360,
  .data_size = 172800 * LV_IMG_PX_SIZE_ALPHA_BYTE,
  .data = alpha_map,
};
```

Am I doing something wrong?

embeddedt, 2022-10-18:
`alpha` is an `lv_img_dsc_t`, so I think you should be passing a pointer to that to `lv_img_disp`, and change `lv_img_disp` to take an `lv_img_dsc_t *`.

Turtgun, 2022-10-18:
```
Cannot cast from type 'const lv_img_dsc_t' to pointer type 'lv_img_dsc_t *'
```

embeddedt, 2022-10-18:
You then need to call `lv_img_disp(&cArr)` and change `lv_img_disp` to call `lv_img_set_src` with the given pointer rather than a pointer to that pointer.

Turtgun, 2022-10-18:
I don't follow, is it like this?

```cpp

    static void lv_img_disp(lv_img_dsc_t* cArr)
    {
        lv_obj_t* img = lv_img_create(lv_scr_act(), NULL);
        lv_img_set_src(img, cArr);
        lv_obj_align(img, NULL, LV_ALIGN_IN_TOP_LEFT, 0, 0);
    }
```
```cpp
	lv_img_disp(&alpha);
```

even so, I get:
```
Candidate function not viable: 1st argument ('const lv_img_dsc_t *') would lose const qualifier
```

Turtgun, 2022-10-18:
Oh, thanks,
I'll go test it either today or tomorrow to tell you what I get.
Either-ways, thank you for your very quick replies, it's quite incredible.

Turtgun, 2022-10-21:
It does work but I have issues with unrelated things so I will now close this issue.

closed 2022-10-21 by Turtgun