lvgl/lvgl #3748: can't add table to group or only can select table once

PawnMa, 2022-10-18:
### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

PawnMa, 2022-10-20:
I did not add obj-btn1 to group. if I add btn1，the phenomenon is the same as you.
You can try to remove btn1 from group.
thanks!!

kisvegabor, 2022-10-21:
Ahh, I think then this is missing:
```c
    lv_obj_add_flag(table, LV_OBJ_FLAG_SCROLL_ON_FOCUS);
```

PawnMa, 2022-10-21:
The problem has been solved Thank you very much

closed 2022-10-21 by PawnMa