lvgl/lvgl #3749: When use the IME_pinyin keyboard can't show the candidate box

Littermonkey, 2022-10-19:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Don't show the Chinese selection box.
### To Reproduce

```c
void setup_scr_screen(lv_ui *ui){

	//Write codes screen
	ui->screen = lv_obj_create(NULL);
	lv_obj_set_scrollbar_mode(ui->screen, LV_SCROLLBAR_MODE_OFF);

	//Write codes screen_ta_1
	ui->screen_ta_1 = lv_textarea_create(ui->screen);
	lv_obj_set_pos(ui->screen_ta_1, 20, 67);
	lv_obj_set_size(ui->screen_ta_1, 200, 30);
	lv_obj_set_scrollbar_mode(ui->screen_ta_1, LV_SCROLLBAR_MODE_OFF);
	lv_textarea_set_text(ui->screen_ta_1,"Hello World");

	lv_obj_t * pinyin_ime = lv_ime_pinyin_create(ui->screen);
    lv_obj_set_style_text_font(pinyin_ime, &lv_font_simsun_16_cjk, 0);
	g_kb = lv_keyboard_create(ui->screen);
	lv_ime_pinyin_set_keyboard(pinyin_ime, g_kb);
	lv_keyboard_set_textarea(g_kb, ui->screen_ta_1);
	lv_obj_add_event_cb(g_kb, kb_event_cb, LV_EVENT_ALL, NULL);
	lv_obj_add_flag(g_kb, LV_OBJ_FLAG_HIDDEN);
	lv_obj_add_event_cb(ui->screen_ta_1, ta_event_cb, LV_EVENT_ALL, g_kb);

}
void setup_ui(lv_ui *ui){
	init_scr_del_flag(ui);
	setup_scr_screen(ui);
	lv_scr_load(ui->screen);
}
```
### Expected behavior

When I look at the source code I find that the keyboard pinyin cand_panel create based on "lv_scr_act()". I think it's the root cause, indeed when I change "lv_scr_act()" to "obj", it works fine.  what do you think about this issue? 
[image]

### Screenshots or video

[image]

kisvegabor, 2022-10-20:
I think `lv_ime_pinyin_set_keyboard` should change the parent of `img_pinyin` to the parent of the keyboard. 

```c
void lv_ime_pinyin_set_keyboard(lv_obj_t * obj, lv_obj_t * kb)
{
    if(kb) {
        LV_ASSERT_OBJ(kb, &lv_keyboard_class);
    }

    LV_ASSERT_OBJ(obj, MY_CLASS);
    lv_ime_pinyin_t * pinyin_ime = (lv_ime_pinyin_t *)obj;

    pinyin_ime->kb = kb;

    //Add this
    lv_obj_set_parent(obj, lv_obj_get_parent(kb));

    lv_obj_add_event_cb(pinyin_ime->kb, lv_ime_pinyin_kb_event, LV_EVENT_VALUE_CHANGED, obj);
    lv_obj_align_to(pinyin_ime->cand_panel, pinyin_ime->kb, LV_ALIGN_OUT_TOP_MID, 0, 0);
}
```

Littermonkey, 2022-10-20:
Hi @kisvegabor 
     According to the method you provided, the keyboard is stuck, I think before the screen loads successfully `lv_scr_act()` is invalid.

kisvegabor, 2022-10-21:
I took any deeper look, tested it an pushed a fix.
Does it work on your side too?

@100ask 
Now the `pinyin_ime` object is hidden and the `cand_panel` button matrix is created which is actually visible for the user. I wonder if `ime_pinyin` could be based on `lv_btnmatrix` directly to avoid having a hidden object just to store data. What do you think?

commit 1bad397c (2022-10-21): fix(ime_pinyin): make ime_pinyin_set_keyboard() change the parent of pinyin IME to the keyboard's parent

fixes: #3749
closed 2022-10-21 by kisvegabor