lvgl/lvgl #3752: LV_GRADIENT_MAX_STOPS redefined warnings

dizcza, 2022-10-20:
### Introduce the problem
Building lvgl as a component for ESP-IDF project outputs 100+ warnings that LV_GRADIENT_MAX_STOPS redefined.

The issue happens not only with me, see https://forum.lvgl.io/t/trouble-setting-up-the-most-recent-version-of-lvgl-in-blank-esp-idf-project/9784. But I'm the first one to report, as I've found no such mentions in the issues.

ESP-IDF v5.1 master branch.

### Steps to reproduce
1. Clone lvgl master branch to `./components` dir of an empty project in ESP-IDF.
2. Add `lv_conf.h` to the `main` folder.
3. Add
```c
#include "lv_conf.h"
#include <lvgl.h>
```
to `main.cpp`.
4. Build.

#### Output

```
[470/803] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.objIn file included from /home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/hal/../draw/lv_draw.h:18,
                 from /home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/hal/lv_hal_disp.h:21,
                 from /home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/hal/lv_hal.h:16,
                 from /home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/lvgl.h:32,
                 from /home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/main/main.cpp:21:
/home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/hal/../draw/../misc/lv_style.h:75: warning: "LV_GRADIENT_MAX_STOPS" redefined
   75 | #define LV_GRADIENT_MAX_STOPS 2
      | 
In file included from /home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/misc/lv_log.h:16,
                 from /home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/lvgl.h:24,
                 from /home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/main/main.cpp:21:
/home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/misc/../lv_conf_internal.h:882: note: this is the location of the previous definition
  882 |         #define LV_GRADIENT_MAX_STOPS CONFIG_LV_GRADIENT_MAX_STOPS
      | 
[577/803] Building C object esp-idf/lvgl/CMakeFiles/__idf_lvgl.dir/src/core/lv_disp.c.objIn file included from /home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/core/../hal/../draw/lv_draw.h:18,
                 from /home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/core/../hal/lv_hal_disp.h:21,
                 from /home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/core/../hal/lv_hal.h:16,
                 from /home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/core/lv_disp.h:16,
                 from /home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/core/lv_disp.c:9:
/home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/core/../hal/../draw/../misc/lv_style.h:75: warning: "LV_GRADIENT_MAX_STOPS" redefined

...

[780/803] Building C object esp-idf/lvgl/CMakeFiles/__idf_lvgl.dir/src/widgets/textarea/lv_textarea.c.objIn file included from /home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/widgets/textarea/../label/../../core/lv_obj.h:20,
                 from /home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/widgets/textarea/../label/lv_label.h:21,
                 from /home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/widgets/textarea/lv_textarea.h:16,
                 from /home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/widgets/textarea/lv_textarea.c:9:
/home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/widgets/textarea/../label/../../core/../misc/lv_style.h:75: warning: "LV_GRADIENT_MAX_STOPS" redefined
   75 | #define LV_GRADIENT_MAX_STOPS 2
      | 
In file included from /home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/widgets/textarea/../label/lv_label.h:16,
                 from /home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/widgets/textarea/lv_textarea.h:16,
                 from /home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/widgets/textarea/lv_textarea.c:9:
/home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/src/widgets/textarea/../label/../../lv_conf_internal.h:882: note: this is the location of the previous definition
  882 |         #define LV_GRADIENT_MAX_STOPS CONFIG_LV_GRADIENT_MAX_STOPS
      | 
[802/803] Generating binary image from built executableesptool.py v4.3
```

<details>
    <summary>lv_conf.h</summary>

```c
/**
 * @file lv_conf.h
 * Configuration file for v9.0.0-dev
 */

/*
[... 763 lines cut ...]
</details>

CmakeList.txt:

```cmake
# The following four lines of boilerplate have to be in your project's CMakeLists
# in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.12)

set(EXTRA_COMPONENT_DIRS "./esp-idf-lib/components")

set(COMPONENTS esp32 esp_rom freertos log main hmc5883l bmprecord esp_timer onlinemean sdcard record mpu9250 lsm303d lsm303dlhc rm3100 gps rtc_ds3231 bsp ssd1306 mappydot lvgl LovyanGFX ads111x)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)

# LVGL custom config file setup
add_compile_definitions( LGFX_USE_V1 )  # use V1 instead of V0
add_compile_definitions( LV_CONF_INCLUDE_SIMPLE=1 )  # allow custom lv_conf.h path
idf_build_set_property(COMPILE_OPTIONS "-I../main" APPEND)  # directory with lv_conf.h

project(magtraffic)
```

### Suggested solution
I dunno.

kisvegabor, 2022-10-23:
Please try to remove these lines:
https://github.com/lvgl/lvgl/blob/0732400e7b564dd0e7dc4a924619d8e19c5b23a0/src/misc/lv_style.h#L73-L76

And enable `//#define LV_GRADIENT_MAX_STOPS 2` in your `lv_conf.h` again.

dizcza, 2022-10-23:
> Please try to remove these lines:
> 
> https://github.com/lvgl/lvgl/blob/0732400e7b564dd0e7dc4a924619d8e19c5b23a0/src/misc/lv_style.h#L73-L76
> 
> And enable `//#define LV_GRADIENT_MAX_STOPS 2` in your `lv_conf.h` again.

Yeap. That worked out. I no longer see those warnings.

kisvegabor, 2022-10-24:
Thanks for the feedback. I've pushed a fix to both the `release/v8.3` and `master` branches.

commit 9d027b35 (2022-10-24): fix(style): remove the reduntant define of LV_GRADIENT_MAX_STOPS

fixes #3752
closed 2022-10-24 by kisvegabor