lvgl/lvgl #3759: LV_TICK_CUSTOM requires esp_timer with latest ESP-IDF v5

dizcza, 2022-10-22:
```c
#define LV_TICK_CUSTOM 1
#if LV_TICK_CUSTOM
    #define LV_TICK_CUSTOM_INCLUDE "esp_timer.h"         /*Header for the system time function*/
    #define LV_TICK_CUSTOM_SYS_TIME_EXPR (esp_timer_get_time() / 1000LL)    /*Expression evaluating to current system time in ms*/
#endif   /*LV_TICK_CUSTOM*/
```

makes a compiler angry showing that `esp_timer.h` is not found. This is because it became a standalone component since ESP-IDF v5.

To fix this, I changed cmake.esp:
```diff
   idf_component_register(SRCS ${SOURCES} ${EXAMPLE_SOURCES} ${DEMO_SOURCES}
       INCLUDE_DIRS ${LVGL_ROOT_DIR} ${LVGL_ROOT_DIR}/src ${LVGL_ROOT_DIR}/../
-                   ${LVGL_ROOT_DIR}/examples ${LVGL_ROOT_DIR}/demos)
+                   ${LVGL_ROOT_DIR}/examples ${LVGL_ROOT_DIR}/demos
+      REQUIRES esp_timer)
```

dizcza, 2022-10-22:
Ticking with `lv_tick_inc` turned out to be significantly faster than getting the systime anyway.

I thought the opposite.

kisvegabor, 2022-10-24:
> Ticking with lv_tick_inc turned out to be significantly faster than getting the systime anyway.
 
It's really interesting that there is a significant difference. How did you measure it?

dizcza, 2022-10-24:
@kisvegabor by FPS in benchmark

dizcza, 2022-10-24:
Perhaps, the comparison is misleading since it was like 2000 vs 8.

kisvegabor, 2022-10-25:
How/Where do you call `lv_tick_inc`? 

FYI ([link](https://docs.lvgl.io/master/porting/tick.html)):
> lv_tick_inc should be called in a higher priority routine than lv_task_handler() (e.g. in an interrupt) to precisely know the elapsed milliseconds even if the execution of lv_task_handler takes more time.

dizcza, 2022-10-25:
I'm calling `lv_tick_inc` from a zero priority zero core soft timer
https://github.com/sukesh-ak/ESP32-LVGL8x-SDSPI/blob/8ba5f97e25f7b45887ce80dde63340c956f9f340/main/helper_display.hpp#L100

dizcza, 2022-10-25:
I better switch to `LV_TICK_CUSTOM` than spending precious high-priority APP core resources on some timer ticking...

kisvegabor, 2022-10-25:
> I'm calling lv_tick_inc from a zero priority zero core soft timer

I don't know how ESP timers are implemented.
[esp_register_freertos_tick_hook()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos_additions.html#_CPPv431esp_register_freertos_tick_hook22esp_freertos_tick_cb_t) can be also an option.

embeddedt, 2022-10-25:
In this case calling `lv_tick_inc(10)` within that hook should work and be quite fast.

dizcza, 2022-11-09:
> In this case calling `lv_tick_inc(10)` within that hook should work and be quite fast.

I need https://github.com/lvgl/lvgl/issues/3769 to be fixed in order to bench

kisvegabor, 2022-11-09:
The [related PR](https://github.com/lvgl/lvgl/pull/3784) is merged so it should work now.

dizcza, 2022-11-25:
Ah, you have another issue unsolved. Could be a regression bug.

```
/home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/demos/benchmark/lv_demo_benchmark.c: In function 'lv_demo_benchmark_run_scene':
/home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/demos/benchmark/lv_demo_benchmark.c:728:38: error: format '%s' expects argument of type 'char *', but argument 4 has type 'int' [-Werror=format=]
  728 |         lv_label_set_text_fmt(title, "%"LV_PRId32"/"LV_PRId32": %s%s", scene_act * 2 + (opa_mode ? 1 : 0),
      |                                      ^~~
  729 |                               (int)(dimof(scenes) * 2) - 2,
      |                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                               |
      |                               int
/home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/demos/benchmark/lv_demo_benchmark.c:728:38: error: too many arguments for format [-Werror=format-extra-args]
/home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/demos/benchmark/lv_demo_benchmark.c: In function 'scene_next_task_cb':
/home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/demos/benchmark/lv_demo_benchmark.c:999:38: error: format '%s' expects argument of type 'char *', but argument 4 has type 'int' [-Werror=format=]
  999 |         lv_label_set_text_fmt(title, "%"LV_PRId32"/"LV_PRId32": %s%s", scene_act * 2 + (opa_mode ? 1 : 0),
      |                                      ^~~
 1000 |                               (int)(dimof(scenes) * 2) - 2,  scenes[scene_act].name, opa_mode ? " + opa" : "");
      |                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                               |
      |                               int
/home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/demos/benchmark/lv_demo_benchmark.c:999:38: error: too many arguments for format [-Werror=format-extra-args]
```

This is the output of building `release/v8.3` branch with the latest ESP-IDF master branch.

Let me know when you fix this.

kisvegabor, 2022-11-28:
@mysterywolf can it be related to #3836?

I've pushed a fix in ae3825871

dizcza, 2022-11-30:
No, it's not fixed

```
/home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/demos/benchmark/lv_demo_benchmark.c: In function 'lv_demo_benchmark_run_scene':
/home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/demos/benchmark/lv_demo_benchmark.c:728:38: error: format '%ld' expects argument of type 'long int', but argument 4 has type 'int' [-Werror=format=]
  728 |         lv_label_set_text_fmt(title, "%"LV_PRId32"/%"LV_PRId32": %s%s", scene_act * 2 + (opa_mode ? 1 : 0),
      |                                      ^~~
  729 |                               (int)(dimof(scenes) * 2) - 2,
      |                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                               |
      |                               int
/home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/demos/benchmark/lv_demo_benchmark.c: In function 'scene_next_task_cb':
/home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/components/lvgl/demos/benchmark/lv_demo_benchmark.c:999:38: error: format '%ld' expects argument of type 'long int', but argument 4 has type 'int' [-Werror=format=]
  999 |         lv_label_set_text_fmt(title, "%"LV_PRId32"/%"LV_PRId32": %s%s", scene_act * 2 + (opa_mode ? 1 : 0),
      |                                      ^~~
 1000 |                               (int)(dimof(scenes) * 2) - 2,  scenes[scene_act].name, opa_mode ? " + opa" : "");
      |                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                               |
      |                               int
cc1: some warnings being treated as errors
```

kisvegabor, 2022-12-02:
I've pushed a fix, there shouldn't be warning now.

JeremiahGillis, 2022-12-02:
Looks like there are some other examples that need to be fixed with 1ed026ca7307957568fe419f1ff39a15b2535b3e. I had to set `LV_BUILD_EXAMPLES` to false to get everything to build.

A few demo examples are:

`lv_example_flex_1.c`

```
<removed>/libraries/lvgl/examples/layouts/flex/lv_example_flex_1.c:31:46: error: format '%u' expects argument of type 'unsigned int', but argument 3 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
   31 |         lv_label_set_text_fmt(label, "Item: %u", i);
      |                                             ~^   ~
      |                                              |   |
      |                                              |   uint32_t {aka long unsigned int}
      |                                              unsigned int
      |                                             %lu
cc1.exe: some warnings being treated as errors
[230/364] Building C object esp-idf/lvgl/CMakeFiles/__idf_lvgl.dir/examples/layouts/grid/lv_example_grid_6.c.obj
ninja: build stopped: subcommand failed.
```

`lv_example_menu_4.c`

```
<removed>/libraries/lvgl/examples/widgets/menu/lv_example_menu_4.c:21:62: error: format '%i' expects argument of type 'int', but argument 3 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
   21 |     lv_label_set_text_fmt(label, "Hello, I am hiding inside %i", btn_cnt);
      |                                                             ~^   ~~~~~~~
      |                                                              |   |
      |                                                              int uint32_t {aka long unsigned int}
      |                                                             %li
<removed>/libraries/lvgl/examples/widgets/menu/lv_example_menu_4.c:25:41: error: format '%i' expects argument of type 'int', but argument 3 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
   25 |     lv_label_set_text_fmt(label, "Item %i", btn_cnt);
      |                                        ~^   ~~~~~~~
      |                                         |   |
      |                                         int uint32_t {aka long unsigned int}
      |                                        %li
cc1.exe: some warnings being treated as errors
[108/138] Building C object esp-idf/lvgl/CMakeFiles/__idf_lvgl.dir/examples/widgets/meter/lv_example_meter_3.c.obj
ninja: build stopped: subcommand failed.
```

kisvegabor, 2022-12-06:
Thanks, fixed.

dizcza, 2022-12-07:
Updated release/v8.3 branch.

I'm calling the `lv_demo_benchmark()` function and I've set up a
```c
/* Setting up tick task for lvgl */
static void lv_tick_cb(void*)
{
    lv_tick_inc(LV_TICK_PERIOD_MS);
}
```
and the board (WT32-SC01) restarts each time the benchmark hits 3/96 - after "Rectangle rounded + opa". The first three examples show that I have 33000 - 36000 FPS which looks fishy. I don't think that either restarting on number 3/96 or >1k FPS is expected behavior, is it?

JeremiahGillis, 2022-12-07:
We need to be a little careful here using `esp_register_freertos_tick_hook` and `lv_tick_inc`. You may run into exceptions if you're also trying to write to flash, such as using the NVS module. The FreeRTOS tick callback function lives in IRAM and will be called during a flash writes. You can make your hook live in IRAM but not `lv_tick_inc` function without modifying the library.

Sure, you can add logic in the hook to not call `lv_tick_inc` when flash is being written to. This is what I ended up doing. It would be cleaner if we could put `lv_tick_inc` in IRAM.

This is about the only downside I see over using `esp_timer`.

dizcza, 2022-12-07:
I ain't using `esp_register_freertos_tick_hook`

kisvegabor, 2022-12-11:
Please take look at this comment: https://github.com/lvgl/lvgl/issues/2972#issuecomment-1105737276

dizcza, 2022-12-13:
@kisvegabor the issue with WDT is solved, thanks. But why do I have 33000 FPS?

Another question popped up after looking at the comments found in the link above. Currently, I'm using
```c
#define LV_MEM_POOL_INCLUDE     "esp_heap_caps.h"
#define LV_MEM_POOL_ALLOC(size) heap_caps_malloc(size, MALLOC_CAP_SPIRAM)
```
and I saw people play with SPIRAM + DMA + LVGL. To exploit DMA in LVGL with SPIRAM, can I just add the `MALLOC_CAP_DMA` flag like so:
```c
#define LV_MEM_POOL_ALLOC(size) heap_caps_malloc(size, MALLOC_CAP_SPIRAM | MALLOC_CAP_DMA)
```
?

kisvegabor, 2022-12-13:
I'm not an expert on ESP so can't really comment on the mem config. cc @espzav 

Just to see the current state clearly :
- Do you measure the FPS with the benchmark demo?
- How do you call `lv_tick_inc`?

dizcza, 2022-12-13:
> Do you measure the FPS with the benchmark demo?

Yes, that's the value I read from the screen.

> How do you call lv_tick_inc?

I've set up a periodic timer with `esp_timer_start_periodic` that ticks every millisecond. Inside the callback (`lv_tick_cb` defined a few comments above), I increment LVGL's tick by 1 ms.

Let me come up with a minimal example.

kisvegabor, 2022-12-15:
I still believe the problem is that `lv_tick_inc` can't be called in time. Other than  I can't image what can cause the incorrect FPS value. Do you have an oscilloscope? If so, could toggle a pin in the timer interrupt and check it on the oscilloscope?

dizcza, 2022-12-15:
First thing first,

https://github.com/lvgl/lvgl/pull/3784 has NOT fixed the bug.

My `lv_conf.h`:
```c
#define LV_TICK_CUSTOM 1
#if LV_TICK_CUSTOM
    #define LV_TICK_CUSTOM_INCLUDE "esp_timer.h"         /*Header for the system time function*/
    #define LV_TICK_CUSTOM_SYS_TIME_EXPR (esp_timer_get_time() / 1000LL)    /*Expression evaluating to current system time in ms*/
#endif   /*LV_TICK_CUSTOM*/
```

Build log:

```
In file included from /home/dizcza/Projects/Embedded/eclipse-workspace/lvgl_fps/components/lvgl/src/hal/../lv_conf_kconfig.h:15,
                 from /home/dizcza/Projects/Embedded/eclipse-workspace/lvgl_fps/components/lvgl/src/hal/../lv_conf_internal.h:15,
                 from /home/dizcza/Projects/Embedded/eclipse-workspace/lvgl_fps/components/lvgl/src/hal/lv_hal_tick.h:16,
                 from /home/dizcza/Projects/Embedded/eclipse-workspace/lvgl_fps/components/lvgl/src/hal/lv_hal_tick.c:9:
/home/dizcza/Projects/Embedded/eclipse-workspace/lvgl_fps/build/config/sdkconfig.h:455:39: fatal error: esp_timer.h: No such file or directory
  455 | #define CONFIG_LV_TICK_CUSTOM_INCLUDE "esp_timer.h"
      |                                       ^~~~~~~~~~~~~
compilation terminated.
[2/11] Performing build step for 'bootloader'
```

dizcza, 2022-12-15:
Secondly, the pull request mentioned above has done something worse.

Now, when I fix the issue manually - as the author should have done in the first place, -
```diff
   idf_component_register(SRCS ${SOURCES} ${EXAMPLE_SOURCES} ${DEMO_SOURCES}
       INCLUDE_DIRS ${LVGL_ROOT_DIR} ${LVGL_ROOT_DIR}/src ${LVGL_ROOT_DIR}/../
-                   ${LVGL_ROOT_DIR}/examples ${LVGL_ROOT_DIR}/demos)
+                   ${LVGL_ROOT_DIR}/examples ${LVGL_ROOT_DIR}/demos
+      REQUIRES esp_timer)
```
I get another build error:
```
In file included from /home/dizcza/Projects/Embedded/eclipse-workspace/lvgl_fps/components/lvgl/src/hal/lv_hal_tick.h:16,
                 from /home/dizcza/Projects/Embedded/eclipse-workspace/lvgl_fps/components/lvgl/src/hal/lv_hal_tick.c:9:
/home/dizcza/Projects/Embedded/eclipse-workspace/lvgl_fps/components/lvgl/src/hal/lv_hal_tick.c: In function 'lv_tick_get':
/home/dizcza/Projects/Embedded/eclipse-workspace/lvgl_fps/components/lvgl/src/hal/../lv_conf_internal.h:253:51: error: implicit declaration of function 'millis' [-Werror=implicit-function-declaration]
  253 |             #define LV_TICK_CUSTOM_SYS_TIME_EXPR (millis())    /*Expression evaluating to current system time in ms*/
      |                                                   ^~~~~~
/home/dizcza/Projects/Embedded/eclipse-workspace/lvgl_fps/components/lvgl/src/hal/lv_hal_tick.c:77:12: note: in expansion of macro 'LV_TICK_CUSTOM_SYS_TIME_EXPR'
   77 |     return LV_TICK_CUSTOM_SYS_TIME_EXPR;
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
[2/11] Performing build step for 'bootloader'
[1/1] cd /home/dizcza/Projects/Embedded/eclipse-workspace/lvgl_fps/build/bootloader/esp-idf/esptool_py && /home/dizcza/.espressif/python_env/idf5.0_py3.9_env/bin/python /home/dizcza/tools/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /home/dizcza/Projects/Embedded/eclipse-workspace/lvgl_fps/build/bootloader/bootloader.bin
Bootloader binary size 0x66f0 bytes. 0x910 bytes (8%) free.
ninja: build stopped: subcommand failed.
```
Although I define the tick as
```c
#define LV_TICK_CUSTOM 1
#if LV_TICK_CUSTOM
    #define LV_TICK_CUSTOM_INCLUDE "esp_timer.h"         /*Header for the system time function*/
    #define LV_TICK_CUSTOM_SYS_TIME_EXPR (esp_timer_get_time() / 1000LL)    /*Expression evaluating to current system time in ms*/
#endif   /*LV_TICK_CUSTOM*/
```

How could this possibly be?

dizcza, 2022-12-15:
And last but not least,
I'm getting a very strange behavior after I set up a minimal code to reproduce the issue with wrong FPS.

`main.cpp`:
```c
#include <string.h>
#include <unistd.h>
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "driver/gpio.h"
#include "esp_timer.h"

#include <lvgl.h>
#include "lv_conf.h"
#include "demos/lv_demos.h"

#include <LovyanGFX.hpp>
#include <LGFX_AUTODETECT.hpp>

// Portrait
#define TFT_WIDTH   320
#define TFT_HEIGHT  480

#define LV_TICK_PERIOD_MS           1
#define LV_TASK_HANDLER_PERIOD_MS   10

using namespace lgfx::v1;

/*** Setup screen resolution for LVGL ***/
static constexpr uint16_t screenWidth = TFT_HEIGHT;
static constexpr uint16_t screenHeight = TFT_WIDTH;
static SemaphoreHandle_t xGuiSemaphore = NULL;

static lv_disp_draw_buf_t draw_buf;

static lv_disp_t *disp;
static lv_theme_t *theme_current;
static lv_color_t bg_theme_color;

static LGFX lcd; // declare display variable

static int64_t tick_prev = 0;
static int64_t tick_duration_min_us = 0;
static int64_t tick_duration_max_us = 0;
static int64_t tick_counter = 0;

static const char *TAG = "main";

extern "C" { void app_main(); }

/* Setting up tick task for lvgl */
static void lv_tick_cb(void*)
{
    lv_tick_inc(LV_TICK_PERIOD_MS);
    const int64_t tick = esp_timer_get_time();
    if (tick_counter == 1) {
        tick_duration_min_us = tick_duration_max_us = tick - tick_prev;
    } else if (tick_counter > 1) {
        const int64_t duration_us = tick - tick_prev;
        if (duration_us < tick_duration_min_us) tick_duration_min_us = duration_us;
        if (duration_us > tick_duration_max_us) tick_duration_max_us = duration_us;
    }
    tick_prev = tick;
    tick_counter++;
}

bool lvgl_acquire(TickType_t xTicksToWait)
{
[... 191 lines cut ...]
I (1488) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1498) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1508) LovyanGFX: [Autodetect] WT32_SC01
I (1518) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1528) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1538) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1548) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (2038) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2038) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2048) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2058) main: lcd.init() OK
I (2058) main: lv_init OK
I (2158) main: lv_tick timer started
I (2158) main: lv_task_handler timer started
I (3378) main: lv_tick_cb period min 0.005, max 178.237 ms
I (4458) main: lv_tick_cb period min 0.005, max 178.237 ms
I (5588) main: lv_tick_cb period min 0.005, max 178.237 ms
I (6798) main: lv_tick_cb period min 0.005, max 178.237 ms
I (7858) main: lv_tick_cb period min 0.005, max 178.237 ms
I (9018) main: lv_tick_cb period min 0.005, max 178.237 ms
```

And the screen is updated as follows. Two squares jump up and down and nothing more changes. The header got stuck with the message "0/96 Rectangle". No FPS message is shown.

Very weird. I cannot explain why this happen without diving deeply into what `lv_demo_benchmark` does.

dizcza, 2022-12-15:
I hope my effort to finally resolve it won't be lost.

dizcza, 2022-12-15:
I'm using ESP-IDF release v5.0.

kisvegabor, 2022-12-18:
> I get another build error: [...]
How could this possibly be?

It's really strange. This is the relevant part of `lv_conf_internal.h`: 
```c
    #ifndef LV_TICK_CUSTOM_SYS_TIME_EXPR
        #ifdef CONFIG_LV_TICK_CUSTOM_SYS_TIME_EXPR
            #define LV_TICK_CUSTOM_SYS_TIME_EXPR CONFIG_LV_TICK_CUSTOM_SYS_TIME_EXPR
        #else
            #define LV_TICK_CUSTOM_SYS_TIME_EXPR (millis())    /*Expression evaluating to current system time in ms*/
        #endif
    #endif
```

So if `LV_TICK_CUSTOM_SYS_TIME_EXPR` is defined, `millis` shouldn't be defined here. Maybe a `clean`ing? 

> I'm getting a very strange behavior after I set up a minimal code to reproduce the issue with wrong FPS.

```c
    bsp_start_timer(&lv_tick_timer, "lv_tick", lv_tick_cb, NULL, LV_TICK_PERIOD_MS);
    [...]
    bsp_start_timer(&lv_task_handler_timer, "lv_task_handler", lv_task_handler_cb, NULL, LV_TASK_HANDLER_PERIOD_MS);
```

It seems these timers have the same priority and they can't interrupt each other. So `lv_tick_cb` can't be called while `lv_task_handler_cb` is running. 

`lv_task_handler` should be called in normal task instead.

espzav, 2022-12-19:
> and I saw people play with SPIRAM + DMA + LVGL. To exploit DMA in LVGL with SPIRAM, can I just add the `MALLOC_CAP_DMA` flag like so:
> 
> ```c
> #define LV_MEM_POOL_ALLOC(size) heap_caps_malloc(size, MALLOC_CAP_SPIRAM | MALLOC_CAP_DMA)
> ```
> 

Hello,
I am sorry for the late reply. The DMA with SPIRAM is not working at this time. The ESP32S3 has a peripheral GDMA for this but there is no code example now and it has never been tested with LVGL yet.

dizcza, 2022-12-19:
> It seems these timers have the same priority and they can't interrupt each other. So `lv_tick_cb` can't be called while `lv_task_handler_cb` is running.
> 
> `lv_task_handler` should be called in normal task instead.

When I commented out these two lines
```c
//    esp_timer_handle_t lv_task_handler_timer;
//    bsp_start_timer(&lv_task_handler_timer, "lv_task_handler", lv_task_handler_cb, NULL, LV_TASK_HANDLER_PERIOD_MS);
```
and in the main loop
```c
    while (1) {
        lv_task_handler();
        vTaskDelay(pdMS_TO_TICKS(LV_TASK_HANDLER_PERIOD_MS));
    }
```
I finally got a reasonable FPS ~5 ranging from 2 to 13. But I was still getting
```
I (3158) main: lv_tick_cb period min 0.906, max 1.088 ms
I (4158) main: lv_tick_cb period min 0.030, max 2.125 ms
I (5158) main: lv_tick_cb period min 0.026, max 2.125 ms
I (6158) main: lv_tick_cb period min 0.026, max 2.125 ms
```
which means that `lv_tick_inc` was called with different periods. Maybe the deviation in the period - from 0.026 to 2.125 ms - led to different values of FPS per bench task.

When I decided to call both functions in the main loop, it broke the FPS measure again, leading to 33000 FPS:
```c
    while (1) {
        lv_task_handler();
        lv_tick_inc(LV_TASK_HANDLER_PERIOD_MS);
        vTaskDelay(pdMS_TO_TICKS(LV_TASK_HANDLER_PERIOD_MS));
    }
```
why?

dizcza, 2022-12-19:
> It's really strange. This is the relevant part of `lv_conf_internal.h`:
>
> ```c
>     #ifndef LV_TICK_CUSTOM_SYS_TIME_EXPR
>         #ifdef CONFIG_LV_TICK_CUSTOM_SYS_TIME_EXPR
>             #define [...]

Full cleaning does not help.

One thing I omitted explaining this custom tick. When I define custom tick only in `lv_conf.h` like so
```c
#define LV_TICK_CUSTOM 1
#if LV_TICK_CUSTOM
    #define LV_TICK_CUSTOM_INCLUDE "esp_timer.h"         /*Header for the system time function*/
    #define LV_TICK_CUSTOM_SYS_TIME_EXPR (esp_timer_get_time() / 1000LL)    /*Expression evaluating to current system time in ms*/
#endif   /*LV_TICK_CUSTOM*/
```
the benchmark got stuck at the very first frame. This means that my custom define `LV_TICK_CUSTOM_SYS_TIME_EXPR` is never called. This is with
```
[ ] Use a custom tick source
```
in the menuconfig. When I edit LVGL component settings in menuconfig like so
```
[*] Use a custom tick source
(esp_timer.h) Header for the system time function
```
it throws me the error with `millis()` I showed above. This `millis` is hardcoded somewhere in your code and users have no choice of redefining it.

Which leads again to the fundamental question I still don't understand.

Why there are `lv_conf.h` and (partially) duplicated options in LVGL menuconfig? Which one is superior?

kisvegabor, 2022-12-20:
> When I decided to call both functions in the main loop, it broke the FPS measure again, leading to 33000 FPS:

It's the same case, LVGL's tick can't keep ticking independently of `lv_task_handler`.

> Why there are lv_conf.h and (partially) duplicated options in LVGL menuconfig? Which one is superior?

`lv_conf.h` is the superior. But are you sure your `lv_conf.h` is picked up by the compiler. You can try it out by adding `#error "test"` in `lv_conf.h`.

dizcza, 2022-12-20:
> `lv_conf.h` is the superior. But are you sure your `lv_conf.h` is picked up by the compiler. You can try it out by adding `#error "test"` in `lv_conf.h`.

It *is* picked by the compiler and 'error test' proves that. But my guess is that it's picked *after* `lv_conf_internal.h`.

kisvegabor, 2022-12-21:
Actually `lv_conf_internal.h` picks `lv_conf.h`. LVGL only includes `lv_conf_internal.h` and never `lv_conf.h` directly.

dizcza, 2022-12-22:
Well, I couldn't find the cause of the problem back one month ago when I experienced that although I define something in `lv_conf.h`, the compiler still picks partly from LVGL menuconfig. That's why I couldn't understand what's superior. But I failed to find the cause.

dizcza, 2022-12-22:
You have all the files and code to check this by yourself.

dizcza, 2022-12-22:
I've even set up a repo for you to play with.

https://github.com/dizcza/lvgl_fps

kisvegabor, 2023-01-03:
Exactly which ESP board and IDF version do you use? 

Please add the exact commands you used to config and build the project.

dizcza, 2023-01-03:
* WT32-SC01
* IDF 5.0 release
* `idf.py build`
all related config is stored in [sdkconfig](https://github.com/dizcza/lvgl_fps/blob/master/sdkconfig)

kisvegabor, 2023-01-03:
I got a very strange error. Maybe GitHub is now working properly now?
```
git clone https://github.com/dizcza/lvgl_fps.git --recurse-submodules 
Cloning into 'lvgl_fps'...
remote: Enumerating objects: 18, done.
remote: Counting objects: 100% (18/18), done.
remote: Compressing objects: 100% (18/18), done.
remote: Total 18 (delta 0), reused 18 (delta 0), pack-reused 0
Receiving objects: 100% (18/18), 23.03 KiB | 1.05 MiB/s, done.
Submodule 'components/LovyanGFX' (git@github.com:lovyan03/LovyanGFX.git) registered for path 'components/LovyanGFX'
Submodule 'components/lvgl' (git@github.com:lvgl/lvgl.git) registered for path 'components/lvgl'
Cloning into '/home/kisvegabor/projects/lvgl/espressif-workspace/lvgl_fps/components/LovyanGFX'...
Warning: Permanently added the ECDSA host key for IP address '140.82.121.3' to the list of known hosts.
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:lovyan03/LovyanGFX.git' into submodule path '/home/kisvegabor/projects/lvgl/espressif-workspace/lvgl_fps/components/LovyanGFX' failed
Failed to clone 'components/LovyanGFX'. Retry scheduled
Cloning into '/home/kisvegabor/projects/lvgl/espressif-workspace/lvgl_fps/components/lvgl'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:lvgl/lvgl.git' into submodule path '/home/kisvegabor/projects/lvgl/espressif-workspace/lvgl_fps/components/lvgl' failed
Failed to clone 'components/lvgl'. Retry scheduled
Cloning into '/home/kisvegabor/projects/lvgl/espressif-workspace/lvgl_fps/components/LovyanGFX'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:lovyan03/LovyanGFX.git' into submodule path '/home/kisvegabor/projects/lvgl/espressif-workspace/lvgl_fps/components/LovyanGFX' failed
Failed to clone 'components/LovyanGFX' a second time, aborting

```

kisvegabor, 2023-01-03:
I don't know what could be the problem, but I manually cloned both components now they work (I mean the cloning)

I've noticed one important thing. You shouldn't include `lv_conf.h` directly [here](https://github.com/dizcza/lvgl_fps/blob/master/main/main.cpp#L11).

dizcza, 2023-01-03:
> I don't know what could be the problem, but I manually cloned both components now they work (I mean the cloning)

You probably could not clone the repo at once because I've set up git SSH keys, not HTTPS.

> I've noticed one important thing. You shouldn't include lv_conf.h directly [here](https://github.com/dizcza/lvgl_fps/blob/master/main/main.cpp#L11).

I didn't know that. Do I still need these lines then https://github.com/dizcza/lvgl_fps/blob/751c7952c545d398cd17bad73863fb7e6229604c/CMakeLists.txt#L13-L14 ?

kisvegabor, 2023-01-05:
> Do I still need these lines then

I'm not sure, but you can try it out. If the build fails, they are needed. It can't happen that it builds without errors but LVGL haven't found `lv_conf.h` and it's in a semi-working state.

dizcza, 2023-01-05:
I've removed the include `lv_conf.h` line from the `main.c` file and am still getting the build error:
```
/home/dizcza/Projects/Embedded/eclipse-workspace/lvgl_fps/build/config/sdkconfig.h:455:39: fatal error: Arduino.h: No such file or directory
  455 | #define CONFIG_LV_TICK_CUSTOM_INCLUDE "Arduino.h"
      |                                       ^~~~~~~~~~~
```

You can try it yourself. You don't need the board to build the code.

dizcza, 2023-01-05:
What also confuses me is the wording in the menuconfig
```
[ ] Uncheck this to use custom lv_conf.h
```

I digest that this field should be empty in order to pick my `lv_conf.h` file, right?

With this unchecked, upon
```
/home/dizcza/Projects/Embedded/eclipse-workspace/lvgl_fps/build/config/sdkconfig.h:455:39: fatal error: Arduino.h: No such file or directory
  455 | #define CONFIG_LV_TICK_CUSTOM_INCLUDE "Arduino.h"
      |                                       ^~~~~~~~~~~
```
I'm also getting
```
   46 |         #pragma message("Possible failure to include lv_conf.h, please read the comment in this file if you get errors")
```
x100 build errors with `lv_conf.h` line removed from the main.c.

dizcza, 2023-01-05:
All right, I found the correct way to build it.

1. Add 
    ```cmake
    add_compile_definitions( LV_CONF_INCLUDE_SIMPLE=1 )  # allow custom lv_conf.h path
    idf_build_set_property(COMPILE_OPTIONS "-I../main" APPEND)  # directory with lv_conf.h
    ```
    to the root CMakeFile. It should really be `-I../main` and not `-Imain` even though the former is confusing since the main folder is in the same folder as the cmake file.

2. Uncheck `[ ] Uncheck this to use custom lv_conf.h` in the menuconfig.
3. Update the `release/v8.3` LVGL branch with esp_timer fix finally included https://github.com/lvgl/lvgl/commit/ecafa88d2c759514788ab6a500a13261d1a6791f
4. Change the following lines in the `lv_conf.h` to be
```c
#define LV_TICK_CUSTOM 1
#if LV_TICK_CUSTOM
    #define LV_TICK_CUSTOM_INCLUDE "esp_timer.h"         /*Header for the system time function*/
    #define LV_TICK_CUSTOM_SYS_TIME_EXPR (esp_timer_get_time() / 1000LL)    /*Expression evaluating to current system time in ms*/
#endif   /*LV_TICK_CUSTOM*/
```

After all these steps, it works like a charm.

dizcza, 2023-01-05:
Overall, with and without `LV_TICK_CUSTOM` I'm getting ~8 FPS.

Finally, after all this trouble, with *55* comments written, you can close the issue (everything seems to be resolved now). Thanks for your support @kisvegabor

kisvegabor, 2023-01-09:
Thank you for your persistence and sharing the steps. I'm sure it will be useful for others too.

closed 2023-01-09 by kisvegabor