lvgl/lvgl #3760: GIF images with only one frame or one cycle cannot be stopped correctly

qingliyu, 2022-10-22:
### To Reproduce

### Expected behavior

qingliyu, 2022-10-22:
The single cycle gif generated by Photoshop cannot be stopped correctly
    lv_obj_t *icon2 = lv_gif_create(lv_scr_act());
    /*From file*/
    lv_gif_set_src(icon2, "B:/resources/test");
    lv_obj_align(icon2, LV_ALIGN_TOP_LEFT, 0, 0);

kisvegabor, 2022-10-24:
Can you send an example gif?

qingliyu, 2022-10-25:
> Can you send an example gif?

This gif is single cycle on PC and cyclic on LVGL project.

[image]

kisvegabor, 2022-10-29:
Thank you, I fixed it in https://github.com/lvgl/lvgl/commit/4972efc3eeab32b83aa4faa4d471aab7dea71687

commit 4972efc3 (2022-10-29): fix(gif): fix the handling of non-repeating gifs

fixes #3760
closed 2022-10-29 by kisvegabor