lvgl/lvgl #3764: Problem with LVGL subpixel rendering with inverted RGB

EnricoLusaEGF, 2022-10-24:
### Describe the bug

I’m working on a project that uses LVGL 8.3.2.

I’m using fonts created with the LVGL converter, with BPP=4 and subpixel rendering.

The display we are using is “upside down”, so the pixels are in BGR order from the user point of view.

To make subpixel rendering work in this configuration I’ve set LV_DRAW_SW_FONT_SUBPX_BGR to 1.

This has the desired effect: the font is sharper and under a lens it’s visible that the curves are correct.

The problem is that now when I set the font to a color different than black, the colors are skewed

[image] vs [image]
_with BGR  vs  original_

(this screenshot is taken from the vscode simulator, where the effect is identical to the one on my hardware).

(The color depth doesn't affect this).

While writing this report I started fixing it and I made some progress, I will leave this here for information and to refer this in the pull request if everything works out.

### To Reproduce

You can reproduce this by simply running the simulator with the widgets demo, setting 
LV_DRAW_SW_FONT_SUBPX_BGR = 1
in lv_conf.h
and replacing some of the default fonts (in lv_demo_widgets.c) with lv_font_montserrat_12_subpx

[image]

### A bit more analysis

I think the problem comes from the fact that with LV_DRAW_SW_FONT_SUBPX_BGR we simply invert B and R.
This is ok with black because we basically either have black pixel or boundary pixel. Inverted black is still black, and the boundary correctly mirror the subpixel.
For example(you can see the inverted colors of the borders):
[image]
This on a real screen (inverted) behave like this:
**(wrong RGB order)**
[image]
**(correct  BGR order)**
[image]

When the text is colored tho, inverting all B and R will change all the colors, not only the boundaries:
[image]

This is even a bigger issue when the background isn't white, because we are blending normal colors with "inverted" colors:
[image]

kisvegabor, 2022-10-25:
Hi,

Thanks for reporting it. It's a big surprise that this issues was not spotted so far. 

Based on your hints I've looked into it. I'm not entirely sure this might solve the issue: please change these lines
https://github.com/lvgl/lvgl/blob/903e94b716ca1b32cdb51de11df679953699e53b/src/draw/sw/lv_draw_sw_letter.c#L491-L497

to
```c
#if LV_FONT_SUBPX_BGR
                res_color.ch.red = (uint32_t)((uint16_t)txt_rgb[0] * font_rgb[2] + (bg_rgb[0] * (255 - font_rgb[2]))) >> 8;
                res_color.ch.blue = (uint32_t)((uint16_t)txt_rgb[2] * font_rgb[0] + (bg_rgb[2] * (255 - font_rgb[0]))) >> 8;
#else
                res_color.ch.red = (uint32_t)((uint16_t)txt_rgb[0] * font_rgb[0] + (bg_rgb[0] * (255 - font_rgb[0]))) >> 8;
                res_color.ch.blue = (uint32_t)((uint16_t)txt_rgb[2] * font_rgb[2] + (bg_rgb[2] * (255 - font_rgb[2]))) >> 8;
#endif
```

EnricoLusaEGF, 2022-10-25:
Nice! 
That was easier than I thought, I was doing a lookahead to detect boundaries :)

I've tested it on my hardware and it seems to be working as expected. I'll try with different BPP just to be sure.

EnricoLusaEGF, 2022-10-25:
Ok tested with all BPPs, colored text, colored background on BGR screen, everything looks good.

Thanks!

kisvegabor, 2022-10-26:
Great! Thanks for testing it.

I've pushed the fix to both `release/v8.3` and `master`.

commit e050f5ca (2022-10-26): fix(draw): fix text color with sub pixel rendering and BGR order

fixes: #3764
commit 064e4787 (2022-10-26): fix(draw): fix text color with sub pixel rendering and BGR order

fixes: #3764
closed 2022-10-26 by kisvegabor