lvgl/lvgl #3766: Why is there no . h file in the compiled product

chenyinen, 2022-10-25:
### Introduce the problem

### Examples and cases

### Suggested solution

kisvegabor, 2022-10-26:
What is the common practice in this case? Should CMake  also create an `inc` folder that you can copy/install where needed?

chenyinen, 2022-10-26:
Yes, I think that cmake should also create an inc file, otherwise a single library file cannot be used by others. Many relative paths are used in the source code, which makes it inconvenient to embed lvgl into other projects

chenyinen, 2022-10-26:
In my project, I hope to directly use the products of lvgl cmake to provide external projects and form a complete compilation system

kisvegabor, 2022-10-26:
> Many relative paths are used in the source code, which makes it inconvenient to embed lvgl into other projects

We can simply copy the `src` folder, rename it to `inc` and remove all the `.c` files. After that `inc/lvgl.h` should work and all relative path will be correct. What do you think?

0x0203, 2022-10-26:
I had similar thoughts with respect to making a library. My one concern with doing this, however, would be to make sure whatever documentation discusses using the library in this way makes a clear, hard to miss disclaimer about how the compiled library will only work for whatever configuration was set in lv_conf.h when it was compiled. Maybe it's obvious to anyone already working with LVGL, but I can see someone trying to link against the .a for a different target that needs a different configuration (e.g., using a different color depth), and not understanding why it doesn't work.

0x0203, 2022-10-26:
As a follow up thought, perhaps it would be useful to name the .a to include the target platform so that you could have multiple system .a's installed, one for each target you need. For example, lvgl_TFT_eSPI.a and lvgl_esp32.a? Target name could be a user-set parameter set in lv_conf.h or a cmake define.

chenyinen, 2022-10-26:
You are right. If a novice uses it directly, it may be misleading that it cannot run, but the main reason is that its kconfig clipping mode+cmake compilation mode are perfectly compatible with my project. For different target platforms, I just need to use different kconfigs, so there will be no problem you mentioned above!

kisvegabor, 2022-10-29:
Are you interested in updating CMake with creating the `inc` folder? Adding an option to set the output name seems useful. 

BTW, our CMake alrady supports `install`:
https://github.com/lvgl/lvgl/blob/6e62fcfd08987946ab99e6fa3369f4268bab83b1/env_support/cmake/custom.cmake#L67-L71

I'm not a CMake expert so I'm not sure it helps.

0x0203, 2022-10-30:
I'm not sure who's being asked, but I'm probably not the right person to implement this. I'm not using cmake for my builds so testing would be difficult, and I have pretty limited experience in using/configuring it.

kisvegabor, 2022-11-01:
I meant: Are _any of_ you interested in updating CMake with creating the inc folder? :slightly_smiling_face: 

Maybe @chenyinen?

closed 2022-11-04 by chenyinen