lvgl/lvgl #3769: undefined reference to lv_demo_benchmark in v8.3.3

dizcza, 2022-10-26:
So you guys told me that master branch is unstable and I should use v8.3. So I did.

But.

```
/home/dizcza/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(wt32_handler.cpp.obj):(.literal._Z15lv_display_initv+0x58): undefined reference to `lv_demo_benchmark'
/home/dizcza/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/11.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(wt32_handler.cpp.obj): in function `_Z15lv_display_initv':
/home/dizcza/Projects/Embedded/eclipse-workspace/magtraffic/main/wt32_handler.cpp:109: undefined reference to `lv_demo_benchmark'
collect2: error: ld returned 1 exit status
```

However, `lv_demo_benchmark` is compiled successfully on the master branch (with a mode specified as an argument).

I'm using ESP-IDF master.

### To Reproduce

```c
#include "demos/lv_demos.h"

    lv_init();         // Initialize lvgl
    lv_demo_benchmark();
```

I've set `LV_USE_DEMO_BENCHMARK` to 1 in both `lv_conf.h` and skdconfig (I *still* don't know which one is superior).

### Expected behavior

The project builds.

### lv_conf.h

[lv_conf.txt](https://github.com/lvgl/lvgl/files/9873490/lv_conf.txt)

kisvegabor, 2022-10-29:
Please add `#error "test"` to `lv_demo_benchmark.c` to see if it's at least picked by the compiler. 

I suggest trying it before and after `#if LV_USE_DEMO_BENCHMARK` too.

dizcza, 2022-10-29:
It's not being picked by the compiler. Tried before and after `#if LV_USE_DEMO_BENCHMARK`

kisvegabor, 2022-11-01:
I cc @tore-espressif from Espressif. Do you have any idea why `[lvgl/demos](https://github.com/lvgl/lvgl/tree/master/demos)` is not compiled?

rmshub, 2022-11-03:
In master branch, demos source code added to complier with lvgl/env_support/cmake/esp.cmake
But this demo folder is not added in v8.3

https://github.com/lvgl/lvgl/blob/master/env_support/cmake/esp.cmake#L20-L39

kisvegabor, 2022-11-04:
Ah, I see. I din't have a ready-to-use ESP-IDF environment at this moment to test it. Could you send a PR for the `release/v8.3` branch?

kisvegabor, 2022-11-09:
As PR is merged I close this issue. Thank you for your contribution.

closed 2022-11-09 by kisvegabor