lvgl/lvgl #3775: Can't run any demo on Arduino IDE

DonatelloX, 2022-10-29:
**Problem:**
Unable to compile LVGL sample software on ESP32 in Arduino IDE.

**Reproduce:**
Arduino Core ESP32: 2.0.5
Arduino IDE: 2.0.1
LVGL: 8.3.3
LVGL example: 8.0.1

1) copy lv_conf_template.h as lv_conf.h into the Arduino Libraries directory next to the lvgl library folder then change the first #if 0 to #if 1 and set LV_TICK_CUSTOM 1
2) copy lv_conf_template.h as lv_demo_conf.h into the Arduino Libraries directory next to the lvgl library folder then change the first #if 0 to #if 1 and set LV_TICK_CUSTOM 1
3) copy lvgl/examples to lvgl/src/examples. Similarly for the demos lvgl/demos to lvgl/src/demos

Compilation error:
LVGL_Arduino.ino:161: undefined reference to `lv_demo_widgets'

DonatelloX, 2022-10-29:
In the `lv_conf.h` and `lv_demo_conf.h` I have enabled:
`#define LV_USE_DEMO_WIDGETS 1`

but there are errors during the compilation:
```
[...]\Documents\Arduino\libraries\lv_examples\src\lv_demo_widgets\lv_demo_widgets.c: In function 'slider_event_cb':
[...]\Documents\Arduino\libraries\lv_examples\src\lv_demo_widgets\lv_demo_widgets.c:1283:41: error: 'lv_obj_draw_part_dsc_t' {aka 'struct <anonymous>'} has no member named 'clip_area'; did you mean 'draw_area'?
             lv_draw_rect(&bg_area, dsc->clip_area, &rect_dsc);
                                         ^~~~~~~~~
                                         draw_area
[...]\Documents\Arduino\libraries\lv_examples\src\lv_demo_widgets\lv_demo_widgets.c:1289:43: error: 'lv_obj_draw_part_dsc_t' {aka 'struct <anonymous>'} has no member named 'clip_area'; did you mean 'draw_area'?
             lv_draw_label(&txt_area, dsc->clip_area, &label_dsc, buf, NULL);
                                           ^~~~~~~~~
                                           draw_area
[...]\Documents\Arduino\libraries\lv_examples\src\lv_demo_widgets\lv_demo_widgets.c: In function 'chart_event_cb':
[...]\Documents\Arduino\libraries\lv_examples\src\lv_demo_widgets\lv_demo_widgets.c:1337:57: error: 'lv_obj_draw_part_dsc_t' {aka 'struct <anonymous>'} has no member named 'clip_area'; did you mean 'draw_area'?
                 _lv_area_intersect(&obj_clip_area, dsc->clip_area, &obj->coords);
                                                         ^~~~~~~~~
                                                         draw_area
[...]\Documents\Arduino\libraries\lv_examples\src\lv_demo_widgets\lv_demo_widgets.c:1399:45: error: 'lv_obj_draw_part_dsc_t' {aka 'struct <anonymous>'} has no member named 'clip_area'; did you mean 'draw_area'?
                 lv_draw_rect(&bg_area, dsc->clip_area, &rect_dsc);
                                             ^~~~~~~~~
                                             draw_area
[...]\Documents\Arduino\libraries\lv_examples\src\lv_demo_widgets\lv_demo_widgets.c:1405:47: error: 'lv_obj_draw_part_dsc_t' {aka 'struct <anonymous>'} has no member named 'clip_area'; did you mean 'draw_area'?
                 lv_draw_label(&txt_area, dsc->clip_area, &label_dsc, buf, NULL);
                                               ^~~~~~~~~
                                               draw_area
[...]\Documents\Arduino\libraries\lv_examples\src\lv_demo_widgets\lv_demo_widgets.c: In function 'shop_chart_event_cb':
[...]\Documents\Arduino\libraries\lv_examples\src\lv_demo_widgets\lv_demo_widgets.c:1450:35: error: 'lv_obj_draw_part_dsc_t' {aka 'struct <anonymous>'} has no member named 'clip_area'; did you mean 'draw_area'?
             lv_draw_rect(&a, dsc->clip_area, &draw_rect_dsc);
                                   ^~~~~~~~~
                                   draw_area
[...]\Documents\Arduino\libraries\lv_examples\src\lv_demo_widgets\lv_demo_widgets.c:1456:35: error: 'lv_obj_draw_part_dsc_t' {aka 'struct <anonymous>'} has no member named 'clip_area'; did you mean 'draw_area'?
             lv_draw_rect(&a, dsc->clip_area, &draw_rect_dsc);
                                   ^~~~~~~~~
                                   draw_area
[...]\Documents\Arduino\libraries\lv_examples\src\lv_demo_widgets\lv_demo_widgets.c:1461:35: error: 'lv_obj_draw_part_dsc_t' {aka 'struct <anonymous>'} has no member named 'clip_area'; did you mean 'draw_area'?
             lv_draw_rect(&a, dsc->clip_area, &draw_rect_dsc);
                                   ^~~~~~~~~
                                   draw_area
```

kisvegabor, 2022-11-01:
Since LVGL v8.2 the demos and examples live in the lvgl repo so the `LVGL example` library shouldn't be used with LVGL 8.3.

DonatelloX, 2022-11-03:
I tried to install version 8.2 instead of 8.3.3, but the compiler gives me an error:
`undefined reference to `lv_demo_widgets'`
How can I try an example of lvgl? I'd like to use it for my next projects.
Thank you

closed 2022-11-26 by lvgl-bot