lvgl/lvgl #3780: Improve scroll direction prediction

FASTSHIFT, 2022-11-02:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
The current lvgl will determine the direction of the responsive scrolling according to the scrolling axis of the component when it is clicked. When the scroll axes of two components are perpendicular to each other, lvgl will take the scrolling direction of the first clicked component as the scrolling direction of the response, not the actual scrolling direction of the user.

### To Reproduce

```c
static void scroll_dir_test()
{
    lv_obj_t * obj = lv_obj_create(lv_scr_act());
    lv_obj_set_size(obj, lv_pct(200), lv_pct(80));

    lv_obj_t * slider = lv_slider_create(obj);
    lv_obj_set_size(slider, 100, lv_pct(80));
    lv_obj_center(slider);
}
```

### Expected behavior
The first scrolling direction should be based on the user's scrolling direction, which is more in line with the user's intuition.

https://user-images.githubusercontent.com/26767803/199457093-1cea4b8b-83d5-4e1b-b3b2-062d7927a9c2.mp4

### Screenshots or video

https://user-images.githubusercontent.com/26767803/199457269-b8f525ce-f5f8-4dae-8322-cfdb0172e2c6.mp4

kisvegabor, 2022-11-03:
Just to be sure I understand correctly: e.g. in case of a horizontal scroll gesture we should check if there is an object (among the parents too) which can be scrolled horizontally.  If so, actually it works like this with containers. The slider is a special case as it "absorbs" the scrolls. It's because it has 
```c
    lv_obj_clear_flag(obj, LV_OBJ_FLAG_SCROLL_CHAIN); //Absorb the scroll
    lv_obj_clear_flag(obj, LV_OBJ_FLAG_SCROLLABLE); //Children shouldn't add scrollbars and make the slider scrollable 
```

Adding
```c
  lv_obj_add_flag(slider, LV_OBJ_FLAG_SCROLL_CHAIN_HOR); //Absorb only vertical scroll
```

to your example should result in the desired behavior. 

Actually we could automatically add/clear `LV_OBJ_FLAG_SCROLL_CHAIN_HOR/VER` in `LV_EVENT_SIZE_CHANGED` of the slider, however it might overwrite the user's settings. Some smart logic can help though. E.g. if the slider was horizontal so far and `SCROLL_CHAIN_VER` was added but `SCROLL_CHAIN_HOR` wasn't, swap the `SCROLL_CHAIN` flags.

What do you think?

kisvegabor, 2022-11-08:
Ah that's true. It's because adjusting the slider is not considered as scroll so LVGL still keep looking for scroll opportunity. 

Just to be sure, all this are related only the the slider (and maybe a few mire widget), right? And not to the scrolling of LVGL in genereal.

If so we can fix it in the slider by making it absorb all scrolls when it's value has been changed and revert it when the slider is released. What do you think?

FASTSHIFT, 2022-11-10:
> Ah that's true. It's because adjusting the slider is not considered as scroll so LVGL still keep looking for scroll opportunity.
>
> Just to be sure, all this are related only the the slider (and maybe [...]

I agree with this.

kisvegabor, 2022-11-13:
I implemented it as discussed.

commit be9a654a (2022-11-13): feat(slider): handle scroll absorbtion in a smarter way

fixes: #3780
closed 2022-11-13 by kisvegabor