lvgl/lvgl #3783: Compilation error in Arduino IDE for the image button widget

DonatelloX, 2022-11-04:
**Bug description**
I am trying to use lvgl widgets. The button works fine, while I'm having a lot of difficulties with the image button.
There are errors in the compilation:

```
[...]\LVGL_Arduino.ino:214:94: error: invalid conversion from 'int' to 'lv_style_prop_t' [-fpermissive]
[...]\LVGL_Arduino.ino:239:16: error: redeclaration of 'lv_obj_t* label'
[...]\LVGL_Arduino.ino:183:15: note: 'lv_obj_t* label' previously declared here
```

**To Reproduce**
Hardware: ESP32 DevKit + RA8875 resistive touchscreen
Arduino Core ESP32: 2.0.5
Arduino IDE: 2.0.1
LVGL: 8.3.3
LVGL example: 8.1.0

I use the example of the image button version 8.3.3:
  ```
    LV_IMG_DECLARE(imgbtn_left);
    LV_IMG_DECLARE(imgbtn_right);
    LV_IMG_DECLARE(imgbtn_mid);

    /*Create a transition animation on width transformation and recolor.*/
    static lv_style_prop_t tr_prop[] = {LV_STYLE_TRANSFORM_WIDTH, LV_STYLE_IMG_RECOLOR_OPA, 0};
    static lv_style_transition_dsc_t tr;
    lv_style_transition_dsc_init(&tr, tr_prop, lv_anim_path_linear, 200, 0, NULL);

    static lv_style_t style_def;
    lv_style_init(&style_def);
    lv_style_set_text_color(&style_def, lv_color_white());
    lv_style_set_transition(&style_def, &tr);

    /*Darken the button when pressed and make it wider*/
    static lv_style_t style_pr;
    lv_style_init(&style_pr);
    lv_style_set_img_recolor_opa(&style_pr, LV_OPA_30);
    lv_style_set_img_recolor(&style_pr, lv_color_black());
    lv_style_set_transform_width(&style_pr, 20);

    /*Create an image button*/
    lv_obj_t * imgbtn1 = lv_imgbtn_create(lv_scr_act());
    lv_imgbtn_set_src(imgbtn1, LV_IMGBTN_STATE_RELEASED, &imgbtn_left, &imgbtn_mid, &imgbtn_right);
    lv_obj_add_style(imgbtn1, &style_def, 0);
    lv_obj_add_style(imgbtn1, &style_pr, LV_STATE_PRESSED);

    lv_obj_align(imgbtn1, LV_ALIGN_CENTER, 0, 0);

    /*Create a label on the image button*/
    lv_obj_t * label = lv_label_create(imgbtn1);
    lv_label_set_text(label, "Button");
    lv_obj_align(label, LV_ALIGN_CENTER, 0, -4);
```

**Expected behavior**
No compilation error.

DonatelloX, 2022-11-04:
I solved the style problem by changing this:
`static lv_style_prop_t tr_prop[] = {LV_STYLE_TRANSFORM_WIDTH, LV_STYLE_IMG_RECOLOR_OPA, 0};`
in
`static lv_style_prop_t tr_prop[] = {LV_STYLE_TRANSFORM_WIDTH, LV_STYLE_IMG_RECOLOR_OPA, 0};`

But i still have problems, in particular I have these errors:
```
undefined reference to `imgbtn_right'
undefined reference to `imgbtn_mid'
undefined reference to `imgbtn_left'
collect2.exe: error: ld returned 1 exit status
```

So I copied the files into my Arduino project:
```
imgbtn_right.c
imgbtn_mid.c
imgbtn_left.c
```
and then I added in my code:
```
#include "imgbtn_right.c"
#include "imgbtn_mid.c"
#include "imgbtn_left.c"
```

Now, I have the problem in imgbtn files:
```
const lv_img_dsc_t imgbtn_right = {
  .header.always_zero = 0,
  .header.w = 8,
  .header.h = 50,
  .data_size = 400 * LV_IMG_PX_SIZE_ALPHA_BYTE,
  .header.cf = LV_IMG_CF_TRUE_COLOR_ALPHA,
  .data = imgbtn_right_map,
};
```
because the compiler gives these errors:
```
[...]\LVGL_Arduino\imgbtn_right.c:229:3: error: expected primary-expression before '.' token
   .header.always_zero = 0,
   ^
[...]\LVGL_Arduino\imgbtn_right.c:230:3: error: expected primary-expression before '.' token
   .header.w = 8,
   ^
[...]\LVGL_Arduino\imgbtn_right.c:231:3: error: expected primary-expression before '.' token
   .header.h = 50,
   ^
[...]\LVGL_Arduino\imgbtn_right.c:233:3: error: expected primary-expression before '.' token
   .header.cf = LV_IMG_CF_TRUE_COLOR_ALPHA,
   ^
[...]\LVGL_Arduino\imgbtn_mid.c:225:3: error: expected primary-expression before '.' token
   .header.always_zero = 0,
   ^
[...]\LVGL_Arduino\imgbtn_mid.c:226:3: error: expected primary-expression before '.' token
   .header.w = 5,
   ^
[...]\LVGL_Arduino\imgbtn_mid.c:227:3: error: expected primary-expression before '.' token
   .header.h = 49,
   ^
[...]\LVGL_Arduino\imgbtn_mid.c:229:3: error: expected primary-expression before '.' token
   .header.cf = LV_IMG_CF_TRUE_COLOR_ALPHA,
   ^
[...]\LVGL_Arduino\imgbtn_left.c:252:3: error: expected primary-expression before '.' token
   .header.always_zero = 0,
   ^
[...]\LVGL_Arduino\imgbtn_left.c:253:3: error: expected primary-expression before '.' token
   .header.w = 8,
   ^
[...]\LVGL_Arduino\imgbtn_left.c:254:3: error: expected primary-expression before '.' token
   .header.h = 50,
   ^
[...]\LVGL_Arduino\imgbtn_left.c:256:3: error: expected primary-expression before '.' token
   .header.cf = LV_IMG_CF_TRUE_COLOR_ALPHA,
   ^
```

kisvegabor, 2022-11-07:
I've tried it out and it worked for me with this simple INO file:
```c
#include <lvgl.h>
#include <examples/lv_examples.h>

void setup()
{
    lv_init();
    lv_example_imgbtn_1();
}

void loop()
{
    lv_timer_handler(); /* let the GUI do its work */
    delay( 5 );
}
```

This in the library folder (LVGL v8.3.3):
[image]

And this in `lvgl/src` (`examples` and `demos` moved there)
[image]

DonatelloX, 2022-11-10:
If you use the predefinite images in `libraries\lvgl\src\examples\assets`, then the compiler does not fail.
I use a custom image. My PNG image has been converted into C array through the _Online image converter tool_ (https://lvgl.io/tools/imageconverter). Then I copy the test.c file into my LVGL project.

My test.c is:
```
#if defined(LV_LVGL_H_INCLUDE_SIMPLE)
#include "lvgl.h"
#else
#include "lvgl/lvgl.h"
#endif

#ifndef LV_ATTRIBUTE_MEM_ALIGN
#define LV_ATTRIBUTE_MEM_ALIGN
#endif

#ifndef LV_ATTRIBUTE_IMG_TEST
#define LV_ATTRIBUTE_IMG_TEST
#endif

const LV_ATTRIBUTE_MEM_ALIGN LV_ATTRIBUTE_LARGE_CONST LV_ATTRIBUTE_IMG_TEST uint8_t Test_map[] = {
#if LV_COLOR_DEPTH == 1 || LV_COLOR_DEPTH == 8
  /*Pixel format: Red: 3 bit, Green: 3 bit, Blue: 2 bit*/
  0x00, _[...]_ 0x00, 
#endif
#if LV_COLOR_DEPTH == 16 && LV_COLOR_16_SWAP == 0
  /*Pixel format: Red: 5 bit, Green: 6 bit, Blue: 5 bit*/
  0x00, _[...]_ 0x00, 
#endif
#if LV_COLOR_DEPTH == 16 && LV_COLOR_16_SWAP != 0
  /*Pixel format: Red: 5 bit, Green: 6 bit, Blue: 5 bit BUT the 2 bytes are swapped*/
  0x00, _[...]_ 0x00, 
#endif
#if LV_COLOR_DEPTH == 32
  /*Pixel format: Fix 0xFF: 8 bit, Red: 8 bit, Green: 8 bit, Blue: 8 bit*/
  0x00, _[...]_ 0xff, 
#endif
};

const lv_img_dsc_t Test = {
  .header.cf = LV_IMG_CF_TRUE_COLOR,
  .header.always_zero = 0,
  .header.reserved = 0,
  .header.w = 52,
  .header.h = 128,
  .data_size = 6656 * LV_COLOR_SIZE / 8,
  .data = Test_map,
};
```

Now I use the image (test.c) with this code:
```
    LV_IMG_DECLARE(test);
    lv_obj_t * img1 = lv_img_create(lv_scr_act());
    lv_img_set_src(img1, &test);
    lv_obj_align(img1, LV_ALIGN_CENTER, 0, -20);
    lv_obj_set_size(img1, 200, 200);

    lv_obj_t * img2 = lv_img_create(lv_scr_act());
    lv_img_set_src(img2, LV_SYMBOL_OK "Accept");
    lv_obj_align_to(img2, img1, LV_ALIGN_OUT_BOTTOM_MID, 0, 20);
```

I receive more compilation errors. I report only a few but they are really many.

`fatal error: lvgl/lvgl.h: No such file or directory` in `#include "lvgl/lvgl.h"`
The problem is in test.c:
```
#if defined(LV_LVGL_H_INCLUDE_SIMPLE)
#include "lvgl.h"
#else
**#include "lvgl/lvgl.h"**
#endif
```

`error: unknown type name 'LV_ATTRIBUTE_LARGE_CONST'`
The problem is in test.c:
`const LV_ATTRIBUTE_MEM_ALIGN **LV_ATTRIBUTE_LARGE_CONST** LV_ATTRIBUTE_IMG_TEST uint8_t Test_map[] = {`

```
error: unknown type name 'uint8_t'
 const LV_ATTRIBUTE_MEM_ALIGN LV_ATTRIBUTE_IMG_PLATIC uint8_t Platic_map[] = {
                                                      ^~~~~~~
error: unknown type name 'lv_img_dsc_t'
 const lv_img_dsc_t Platic = {
       ^~~~~~~~~~~~
error: field name not in record or union initializer
   .header.cf = LV_IMG_CF_TRUE_COLOR,
   ^
```
[...]

Please, how can it be solved?
Thank you

kisvegabor, 2022-11-13:
Where did you place the new image? In Arduino it should be directly next to the `ino` file.

DonatelloX, 2022-11-13:
Yes, the image array is in INO folder. I also tried to include the file in the Arduino project but it doesn't work.
Moreover, I also to copy the image array in the LVGL folders but the compilation gives errors.

kisvegabor, 2022-11-13:
Arduino compiles only the `src` folder. Does it work in `lvgl/src`?

DonatelloX, 2022-11-13:
Yes, I first copied the image array file to INO folder, then I moved image array file to `lvgl/scr` folder. In both cases, the compilation errors are the same.

DonatelloX, 2022-11-13:
I copied the image file into `lvgl/scr` and then changed the initial part of the file from:
```
#if defined(LV_LVGL_H_INCLUDE_SIMPLE)
#include "lvgl.h"
#else
#include "lvgl/lvgl.h"
#endif
```
in:
`#include "lvgl.h"`

In this way there are no compilation errors and the image is shown on the display.

But the image must be in the `lvgl/scr` folder and therefore it is not in the Arduino project. This can be a problem because when copying the Arduino project there will be no images.

kisvegabor, 2022-11-16:
Aaah, I see. 

Please try this at the beginning of the image and keep it next to the `ino` file.
```c
#ifdef __has_include
    #if __has_include("lvgl.h")
        #ifndef LV_LVGL_H_INCLUDE_SIMPLE
            #define LV_LVGL_H_INCLUDE_SIMPLE
        #endif
    #endif
#endif

#if defined(LV_LVGL_H_INCLUDE_SIMPLE)
  #include "lvgl.h"
#else
  #include "lvgl/lvgl.h"
#endif

```

DonatelloX, 2022-11-19:
If I may suggest: it would be useful to change the output image C array of the "image converter" online tool with the code shown in the previous post. Also change the description `Chroma keyed Make LV_COLOR_TRANSP (lv_conf.h) pixels to transparent` in "image converter" online tool, because `LV_COLOR_TRANSP` no longer exists in 8.3.3 but there is `LV_COLOR_CHROMA_KEY`.

kisvegabor, 2022-11-24:
Does it mean the above mentioned code snippet worked well?

DonatelloX, 2022-11-24:
Yes, it works well.

closed 2022-11-24 by DonatelloX

kisvegabor, 2022-11-24:
@embeddedt Could you update the image converter to include `lvgl.h` with:
```c
#ifdef __has_include
    #if __has_include("lvgl.h")
        #ifndef LV_LVGL_H_INCLUDE_SIMPLE
            #define LV_LVGL_H_INCLUDE_SIMPLE
        #endif
    #endif
#endif

#if defined(LV_LVGL_H_INCLUDE_SIMPLE)
  #include "lvgl.h"
#else
  #include "lvgl/lvgl.h"
#endif
```

embeddedt, 2022-11-28:
@kisvegabor Done. I think the version on lvgl.io will also need to be updated.

kisvegabor, 2022-11-28:
Updated, thank you!