lvgl/lvgl #3785: bug about label long mode

yin2021, 2022-11-07:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
The width of label is 88 , which is larger than the length occupied by the text of label. When LV_LABEL_LONG_CLIP is set, full of text can be shown. When LV_LABEL_LONG_WRAP(LV_LABEL_LONG_DOT) is set, label will be wrap(write dots)
[image]

### To Reproduce
void lv_example_label_1(lv_obj_t *par)
{
    lv_obj_t * bg = lv_obj_create(lv_scr_act());
    lv_obj_set_size(bg, 300, 300);
    lv_obj_align(bg, LV_ALIGN_TOP_LEFT, 0, 0);

    //LV_LABEL_LONG_CLIP
    lv_obj_t * label1 = lv_label_create(bg);
    lv_label_set_long_mode(label1, LV_LABEL_LONG_CLIP);
    lv_label_set_text(label1, "aaaaabbbbb");
    lv_obj_set_width(label1, 88);
    lv_obj_align(label1, LV_ALIGN_TOP_LEFT, 0, 0);

    //LV_LABEL_LONG_DOT
    lv_obj_t * label2 = lv_label_create(bg);
    lv_label_set_long_mode(label2, LV_LABEL_LONG_DOT);
    lv_label_set_text(label2, "aaaaabbbbb");
    lv_obj_set_size(label2, 88, 20);
    lv_obj_align(label2, LV_ALIGN_TOP_LEFT, 0, 40);

    //LV_LABEL_LONG_WRAP
    lv_obj_t * label3 = lv_label_create(bg);
    lv_label_set_long_mode(label3, LV_LABEL_LONG_WRAP);
    lv_label_set_text(label3, "aaaaabbbbb");
    lv_obj_set_width(label3, 88);
    lv_obj_align(label3, LV_ALIGN_TOP_LEFT, 0, 80);
}

### Expected behavior

### Screenshots or video
[image]

kisvegabor, 2022-11-09:
For me the label's width is 105, probably because we used a different fonts.

I printed the label's default width, set the width to that value and it working well:
[image]

```c
    lv_obj_t * label1 = lv_label_create(bg);
    lv_label_set_long_mode(label1, LV_LABEL_LONG_CLIP);
    lv_label_set_text(label1, "aaaaabbbbb");
    lv_obj_align(label1, LV_ALIGN_TOP_LEFT, 0, 0);
    
    lv_obj_update_layout(label1);
    printf("%d\n", lv_obj_get_width(label1));

    lv_obj_set_width(label1, 105);

    //LV_LABEL_LONG_DOT
    lv_obj_t * label2 = lv_label_create(bg);
    lv_label_set_long_mode(label2, LV_LABEL_LONG_DOT);
    lv_label_set_text(label2, "aaaaabbbbb");
    lv_obj_set_size(label2, 105, 20);
    lv_obj_align(label2, LV_ALIGN_TOP_LEFT, 0, 40);

    //LV_LABEL_LONG_WRAP
    lv_obj_t * label3 = lv_label_create(bg);
    lv_label_set_long_mode(label3, LV_LABEL_LONG_WRAP);
    lv_label_set_text(label3, "aaaaabbbbb");
    lv_obj_set_width(label3, 105);
    lv_obj_align(label3, LV_ALIGN_TOP_LEFT, 0, 80);
```

yin2021, 2022-11-09:
> For me the label's width is 105, probably because we used a different fonts.
>
> I printed the label's default width, set the width to that value and it working well: [image]
>
> ```c
>     lv_obj_t * [...]

1.

yin2021, 2022-11-09:
> For me the label's width is 105, probably because we used a different fonts.
>
> I printed the label's default width, set the width to that value and it working well: [image]
>
> ```c
>     lv_obj_t * [...]

1. I have try your code，and i think 105 is not a accuracy value of label width. You can execute the same code  only reducing the width of the three labels, and you will find the same problem.  
[image]

2. In the above picture, the width of the three label is same, however, when serting LV_LABEL_LONG_CLIP mode for label1, ten characters of the string "aaaaabbbbbc" can be shown in screen and the last character 'c' can be shown partly, which meaning the width of label is enough for ten characters. So, I think when it is LV_LABEL_LONG_DOT mode, there should not be write dots at the end of the label's text, because the string “aaaaabbbbb” is not too long for the label2. When it is LV_LABEL_LONG_WRAP mode, the object height should not be expand for label3. Do you think?

3. In the above picture, we can find label2 and label3 is shorter than label1. I think the label should not Shorten when LV_LABEL_LONG_WRAP mode and LV_LABEL_LONG_DOT mode are met.

kisvegabor, 2022-11-10:
>  lv_obj_update_layout(label1);
    printf("%d\n", lv_obj_get_width(label1));

It tell the width what LVGL thinks the width is. Considering all kerning, letter spaces etc, LVGL calculated 105px width. 

The case of extra "c" letter is interesting though. `lv_label` sets some extra draw area for itself to allow drawing a little bit of content outside of the actual bounding boxes. This hack was required because there are some handwritten style fonts which has some "tendrils" overflowing the assumed character box. However this extra draw area shows a little bit from the next character in clip mode.

So I assume the calculated width is correct and the clipping is not accurate.

yin2021, 2022-11-10:
> 

> It tell the width what LVGL thinks the width is. Considering all kerning, letter spaces etc, LVGL calculated 105px width.
>
> The case of extra "c" letter is interesting though. `lv_label` sets some [...]
Maybe I misunderstood LV_LABEL_LONG_CLIP mode. But, i am confuse in LV_LABEL_LONG_DOT mode and LV_LABEL_LONG_WRAP mode. In the picture below, LV_LABEL_LONG_DOT mode happends in the label2. However, the label2 show less characters than label1. Why is that?

code in following：
void lv_example_label_1(lv_obj_t *par)
{
    lv_obj_t * bg = lv_obj_create(lv_scr_act());
    lv_obj_set_size(bg, 200, 200);
    lv_obj_align(bg, LV_ALIGN_TOP_LEFT, 0, 0);

    lv_obj_t *btn1 = lv_obj_create(bg);
    lv_obj_set_style_radius(btn1, 0, 0);
    lv_obj_set_style_bg_opa(btn1, LV_OPA_100, 0);
    lv_obj_set_style_bg_color(btn1, LV_COLOR_BLACK, 0);
    lv_obj_set_size(btn1, 78, 22);
    lv_obj_align(btn1, LV_ALIGN_TOP_LEFT, 20, 20);

    lv_obj_t * label1 = lv_label_create(btn1);
    lv_obj_set_style_text_color(label1, LV_COLOR_RED, 0);
    lv_label_set_long_mode(label1, LV_LABEL_LONG_DOT);
    lv_obj_set_size(label1, 78, 20);
    lv_label_set_text(label1, "aaaaabbb");
    lv_obj_update_layout(label1);
    printf("label1 width: %d\n", lv_obj_get_width(label1));
    lv_obj_align(label1, LV_ALIGN_CENTER, 0, 0);

    lv_obj_t *btn2 = lv_obj_create(bg);
    lv_obj_set_style_radius(btn2, 0, 0);
    lv_obj_set_style_bg_opa(btn2, LV_OPA_100, 0);
    lv_obj_set_style_bg_color(btn2, LV_COLOR_BLACK, 0);
    lv_obj_set_size(btn2, 78, 22);
    lv_obj_align(btn2, LV_ALIGN_TOP_LEFT, 20, 45);

    lv_obj_t * label2 = lv_label_create(btn2);
    lv_obj_set_style_text_color(label2, LV_COLOR_RED, 0);
    lv_label_set_long_mode(label2, LV_LABEL_LONG_DOT);
    lv_obj_set_size(label2, 78, 20);
    lv_label_set_text(label2, "aaaaabbbb");
    lv_obj_update_layout(label2);
    printf("label2 width: %d\n", lv_obj_get_width(label2));
    lv_obj_align(label2, LV_ALIGN_CENTER, 0, 0);
}

kisvegabor, 2022-11-13:
I'd like to test your code as it's but I'm using a different font and the label sizes are different. 
So which font do you use?

yin2021, 2022-11-14:
> I'd like to test your code as it's but I'm using a different font and the label sizes are different. So which font do you use?

As you see, i do not specify any font. So, maybe it is default font.
[image]

/*Always set a default font*/
#define LV_FONT_DEFAULT &lv_font_montserrat_14

yin2021, 2022-11-14:
I am sorry for ambiguous font. You can try the following code, which specified font(including lv_font_montserrat_14 and lv_font_montserrat_18)

void lv_example_label_1(lv_obj_t *par)
{
    int font_14_len = 78;
    int font_18_len = 78 + 20;

    lv_obj_t * bg = lv_obj_create(lv_scr_act());
    lv_obj_set_size(bg, 200, 200);
    lv_obj_align(bg, LV_ALIGN_TOP_LEFT, 0, 0);

    //lv_font_montserrat_14
    lv_obj_t *btn1 = lv_obj_create(bg);
    lv_obj_set_style_radius(btn1, 0, 0);
    lv_obj_set_style_bg_opa(btn1, LV_OPA_100, 0);
    lv_obj_set_style_bg_color(btn1, LV_COLOR_BLACK, 0);
    lv_obj_set_size(btn1, font_14_len, 22);
    lv_obj_align(btn1, LV_ALIGN_TOP_LEFT, 20, 20);

    lv_obj_t * label1 = lv_label_create(btn1);
    lv_obj_set_style_text_font(label1, &lv_font_montserrat_14, 0);
    lv_obj_set_style_text_color(label1, LV_COLOR_RED, 0);
    lv_label_set_long_mode(label1, LV_LABEL_LONG_DOT);
    lv_obj_set_size(label1, font_14_len, 20);
    lv_label_set_text(label1, "aaaaabbb");
    lv_obj_update_layout(label1);
    printf("label1 width: %d\n", lv_obj_get_width(label1));
    lv_obj_align(label1, LV_ALIGN_CENTER, 0, 0);

    lv_obj_t *btn2 = lv_obj_create(bg);
    lv_obj_set_style_radius(btn2, 0, 0);
    lv_obj_set_style_bg_opa(btn2, LV_OPA_100, 0);
    lv_obj_set_style_bg_color(btn2, LV_COLOR_BLACK, 0);
    lv_obj_set_size(btn2, font_14_len, 22);
    lv_obj_align(btn2, LV_ALIGN_TOP_LEFT, 20, 45);

    lv_obj_t * label2 = lv_label_create(btn2);
    lv_obj_set_style_text_font(label2, &lv_font_montserrat_14, 0);
    lv_obj_set_style_text_color(label2, LV_COLOR_RED, 0);
    lv_label_set_long_mode(label2, LV_LABEL_LONG_DOT);
    lv_obj_set_size(label2, font_14_len, 20);
    lv_label_set_text(label2, "aaaaabbbb");
    lv_obj_update_layout(label2);
    printf("label2 width: %d\n", lv_obj_get_width(label2));
    lv_obj_align(label2, LV_ALIGN_CENTER, 0, 0);

    //lv_font_montserrat_18
    lv_obj_t *btn3 = lv_obj_create(bg);
    lv_obj_set_style_radius(btn3, 0, 0);
    lv_obj_set_style_bg_opa(btn3, LV_OPA_100, 0);
    lv_obj_set_style_bg_color(btn3, LV_COLOR_BLACK, 0);
    lv_obj_set_size(btn3, font_18_len, 22);
    lv_obj_align(btn3, LV_ALIGN_TOP_LEFT, 20, 85);

    lv_obj_t * label3 = lv_label_create(btn3);
    lv_obj_set_style_text_font(label3, &lv_font_montserrat_18, 0);
    lv_obj_set_style_text_color(label3, LV_COLOR_RED, 0);
    lv_label_set_long_mode(label3, LV_LABEL_LONG_DOT);
    lv_obj_set_size(label3, font_18_len, 20);
    lv_label_set_text(label3, "aaaaabbb");
    lv_obj_update_layout(label3);
    printf("label3 width: %d\n", lv_obj_get_width(label3));
    lv_obj_align(label3, LV_ALIGN_CENTER, 0, 0);

    lv_obj_t *btn4 = lv_obj_create(bg);
    lv_obj_set_style_radius(btn4, 0, 0);
    lv_obj_set_style_bg_opa(btn4, LV_OPA_100, 0);
    lv_obj_set_style_bg_color(btn4, LV_COLOR_BLACK, 0);
    lv_obj_set_size(btn4, font_18_len, 22);
    lv_obj_align(btn4, LV_ALIGN_TOP_LEFT, 20, 110);

    lv_obj_t * label4 = lv_label_create(btn4);
    lv_obj_set_style_text_font(label4, &lv_font_montserrat_18, 0);
    lv_obj_set_style_text_color(label4, LV_COLOR_RED, 0);
    lv_label_set_long_mode(label4, LV_LABEL_LONG_DOT);
    lv_obj_set_size(label4, font_18_len, 20);
    lv_label_set_text(label4, "aaaaabbbb");
    lv_obj_update_layout(label4);
    printf("label4 width: %d\n", lv_obj_get_width(label4));
    lv_obj_align(label4, LV_ALIGN_CENTER, 0, 0);
}

[image]